linuxboot/heads #1429: Coreboot Utilities

bluesteal, 2023-07-01:
Security assessment of Coreboot Utilities

-> What prevents a Coreboot utility from being a malicious binary? The list of Coreboot Utilities is here:
https://doc.coreboot.org/util.html

-> The ability to understand which Coreboot utilities have been included in a build and the ability to audit those binaries and 
attest their integrity.

tlaurion, 2023-07-01:
@bluesteal there is a major difference between utils available at build time under the src directory of a coreboot version and what is packed under initrd as tools.

Also, most of the tools are available as source code and built by coreboot to build the ROM itself.

As of now, none of the tools presented at the link presented are provided under Heads.

With the exception of cbmem
https://github.com/osresearch/heads/blob/91f65bed793c0165203b1cbb9ac4f390f15ffa0e/Makefile#L533

Built from.each version of coreboot version used to build a ROM (which permits to access coreboot boot log, stage boot time, tcpa log to review TPM extend operations).

flashtool module produces a binary of a similar name of the above called cbfs, used under heads to extract/inject from coreboot filesystem user related files from the ROM (gpg public keyring, gpg trustdb, config.user if user saved settings back to ROM).

Other then that, none of those tools are included. Bit omcr again, the source code is there for anybody to inspect and I'm not sure where and why you raise a concern about those tools.

You can see under hashes.txt of a ROM what tools are included in a build.

tools.cpio contains modules/ build output (libs and tools) and are detailed with hashes I'm the files listed above prior of being packed.

heads.cpio contains scripts and all other files contained under initrd source dir and detailed above that cpio in that hashes
txt

modules.cpio contains packed kernel modules compiled as modules which are loaded on demand and are detailed above that cpio under hashes.text

Then tools.cpio heads.cpio and modules.cpio are packed under initrd.cpio.xz and stitched with the kernel (bzImage) as coreboot payload.

> -> The ability to understand which Coreboot utilities have been included in a build and the ability to audit those binaries and attest their integrity.

Unfortunately to accomplish that goal, you need to get how those are built and audit the code. tools packed in tools.cpio are built by musl-cross-make and should be reproducible across all boards for a same architecture. As for the inspection of the source code, you can question integrity upstream at https://review.coreboot.org for the commit that corresponded to the one at release used for a board, see its history and collaborators.

bluesteal, 2023-07-01:
There is concern about firmware because of its privilege level and also because there has been some firmware for Android devices found in the wild which contain malicious modular plugins which allow for remote access. Why couldn't a similar attack affect Coreboot or Libreboot?

bluesteal, 2023-07-01:
These plugins were firmware binaries, maybe similar to cbmem or cbfs. I am using Coreboot as an example of how that attack might be accomplished for Android devices

tlaurion, 2023-07-04:
> There is concern about firmware because of its privilege level and also because there has been some firmware for Android devices found in the wild which contain malicious modular plugins which allow for remote access. Why couldn't a similar attack affect Coreboot or Libreboot?

@bluesteal your concerns are valid, but requires to dig into the project to understand how it works, unfortunately.
Your Android case, if it was more directed to direct example, could or couldn't be tackled more directly but the question here is being directed at coreboot tools included into the builds, which was already addressed, so i'm not quite sure how to address it further.

> These plugins were firmware binaries, maybe similar to cbmem or cbfs. I am using Coreboot as an example of how that attack might be accomplished for Android devices

Heads, just as Android in your example, are a combination of firmware and linux as an OS. When Android boots, its normally a combination of u-boot as a bootloader and then linux as an OS, in verified boot with dual R/O R/W partition. As opposed to Heads, Android will boot the monolithic linux kernel; there is no kexec involved into the final boot of the operating system.

Here, coreboot initializes the platform. Loads and apply microcode updates if present, then load FSP blobs from intel if requires for the platform, and then loads and boot the linux kernel which is coreboot's payload. Coreboot measures itself and the payload and then boots it. Then Heads measures its own stuff and boots if the measurements are as expected. The tools packed inside of Heads, if I understand your concern well, are not coreboot tools as per referenced link, but result of compilation of modules (see modules directory under Heads repository) and some patches to those modules if requires (see patches directory under Heads).

Those modules measure the source code downloaded prior of extracting those modules, ant then those modules are compiled through a locally bootstrapped cross-compiler, the goal being to produce the same binaries to be packed under the final ROM.

The guarantees here are that the source tarballs downloaded are in their authentic state. That authentic state doesn;t imply that there is no vulnerabilities in the code downloaded and used under Heads, simply that they are as intended prior of being patched and compiled, then packed as a linux coreboot payload. Heads is to be used in a "directed" way from the user, that is, using menus to accomplish tasks in an expected process producing expected goals. Coreboot+Heads as opposed to UEFI permits access to a standard "recovery shell", that is a shell containing linux standard tools, which is definitely more powerful the EFI shell, simply because those tools are there and exists. But if you, or someone else, use that shell in the goal of tampering the firmware, then the firmware measurements would be invalidated and Heads would notice. Heads hence permits to validate the firmware, /boot integrity and LUKS header integrity prior of booting. If any of those changes, Heads will notice and refuse to boot. That is the guarantee. Then, when Heads boots the final OS, it parses syslinux/grub config, measured and verified against detached signatures from the user, and kexec (replace the memory kernel) by the one of the OS with OS provided initrd. The only things that is passed to the final OS here are: the framebuffer address to be reused by the final OS, and additional cpio containing a disk encryption key that was unsealed from TPM if measurements+TPM disk unlock key passphrase were good and that cpio containing the disk unlock key and crypttab file is used to unlock the LUKS container prior of the OS taking total control of the system at this point. 

Depending of if if the firmware depends on FSP or other blobs, the firmware has as much control as a firmware can over the platform to initialize it. Then linux acts as an OS (Heads) but is replaced in memory by another OS on kexec call. What was present in ramfs (Heads) vanishes and is replaced by what the final OS boots into. If the OS can write to the SPI (mtd, `iomem=relaxed` kernel boot option) and coreboot+Heads disn't finalize platform lockdown, then the OS *could* write into the firmware (SPI). If that happens, Heads will notice on next boot since HOTP/TOTP sealed measurements inside of the TPM would now be invalid, and it would be impossible for a user to type his TPM disk unlock key passphrase to have the TPM unseal the TPM disk unlock key.

As for the whole stack of software packed into Heads, one last time if Android is the comparison. Android relies, or not, on vboot for verified boot, which security is bound to an external entity signing the builds. This guarantees that a third party signed those builds, but is no guarantee that what is included in those builds has no vulnerability, or worse, backdoors or software intended to cause harm. When something is packed and you relinquish your own right of inspecting the build and or the software included in a build, then you trust that third party.

Heads is based on the concept of "trust but verify". Unfortunately, you have to trust me, and the code reviewers that reviewed my pull requests, that the scripts, build recipes and patches produced were not ill intended. 

[... cut ...] re you want to type your TPM released LUKS decryption key to the final OS. So on a default boot path: no network enabled. No USB storage module. No USB keyboard drivers.

I hope this addresses your concerns. There are articles out there that are produced by really good technical writers and simplifies the reading so much more then I will ever be able to do. 
One of those articles is this one: https://tech.michaelaltfield.net/2023/02/16/evil-maid-heads-pureboot/

I think this issue can be closed?

bluesteal, 2023-07-05:
Is there a minimal version of heads with only required modules compiled into the build?

tlaurion, 2023-07-05:
@bluesteal I don't understand where this is going.

Open a board configuration. Let's say x230-maximized board config.

You will see there the modules and variables referring to modules that are defined to be packed under either modules.cpio heads.cpio or tools.cpio.

Then look at modules/ directory to see what they are referred to and the output which are the libraries and binaries compiled and added into the initrd.

As said before, you want to see hashes.txt of a board build to see exactly what are the final hashes of the actual binaries and libraries added into the tools.cpio that will directly answer your question.

The global Makefile explains the logic of how things are packed and in which order.

Heads will build the cross compiler. To do so download Linux kernel, install headers under install directory, used as sysroot for cross compilation. Then packages instructed to be built by board config are built, following their dependencies. Then coreboot is built. The Linux kernel is added alongside its initrd as coreboot payload. That's it.

For heads to be heads, fbwhiptail/whiptail is needed, alongside qrencode, totp utils, TPM util, flashrom, flashtools, pciutils etc etc etc.

Boards are already trying to install the bare minimal. If you are interested in such board config, where space is really limited, look into x230-legacy board, which is a struggle to maintain. Such board loose kernel modules, networking possibilities, has console based menu (whiptail) as opposed to fbwhiptail, etc.

Users wanted to have ifdtool and nvramutil from libreboot to be able to generate random Mac address and reinject the GBE file into the spi flash. To do so, those two additional binaries would need to be added. Or maybe newer flashrom would be able to only extract GBE, nvramutil being able to generate a new one and have it flashed back into spi, we will know soon.

But then other users want to modify coreboot settings. That would be nvramui. That tool would also need to be added, where cbfs will be able to extract/inject it back.

So question would be: what is minimal tools needed? Everyone wants everything and minimal at the same time.

This is why board configurations are there. To he customized to your needs, where the ones maintained are the best tradeoff the community defines what is needed and what is not.

Closing this issue before it goes in all directions.

closed 2023-07-05 by tlaurion