linuxboot/heads #1435: Including BITS and CHIPSEC as Coreboot Payloads

bluesteal, 2023-07-06:
Is it currently possible to include BITS and/or CHIPSEC to the Coreboot payload as binary utilities?

https://www.osfc.io/2018/talks/bits-and-chipsec-as-coreboot-payloads/

tlaurion, 2023-07-06:
@piotr?

pietrushnic, 2023-07-06:
Yes, it is possible. 3mdeb did that in the past, as presented in the above talk. We did that for validation purposes.

@bluesteal what would be your goal?

bluesteal, 2023-07-06:
just want to be able to understand and analyze firmware in general, it would be a handy tool.

pietrushnic, 2023-07-06:
@bluesteal I guess you could find some branches in our repos, but I doubt it can be compiled without any issues.

bluesteal, 2023-07-11:
do you mean dependencies?

pietrushnic, 2023-07-11:
Those branches are 5 years old:
* https://github.com/3mdeb/coreboot/tree/bits_chipsec_payload
* https://github.com/3mdeb/bits/pull/2
* https://github.com/3mdeb/chipsec - but it should be figure out from build process of coreboot if it is even included