linuxboot/heads #1447: Qemu-coreboot-fbwhiptail always built regardless of board selection

Tickmeister1, 2023-08-01:
## Context of the Build

**1. What board are you trying to build?**  UNTESTED_kgpe-d16_workstation and UNTESTED_kgpe-d16_server and x230-hotp-maximized

**2. What repository:branch are you using to build from?**
- [ X] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] 4.17
- [ X] Other (please specify) 4.11 and 4.19

**4. In building the rom where did you get the blobs?**
- [ X] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [X ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ X] No

**6. What operating system are you using**  Debian 11

*please specify*

## Please describe the problem
Regardless of which board rom I attempt to compile, an incorrectly sized rom called "heads--v0.2.0-1713-g06b1b09.rom" gets built and placed in heads/build/x86/qemu-coreboot-fbwhiptail-tpm1

**Describe the bug**
A clear and concise description of what the bug is.

Expected folder named for the respective board to appear in heads/build/x86.  This does not occur.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Expected folder named for the respective board to appear in heads/build/x86.  This does not occur.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.  Run on x230 laptop, itself using heads.  Vanilla denian 11 on bare metal.

tlaurion, 2023-08-01:
@Tickmeister1 : [qemu-coreboot-fbwhiptail-tpm1 is default board target when no board specified to be built](https://github.com/osresearch/heads/blob/06b1b0948d911ab1ee15d08cc4776e41889349cd/Makefile#L43) through `make BOARD=board_name_under_board_dir`

Let me know if that was basic issue and close accordingly after pointing what followed instructions lead you to build without BOARD=XYZ ?
----

Can try to replicate under debian-11 with the packages installed as per CircleCI 
- [ ] `make BOARD=x230-maximized` (needs blobs/xx30 download/extract script ran first to download+extrct+neuter ME, just as under CircleCI, will need to fix this so board calls it itself, just like what the t440p board config brought as a fix)
- [ ] `make BOARD=UNTESTED_kgpe-d16_workstation`
- [ ] `make BOARD=UNTESTED_kgpe-d16_server`

Tickmeister1, 2023-08-02:
The issue was that I was not capitalizing BOARD in the make command.

closed 2023-08-02 by Tickmeister1