linuxboot/heads #1461: EFIFB support next steps

tlaurion, 2023-08-10:
- [ ] add https://review.coreboot.org/c/coreboot/+/76564/2 under coreboot 4.19 patches and other forks patches? 
- [ ] Add imagemagik into CircleCI apt dependencies so that any jpeg can be injected in rom, at condition that that jpeg is smaller then framebuffer (it will be centered in framebuffer per other patch).
- [ ] Push forks to add those coreboot patches (@krystian-hebel)
- [ ] Push Purism to add fbwhiptail patch so efifb/simplefb kernel driver can be used (works flawlessly) @JonathonHall-Purism 
- [ ] Adapt nv41 and nv50 board once this PR is  merged (@daringer)

_Originally posted by @tlaurion in https://github.com/osresearch/heads/issues/1403#issuecomment-1666530226_

JonathonHall-Purism, 2023-08-10:
@tlaurion The 1366x768 patch is in fbwhiptail 1.2 :tada:

tlaurion, 2023-08-10:
@JonathonHall-Purism so I should add it under  #1403?

tlaurion, 2023-08-11:
Also to be noted, dgpu flavors of t530/w530 are currently enabling optional rom in "insecure mode", where coreboot is said to be significantly slower initializing if we were to change the defaults:

```
CONFIG_PCI_OPTION_ROM_RUN_YABEL:                                                                                                              │  
  │                                                                                                                                               │  
  │ If you select this option, the x86emu CPU emulator will be used to                                                                            │  
  │ execute PCI Option ROMs.                                                                                                                      │  
  │                                                                                                                                               │  
  │ This option prevents Option ROMs from doing dirty tricks with the                                                                             │  
  │ system (such as installing SMM modules or hypervisors), but it is                                                                             │  
  │ also significantly slower than the native Option ROM initialization                                                                           │  
  │ method.                                                                                                                                       │  
  │                                                                                                                                               │  
  │ This is the default choice for non-x86 systems.
```

tlaurion, 2023-08-23:
Seems like qubesos is not providing efifb in compiled kernels @marmarek ? 

Would be a good idea to provide it so that early print messages can be shown on consoles prior of i915drmfb taking ownership if the framebuffer, otherwise messages show on console after i915drmfb is loaded and ownership of FB is active.

marmarek, 2023-08-23:
And one more thing: dom0's kernel config isn't really relevant for early Xen messages. For that, there is a bunch of command line arguments, including `vga=current` which might be enough here. Xen does support getting framebuffer parameters via EFI, but again - I don't think this boot path is used under Heads at all.

tlaurion, 2023-08-23:
> We do have `CONFIG_FB_EFI=y` in our kernel. But, there is also `CONFIG_SYSFS_SIMPLEFB=y` (which IIUC is the newer approach for the same thing).
> That said, I'm a bit confused how it is supposed to work. Heads isn't booting OS via EFI entry point and does not synthesize EFI system table, no? So, how OS would use EFI for locating the framebuffer? What am I missing? 

@marmarek : 
Since https://github.com/osresearch/heads/pull/1403 Heads avoids providing specific DRM/FB drivers, thanks to patches added to coreboot so that GOP/native GFX/vbios rom/libgfxinit offers a efifb compatible framebuffer to the kernel.

Heads kernel then kexec's into next kernel passing the FB address to next kernel, without needing to hack around newer kernel's restrictions not exposing addresses which otherwise required Heads to taint it's kernel exposing FB address (considered dangerous) and disabling framebuffer compression on i915drm module and a hack on kexec to force passing i915fb FB address as VLFB. Kexec detects Heads FB as efifb and gently provides the FB address to the next kernel without any previously needed hacks.

As if today (rudimentary testing booting latest 4.2 ISO with bug report since inst.repo is not found) Heads stays in the dark until i915drmfb takes ownership of the console.

Attempts under Heads to support simplefb didn't work since it is not possible to pass simplefb FB address through VLFB hack, there is no workaround permitting that driver to taint kernel in an insecure flashing like it is possible with i915drmfb.

I am confused into why simplefb is not kicking in prior of i915drmfb taking ownership of the FB under qubes as if now and will need to check this out.

You are right saying that OS deprecation plan for legacy boot support is to remove all advanced DRM drivers from their initrd, letting simplefb, simpledrm and efifb there for the moment, leaving initrd construction with proper advanced drm modules enablement to the final installed OS to construct it from detected GPUs.

Installation media consequently provide/will provide simpledrm/simplefb/efifb to not bloat initrd/unified kernels with all DRM/FB drivers that exist anymore and rely on UEFI to at least provide efifb compatible drivers which can after that be switched to simplefb/simpledrm or more advanced DRM drivers.

tlaurion, 2023-08-23:
> And one more thing: dom0's kernel config isn't really relevant for early Xen messages. For that, there is a bunch of command line arguments, including `vga=current` which might be enough here. Xen [...]

@marmarek AFAIK coreboot tables are now exposed but this is irrelevant here? How to check I'm not presently aware

tlaurion, 2023-08-24:
> @marmarek : Since #1403 Heads avoids providing specific DRM/FB drivers, thanks to patches added to coreboot so that GOP/native GFX/vbios rom/libgfxinit offers a efifb compatible framebuffer to the [...]

Maybe its more practical to see with an example.
Under Heads linux-qemu.config under master, here is when the framebuffer is owned by efifb (QEMU TCG here, since qemu under Qubes so cannot have kvm):
[    4.837119] efifb: probing for efifb
[    4.841420] efifb: framebuffer at 0xc0000000, using 3072k, total 3072k
[    4.842005] efifb: mode is 1024x768x32, linelength=4096, pages=1
[    4.842428] efifb: scrolling: redraw
[    4.842767] efifb: Truecolor: size=8:8:8:8, shift=24:16:8:0
[    4.917478] Console: switching to colour frame buffer device 128x48
[    4.983187] fb0: EFI VGA frame buffer device

I posted a full [rdosreport.txt](https://github.com/QubesOS/qubes-issues/files/12415256/rdsosreport.txt) file under https://github.com/QubesOS/qubes-issues/issues/8443. 

Excerpts::

```
[    1.928708] localhost kernel: pci 0000:00:00.0: [8086:0154] type 00 class 0x060000
[    1.929042] localhost kernel: pci 0000:00:02.0: [8086:0166] type 00 class 0x030000
[    1.929107] localhost kernel: pci 0000:00:02.0: reg 0x10: [mem 0x82c00000-0x82ffffff 64bit]
[    1.929152] localhost kernel: pci 0000:00:02.0: reg 0x18: [mem 0x90000000-0x9fffffff 64bit pref]
[    1.929185] localhost kernel: pci 0000:00:02.0: reg 0x20: [io  0x1000-0x103f]
[    1.929259] localhost kernel: pci 0000:00:02.0: Video device with shadowed ROM at [mem 0x000c0000-0x000dffff]
[    1.952993] localhost kernel: pci 0000:00:02.0: vgaarb: setting as boot VGA device
[    1.952993] localhost kernel: pci 0000:00:02.0: vgaarb: bridge control possible
[    1.952993] localhost kernel: pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    1.953005] localhost kernel: vgaarb: loaded
[   19.375778] localhost kernel: i915 0000:00:02.0: [drm] VT-d active for gfx access
[   19.375889] localhost kernel: i915 0000:00:02.0: vgaarb: deactivate vga console
[   19.468346] localhost kernel: tmpfs: Unsupported parameter 'huge'
[   19.468365] localhost kernel: i915 0000:00:02.0: [drm] Transparent Hugepage support is recommended for optimal performance when IOMMU is enabled!
[   19.468380] localhost kernel: i915 0000:00:02.0: [drm] DMAR active, disabling use of stolen memory
[   19.577394] localhost kernel: i915 0000:00:02.0: [drm] Skipping intel_backlight registration
[   19.578870] localhost kernel: [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[   19.579637] localhost kernel: ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[   19.579909] localhost kernel: input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input8
[   19.775834] localhost kernel: acpi device:0d: registered as cooling_device5
[   19.872710] localhost kernel: fbcon: i915drmfb (fb0) is primary device
[   20.680374] localhost kernel: Console: switching to colour frame buffer device 170x48
[   20.700506] localhost kernel: i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
```

Fb0 is redrawnn at around 20 seconds on real hardware and around 5s on qemu.

-----

Under Heads, it requires of the kernel config to be aware of EFI.
You can see the changes that were required under https://github.com/osresearch/heads/pull/1403/files#diff-816b79cd9a1192fa8cf78b9737f0d3fed5a943a0d2c952da591f8a2446edc1db
- Yes, the kernel needs to be EFI compatible (`CONFIG_EFI=y`)
- Coreboot tables awareness needs activated in kernel (`CONFIG_GOOGLE_COREBOOT_TABLE=y`)
- And then, EFI->EFIFB will pick up fb early, and then i915drmfb will take ownership later on. The user will get fb really early.

tlaurion, 2023-08-28:
@marmarek interesting https://forum.qubes-os.org/t/how-to-kexec-into-qubes-from-linux-livecd/17712/22

Might explain why https://github.com/marmarek/qubes-linux-kernel/commit/c4ff0e4e0b3a9e881b28851575e62e7fac823389 is not enough

@JonathonHall-Purism any hunch?

tlaurion, 2023-08-29:
Hmmm 2.0.27 supports device tree https://github.com/horms/kexec-tools/commit/806711fca9e9d52a677bf090565c32c858f2b12e

tlaurion, 2023-11-08:
All OP tasks under #1522. 

Q4.2 still cannot offer EFIFB direct console output on installed systems which takes 6 seconds under x230 to show console once i915 takes ownership of framebuffer. On Q4.1/Q4.2, booting to a console at kexec call to working console takes 20 seconds per #1522 state as of today, but I think this will take more time to resolve.

tlaurion, 2023-12-12:
Unless QubesOS dom0 can use efifb for early console ownership and then switch to i915+drm or whatever else needed there (something unknown is missing) all of this is fixed under #1522 now merged under master. 

Maybe reopen the ticket later if we understand what is missing from coreboot->libgfxinit->tampoline patch to expose coreboot fb to linux efifb->kexec xen + kernel in multiboot->dom0 kernel seeing coreboot initialized efifb compliant framebuffer. But I have not figured that out yet.

@marmarek again, if you feel like reopening this, as of today, Q4.2 installer takes 20 seconds to have i915+drm initiaalized (not sure why) where installed system takes 6 seconds to init i915+drm and show console after kexec call on x230.

Closing

tlaurion, 2025-03-25:
Seems like Ubuntu won't add efifb to initramfs https://github.com/linuxboot/heads/issues/1641#issuecomment-2752521480