linuxboot/heads #1475: Adiantum crypto module for CPUs without AES-NI

natterangell, 2023-08-23:
The Thinkpad x230i is supported by the x230* roms, and only differs from the regular x230 in that it shipped with a non-upgradable Ivy Bridge i3 CPU which doesn't support AES-NI. 

LUKS is compatible with adiantum, developed by Google primarily for low-powered devices that cannot do hardware accelerated AES encryption. It is included from kernel 5.0 and cryptsetup 2.x (not sure what version) supports it.

This software cipher is a lot faster on CPUs without AES-NI.

[image]

While this might be an edge case, given that most Heads users will not see much difference due to AES-NI support, it might be worth considering for the x230i, and maybe other boards (i3 x220, others?)

I can build my own rom to test, but I wonder what options must be set for building kernel (and maybe cryptsetup?) with support included? With [#1398](https://github.com/osresearch/heads/pull/1398) merged it should be possible.

natterangell, 2023-08-23:
It might seem like the following is enough:
CONFIG_CRYPTO_ADIANTUM=y
CONFIG_CRYPTO_NHPOLY1305=y
CONFIG_CRYPTO_CHACHA20=y

tlaurion, 2023-08-23:
@natterangell : Heads uses cryptsetup for two things and is unrelated to what the final OS is providing:
- cryptsetup open operations to unlock container under recovery shell.
- cryptsetup-reencrypt operations, which reuses the current container settings

To change those, the user either has to:
- Manually call cryptsetup-reencrypt under Heada and pass the desired algo options (where Heads would need those additional algos)
- Manually go to shell under OS installer and manually partition the drive at install, make the LUKS container available to the installer and then continue from there. Again here, your custom chosen algo would need to be supported under Heads as well.

Now my opinions:
- If this is really desired, we could now duplicate x230 and other boards to point to a different linux kernel having those crypto options selected. (Note that you are the only reported x230i user, since Heads is mostly used while not limited to QubesOS where i3 do not provide qubesos requirements and therefore are not chosen by end users if goal is to use QubesOS. I am not aware of any other board owners of xx20 or xx30 which have an i3 vs i5 i7 which have AES-NI on those older models
- linux-x230-maximized is shared across xx20 and xx30 boards for maintainership simplicity, where xx30 ony have 8mb of flash so adding AES crypto without AES-NI is find (and merged) but readding crypto modules used only by some for coner case situations and where OS installers do not create LUKS containers with those options is really geeeky and I have some reluctance adding those into shared linux-x230-maxized.config where need might be later on to remove it.
- You can definitely clone x230-maximized board config and duplicate linux-x230-maximized.config to linux-x230i-maximized config, specifying under the board config that this device is an i3 for others to use, but AFAIK x230i are pretty rare out there.

Thoughts?

natterangell, 2023-08-23:
Completely fine with me not to create a separate x230i board. I suspected as much that few (if anyone) use this system, this is a laptop I got more or less for free, and not my main system anyway.

I'll experiment a litte when I get around to it and document in fork whether it works.

closed 2023-08-23 by natterangell