linuxboot/heads #1480: Use TPM2_Quote for attestation

DemiMarie, 2023-08-25:
**Is your feature request related to a problem? Please describe.**
TOTP requires synchronizing time, and HOTP requires synchronizing state.

**Describe the solution you'd like**
Use [TPM2_Quote] for attestation.  TPM2_Quote signs a set of PCRs and a nonce with a key in the TPM.  It is possible to have a key that can only be used for TPM2_Quote and cannot be exported from the TPM.

**Describe alternatives you've considered**
None.

**Additional context**
This is more secure than TOTP or HOTP for two reasons:

1. The secret used to sign the quote never leaves the TPM, making it much less likely to be accidentally leaked.
2. The qualifying data can be much longer than a 6 digit passcode.

[TPM2_Quote]: https://tpm2-tools.readthedocs.io/en/latest/man/tpm2_quote.1

tlaurion, 2023-12-21:
@DemiMarie Not directly related but https://github.com/linuxboot/heads/issues/135#issuecomment-289150160 was replicated and works.

Also note that tpm-luks is a thing on which we could potentially try to comply to, so that unsealing of nvram based on sealed secret could be unsealed without heads having to construct cpio and pass to kexec. https://github.com/Rohde-Schwarz/TrustedGRUB2/issues/2

As for Quote, we would have to made the code compliant for both TPM1.2/TPM2 cases, which needs time and effort.

Related to #1550

JonathonHall-Purism, 2023-12-21:
This is a great idea.  The Nitrokey firmware will have to support this sort of challenge-response in addition to the reverse HOTP mode we currently use (init with the desired PCR state and public key, challenge with a nonce, respond with the quote including the nonce).  Seems like something it could do, I couldn't guess the complexity through as I have not dug into the Nitrokey firmware before.

DemiMarie, 2023-12-22:
Password + TPM would be ideal, but only discrete TPMs provide strong enough protection to allow weak passwords to be used.