linuxboot/heads #1481: make clear what board build does what

firefoxlover, 2023-08-27:
**Is your feature request related to a problem? Please describe.**
I have a T430 but there are three board names

**Describe the solution you'd like**
I would like an explanation of the builds, what they are and what the differences are.

**Describe alternatives you've considered**
guessing its probably to hotp version, legacy? I dont know!

tlaurion, 2023-08-27:
@firefoxlover hotp vs non-hotp https://osresearch.net/FAQ/#hotp-vs-totp

Maximized va legacy https://osresearch.net/Prerequisites#legacy-vs-maximized-boards

How would you prefer to see this explained?

tlaurion, 2023-08-27:
Maximized builds produce 3 ROMs.
A full image Tobe flasbedinternally and top and bottom images to be externally flashed.

tlaurion, 2023-08-27:
> I have a T430 but there are three board names

There is:
- legacy-flash
- legacy: hotp or not (internally flashable through legacy-flash) 
- maximized : hotp or not

I think I will move legacy boards and make them disappear soon. Too much ink for no known users of them.

Use maximized ones and flash externally. Choose hotp or non-hotp depending if you own a librem key or nitro key pro2/nitrokey storage 2.

firefoxlover, 2023-08-29:
@tlaurion thanks for the response! I think simply linking the Wiki to the README would be great.

So:
- flash is the small 4mb image possible to be used with 1vyrain but not recommended
- legacy is the 7,5mb image that sits in between and is also not complete
- maximized can only be flashed by hardware, not by software

me_cleaner needs to be seperately done.

you need a flashing tool that does not use 5V and toast your chip

tlaurion, 2023-08-29:
> legacy is the 7,5mb image that sits in between and is also not complete

Just that is partly true.
Once flashing x230-flash, flash Rom is called internally to only flash the IFD defined BIOS region. ME and GBE region are untouched. 

The readme already points to https://osresearch.net, which is where heads-wiki is rendered already. 

The wiki already documents all of this. PR welcome against heads-wiki, I cannot really do better then what was already done. Each reader comes with different background. I give those explanations once in a while in the hope that those readers with their background with do PR to simplify the wiki for themselves, helping others with the same background and similar pairs of eyes.

tlaurion, 2026-09-17:
legacy not supported anymore, issue opened in 2023. Closing

closed 2026-09-17 by tlaurion