linuxboot/heads #1486: CAPSLOCK Detection

barbed-overflow, 2023-09-01:
CAPSLOCK Detection when typing in password to avoid lockout.

tlaurion, 2023-09-01:
Have not found a unified way to do that with busybox yet, thanks for the reminder.

One way, not even sure its there since those ThinkPad do not have the caplsock led which is the problem here, could be

`cat /sys/class/leds/tpacpi::kbd_backlight/max_brightness`

xset could do it, but not sure. Will have to dig down here

tlaurion, 2024-01-13:
```
#!/bin/bash

# Use setleds to query the LED status
output=$(busybox setleds -q)

# Extract the capslock status from the output
capslock=$(echo "$output" | grep -o "CapsLock.*" | cut -d " " -f 2)

# Return true if capslock is on, false otherwise
if [ "$capslock" = "on" ]; then
    exit 0
else
    exit 1
fi
```

Would depend on busybox missing features to be added in busybox.config:
```
CONFIG_SETLOGCONS=y
CONFIG_FEATURE_SETLEDS=y
```

Would address point 1 of #1588

tlaurion, 2024-01-13:
Nope. Just changes the led, not interact with console....

tlaurion, 2024-01-20:
Seems like kbd tool would permit this. Smallest poc would be c code, ioctl on try to check input modifiers.

Still ugly. Is there something from keyboard drivers exposed?

tlaurion, 2025-09-22:
> Seems like kbd tool would permit this. Smallest poc would be c code, ioctl on try to check input modifiers.
> 
> Still ugly. Is there something from keyboard drivers exposed? 

We now have and depend on kbd. Revisit this