linuxboot/heads #1497: Can´t mount USB-storage-device after update.

Oessel, 2023-09-07:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**
t430-hotp-maximized
**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [x] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [x] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [x] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [ ] Other

**5. Are you using the PGP key to provide HOTP verification?**
- [x] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ ] First-time flash
- [x] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [x] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [ ] External flashing

**3. How was Heads initially flashed**
- [x] External flashing
- [ ] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [x] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [ ] Yes
- [x] No
- [ ] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [ ] I downloaded it
- [x] I built it

**2. If you downloaded your rom, where did you get it from?**
- [ ] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*
v0.2.0-1796-g35f3b22
**3. If you built your rom, which repository:branch did you use?**
- [x] Heads:Master
- [ ] Other (please identify)

**4. What version of coreboot did you use in building?**
- [ ] 4.8.1 (current default in heads:master)
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [x] Other (4.19)
- [ ] I don't know

**5. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [x] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

## Please describe the problem

**Describe the bug**

I updated my firmware after the latest -pull, after reboot i tried to perform a factory reset which was interrupted because no USB-storage devices were found to mount anymore (to save the new generated public key at factory reset). At first i tried the USB-stick from which i flashed heads minutes before, the Nitrokey Pro was found, the error occured only with any storage device i tried 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

tlaurion, 2023-09-07:
@Oessel what is the filesystem of the usb thumb drive?

Oessel, 2023-09-07:
FAT-32, but i tried 4 thumb drives, 1 with EXT-4, nothing worked. the first one i tried is my "usual heads-update-stick". got the error reproductible on my dirty build (changed gpg-key attributes) but also on a clean build i made for testing. after rollback to 1759-g45a4f9d (also built by myself) everything works fine

tlaurion, 2023-09-07:
@Oessel That was a big error on my side. I assumed my search and replace replaced all calls to mount usb devices in read write mode were searched and replaced, but I missed one.

This is fixed in builds happening right now under master with https://github.com/osresearch/heads/commit/97bb0a82cb0ff07f341ccb1228911b748ee1281d

PR #1498 (2023-09-07): oem-factory-reset: fix call to mount-usb --mode rw 
fixes #1497
commit cba8c454 (2023-09-07): oem-factory-reset: fix call to mount-usb --mode rw (fix #1497)
closed 2023-09-07 by tlaurion