linuxboot/heads #1503: Connect with Nitrokey to officially support T430 and the NV41 and NS50

firefoxlover, 2023-09-17:
Nitrokey also puts Heads on their laptops. Its a great service and a good proof that these models work great.

I guess there is the option to connect with them and have these Laptops be certified as officially working (no support etc.).

I would also see if Nitrokey uses some different settings for their laptops.

tlaurion, 2023-09-18:
@firefoxlover 
Can you clarify your issue in OP and detail what is not working for you? 

Is it linked to nitrokey 3? 
Do you want to know if their firmware for t430 now works with nitrokey3 (NK3?)

firefoxlover, 2023-09-19:
Hi, my request was that there is a fully working T430 image that Nitro uses for their laptops.

So would there stand anything against connecting with them, and make the T430 image official? This would be between them and you. Naming is a bit annoying, I dont know if they are called Nitro or Nitrokey.

tlaurion, 2023-09-23:
I'm not sure I still understand.

If you look at nitrokey GitHub repository for Heads fork, you will see which commit they use in their releases?

firefoxlover, 2023-09-25:
I would like the T430 builds to be rephrased as "unsupported but working well", as Nitrokey uses these images in their Laptops! Their builds are older and not rolling, so the latest images may not be supported. But I meant that its kinda misleading if these builds are actually dangerous or something

nestire, 2023-09-25:
Hi

I'm working for nitrokey, we test actually quit regularly upstream builds to see if they are working. And we try to release more often images for updates  then in the past so that we don't let the gap between our production version and upstream get to big. 
So I would say special with the latest merges for the NK3 and the NV41/NS50 heads fully supports #1485 our devices and we are committed to keep it that way.

tlaurion, 2023-09-25:
> I would like the T430 builds to be rephrased as \"unsupported but working well\", as Nitrokey uses these images in their Laptops! Their builds are older and not rolling, so the latest images may not [...]

I think there is misunderstanding.

Heads is the upstream project where nitrokey owns a fork of the project. The t430 is supported, as all other boards under the board directory that are not named as UNTESTED.

Heads repository is a rolling release as of today until releases are made again. Nitrokey sells products with Heads deployed on which they offer support for the releases they made tested and support.

Forks un open source projects often contain differences when compared to the parent project. For example, pureboot from Purism contained features that were not available under Heads. Nitrokey decided for a long time to release new firmware images on their terms. Nitrokey rebranded t430 to Nitropad t430 in their repo. Easier rebranding has its specific issue that will be tackled with stakeholders in the future. For exemple, recent Heads will provide a bootsplash that shows current Heads bootsplash, where forks might want to modify this to shown their own bootsplash. Circleci builds shows the default bootsplash for all ROMs built supporting bootsplash early in coreboot ram init. 

Heads asks willing testers, called board owners, which have external programmers when important changes can brick computer.

T430 is supported. I'm still confused on what concepts are mixed in here leading to think that the t430 is not supported. The documentation is a community effort, where board owners are welcome to propose pull request to binify the documentation, for example in regard to disassembling the platform.

If the issue is clarified please close it. Otherwise please clarify which concept needs to be clarified. I repeat, Heads is the main codebase for which forks are made to sell products fixated on some commit that are considered tested and supported by the sellers. Otherwise, Heads is basically scripts that run across all boards that are evolving upstream in Heads and for which forks rebase themselves on top of it whenever they see it fit for their businesses.

firefoxlover, 2023-09-25:
The issue is clear. Nitrokey has some differences and only supports their images.

closed 2023-09-25 by firefoxlover

tlaurion, 2023-10-17:
> The issue is clear. Nitrokey has some differences and only supports their images.

@firefoxlover rhere is nothing stopping your from using heads rolling release (edge) instead of their slightly older versions (considered stable from Nitrokey).