linuxboot/heads #1504: libksba-1.6.3 prevent building boards on debian 12

123ahaha, 2023-09-23:
## Context of the Build

**1. What board are you trying to build?**

**2. What repository:branch are you using to build from?**
- [X] Heads:Master

**3. What version of coreboot are you trying to build**
- [X] Other (please specify) coreboot-nitrokey downloaded when `make BOARD=nitropad-nv41`

**4. In building the rom where did you get the blobs?**
- [X] Extracted from the online bios using the automated tools provided in Heads

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
Not relevant here, the process is automated for this the nv41 board, but I ve tried also the BOARD=x230-hotp-maximized and the BOARD=t430-hotp-maximized with the blobs/xx30/download_clean_me.sh runned before and still it didn t worked

**6. What operating system are you using**
```Linux workstation 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) x86_64 GNU/Linux```

## Please describe the problem

**Describe the bug**
I clone the github repo then do make BOARD=nitropad-nv41, but the build process is stopped by this error: 
```
2023-09-23 16:58:50+02:00 DONE libgcrypt
2023-09-23 16:58:50+02:00 CONFIG libksba
make: *** [Makefile:472: /home/user/heads/build/x86/libksba-1.6.3/.configured] Error 1
```
On every board I try to build.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a debian 12 iso 
2. install the required packages
3. clone the repo
4. ```make BOARD=nitropad-nv41``` or x230-hotp-maximized or t430-hotp-maximized or librem_14
5. See error

**Expected behavior**
It should build a .rom 

**Additional context**
I remeber there was some issue with gnat/gcc on debian 12 but 3 months ago. Since then I have been able to build some board but one day i had to rm -rf heads and reclone and since then it didn't work.
What's even weirder is that I built the rom I wanted (in a qube) under debian 12 but on another ssd with QubesOS on it in a repo I'd cloned some time ago.
I tried reclone, ```make real.clean``` and ```make BOARD=nitropad-nv41``` (and others) but it didn't work.

tlaurion, 2023-09-23:
As of today, to replicate Circleci results, one has to build on top of debian-11 which is the only supported build platform.

Other issues tackle the reasons why.

You can relaunch a clean build having installed proper dependencies replicating Circleci config with apt install call there and do V=1

`make BOARD=xyz V=1`

And post results here.
I build locally over debian-12

123ahaha, 2023-09-23:
```
checking for gpg-error-config... /home/user/heads/install/x86/bin/gpg-error-config
checking for gpgrt-config... //bin/gpgrt-config
./configure: line 14436: test: : integer expression expected
./configure: line 14439: test: : integer expression expected
checking for GPG Error - version >= 1.8... no
configure: error: libgpg-error is needed.
                See ftp://ftp.gnupg.org/gcrypt/libgpg-error/ .
make: *** [Makefile:472: /home/user/heads/build/x86/libksba-1.6.3/.configured] Error 1
```

I tried to install libgpg-error from https://gnupg.org/ftp/gcrypt/libgpg-error/ but the more recent version is 1.47 ( =< 1.8 asked)
I ve tried to install anyway but it didn't work

Also to clarify, i succeed to build heads on debian-12 when i'm in a qubes under debian-12, not using V=1 option

tlaurion, 2023-09-23:
This means the build system has a bug, where host dependencies are bleading into the build.

123ahaha, 2023-09-24:
I might have found why is it possible to build heads under qubes's debian 12. The kernel is not the actual debian 12 kernel, but its : 
```
$ uname -ar
Linux personal 6.4.8-1.qubes-fc32.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Aug 8 22:09:56 CEST 2023 x86_64 GNU/Linux
```
this difference might be enough to build heads

tlaurion, 2023-09-24:
> gpgrt-config... //bin/gpgrt-config

This is wrong and should be updated in the module. It looks for OS installed tools and not use Heads compiled tool. As discussed in other issues, host tools changes autotools and other tools that changes across OSes and versions and changes the behavior of modules compilation.

tlaurion, 2023-09-24:
A quick verification in the gnupg toolstack modules show that modules are incomplete since their upgrade to latest versions.

For example, only gpg-error-config is set to be respected as of now. See gpg-error-config at https://github.com/osresearch/heads/blame/master/modules/libassuan#L17

akunterkontrolle, 2024-02-13:
In case you encounter that error, it may help to add ```PKG_CONFIG_PATH=[PATH to the heads dir]/install/x86/lib/pkgconfig/ \``` to the libksba build-formula. It is possibly also needed for gpg(2).

tlaurion, 2025-02-13:
Irrelevant since #1661

closed 2025-02-13 by tlaurion