linuxboot/heads #1507: openssl-3.0.8 prevent building nitropad-nv41 board on debian 11

123ahaha, 2023-10-09:
## Context of the Build

**1. What board are you trying to build?**
nitropad-nv41

**2. What repository:branch are you using to build from?**
- [x] Heads:Master

**3. What version of coreboot are you trying to build**
- [x] Default on debian-11 for board nitropad-nv41 (coreboot-nitrokey)

**4. In building the rom where did you get the blobs?**
- [x] Extracted from the online bios using the automated tools provided in Heads

**6. What operating system are you using**
debian 11.8.0 :
```
$ uname -ar 
Linux debian 5.10.0-26-amd64 #1 SMP Debian 5.10.197-1 (2023-09-29) x86_64 GNU/Linux
```

## Please describe the problem
When I'm running make BOARD=nitropad-nv41 it doesn't build a rom, it output an error when building openssl

**Describe the bug**
```
2023-10-09 17:14:00-04:00 MAKE openssl
tail /home/user/heads/build/x86/log/openssl.log
-----
	apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
	apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
	apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
	apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
	apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
	apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
	apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
	apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
	apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
	apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
	apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
	apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
	apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
	apps/openssl-bin-version.o apps/openssl-bin-x509.o \
	apps/libapps.a -lssl -lcrypto -ldl -pthread 
x86_64-linux-musl-gcc: error: apps/openssl: No such file or directory
make[2]: *** [Makefile:18052: apps/openssl] Error 1
make[2]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make[1]: *** [Makefile:2607: build_programs] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make: *** [Makefile:473: /home/user/heads/build/x86/openssl-3.0.8/.build] Error 1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install a clean debian 11
2. Install the required packages (sudo apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg gawk iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract sudo libssl-dev device-tree-compiler u-boot-tools sharutils e2fsprogs parted curl unzip imagemagick)
3. make BOARD=nitropad-nv41
4. See error

**Expected behavior**
It should output a rom binary

**Additional context**
It might still be linked to the previous issue [1504](https://github.com/osresearch/heads/issues/1504)
But this time I'm under debian-11
I have tested 
```
$ make BOARD=t430-hotp-maximized 
```
and
```
$ make BOARD=x230-hotp-maximized 
```
and
```
$ make BOARD=librem_14
```
thoses work very well though and produce the expected binary. So it might be a nitrokey only issue

tlaurion, 2023-10-09:
@123ahaha will take a look tomorrow. This is weird since Circleci is doing it from debian-11 docker+dependencies under Circleci.

Can you add V=1 to your make call?

123ahaha, 2023-10-09:
thanks for the quick feedback here is the output :
```
$ make real.clean && make BOARD=nitropad-nv41 V=1

...

2023-10-09 18:10:33-04:00 MAKE openssl
make -C "/home/user/heads/build/x86/openssl-3.0.8" -j12 --max-load 16 build_programs && make -C /home/user/heads/build/x86/openssl-3.0.8 DESTDIR=/home/user/heads/install/x86 LIBDIR=lib install_sw 
make[1]: Entering directory '/home/user/heads/build/x86/openssl-3.0.8'
make depend && make _build_programs
make[2]: Entering directory '/home/user/heads/build/x86/openssl-3.0.8'
make[2]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make[2]: Entering directory '/home/user/heads/build/x86/openssl-3.0.8'
rm -f apps/openssl
${LDCMD:-/home/user/heads/crossgcc/x86/bin/x86_64-linux-musl-gcc -fdebug-prefix-map=/home/user/heads=heads -gno-record-gcc-switches -D__MUSL__ -isystem /home/user/heads/install/x86/include -L/home/user/heads/install/x86/lib } -pthread -m64 -Wa,--noexecstack -Os -L.  -L \
	-o apps/openssl \
	apps/lib/openssl-bin-cmp_mock_srv.o \
	apps/openssl-bin-asn1parse.o apps/openssl-bin-ca.o \
	apps/openssl-bin-ciphers.o apps/openssl-bin-cmp.o \
	apps/openssl-bin-cms.o apps/openssl-bin-crl.o \
	apps/openssl-bin-crl2pkcs7.o apps/openssl-bin-dgst.o \
	apps/openssl-bin-ec.o apps/openssl-bin-ecparam.o \
	apps/openssl-bin-enc.o apps/openssl-bin-engine.o \
	apps/openssl-bin-errstr.o apps/openssl-bin-fipsinstall.o \
	apps/openssl-bin-genpkey.o apps/openssl-bin-genrsa.o \
	apps/openssl-bin-info.o apps/openssl-bin-kdf.o \
	apps/openssl-bin-list.o apps/openssl-bin-mac.o \
	apps/openssl-bin-nseq.o apps/openssl-bin-ocsp.o \
	apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
	apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
	apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
	apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
	apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
	apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
	apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
	apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
	apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
	apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
	apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
	apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
	apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
	apps/openssl-bin-version.o apps/openssl-bin-x509.o \
	apps/libapps.a -lssl -lcrypto -ldl -pthread 
x86_64-linux-musl-gcc: error: apps/openssl: No such file or directory
make[2]: *** [Makefile:18052: apps/openssl] Error 1
make[2]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make[1]: *** [Makefile:2607: build_programs] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
tail /home/user/heads/build/x86/log/openssl.log
-----
	apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
	apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
	apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
	apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
	apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
	apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
	apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
	apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
	apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
	apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
	apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
	apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
	apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
	apps/openssl-bin-version.o apps/openssl-bin-x509.o \
	apps/libapps.a -lssl -lcrypto -ldl -pthread 
x86_64-linux-musl-gcc: error: apps/openssl: No such file or directory
make[2]: *** [Makefile:18052: apps/openssl] Error 1
make[2]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make[1]: *** [Makefile:2607: build_programs] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make: *** [Makefile:473: /home/user/heads/build/x86/openssl-3.0.8/.build] Error 1
```

tlaurion, 2023-10-10:
> thanks for the quick feedback here is the output :
> ```
> $ make real.clean && make BOARD=nitropad-nv41 V=1
>
> ...
>
> 2023-10-09 18:10:33-04:00 MAKE openssl
> make -C [...]

@123ahaha 

Can you wipe heads and reclone and build again from a clean commit?

123ahaha, 2023-10-10:
```
$ rm -rf heads
$ git clone https://github.com/osresearch/heads.git
$ cd heads
$ make BOARD=nitropad-nv41 V=1

...

x86_64-linux-musl-gcc: error: apps/openssl: No such file or directory
make[2]: *** [Makefile:18034: apps/openssl] Error 1
make[2]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make[1]: *** [Makefile:2589: build_programs] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
tail /home/user/heads/build/x86/log/openssl.log
-----
	apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
	apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
	apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
	apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
	apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
	apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
	apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
	apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
	apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
	apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
	apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
	apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
	apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
	apps/openssl-bin-version.o apps/openssl-bin-x509.o \
	apps/libapps.a -lssl -lcrypto -ldl -pthread 
x86_64-linux-musl-gcc: error: apps/openssl: No such file or directory
make[2]: *** [Makefile:18034: apps/openssl] Error 1
make[2]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make[1]: *** [Makefile:2589: build_programs] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make: *** [Makefile:473: /home/user/heads/build/x86/openssl-3.0.8/.build] Error 1
```

tlaurion, 2023-10-12:
> make[2]: Entering directory '/home/user/heads/build/x86/openssl-3.0.8'\nrm -f apps/openssl\n${LDCMD:-/home/user/heads/crossgcc/x86/bin/x86_64-linux-musl-gcc -fdebug-prefix-map=/home/user/heads=heads -gno-record-gcc-switches -D__MUSL__ -isystem /home/user/heads/install/x86/include -L/home/user/heads/install/x86/lib } -pthread -m64 -Wa,--noexecstack -Os -L.  -L \\\n

Really weird. Will try to replicate. 
Here it seems that it's deleted and then searched for and not found?!

tlaurion, 2023-10-16:
> Will try to replicate.

@123ahaha

Seems like libncurses5-dev is missing from explicit Circleci apt call and could result in such error. 

```
sudo apt-get update sudo apt-get install autoconf automake bc bison build-essential bzip2 ccache cmake coreboot-utils cpio curl device-tree-compiler doxygen e2fsprogs flex gawk gnat git gnupg graphviz iasl imagemagick innoextract libdigest-sha-perl libelf-dev libncurses5-dev libssl-dev libtool libudev-dev libudev1 libusb-1.0-0-dev m4 nasm ncurses-dev parted patch pkg-config python python2 python3 rsync sharutils sudo texinfo u-boot-tools udev unzip uuid-dev wget zlib1g-dev
```

Can you try this and report?
Otherwise I'll try from qubes debian-11-minimal when I have a second.

Seems like libncurses5-dev is present in qubes debian-11 template and Circleci debian-11 docker but not on debian-11 default install.

123ahaha, 2023-10-16:
I couldn't install coreboot-utils on bullseye
and it didn't worked

tlaurion, 2023-10-17:
```
2023-10-16 21:19:16-04:00 MAKE openssl
tail /home/user/heads/build/x86/log/openssl.log
-----
	apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
	apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
	apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
	apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
	apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
	apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
	apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
	apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
	apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
	apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
	apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
	apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
	apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
	apps/openssl-bin-version.o apps/openssl-bin-x509.o \
	apps/libapps.a -lssl -lcrypto -ldl -pthread 
x86_64-linux-musl-gcc: error: apps/openssl: No such file or directory
make[2]: *** [Makefile:18052: apps/openssl] Error 1
make[2]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make[1]: *** [Makefile:2607: build_programs] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make: *** [Makefile:473: /home/user/heads/build/x86/openssl-3.0.8/.build] Error 1
```

```
root@heads-tests-deb11min:~# apt install libncurses5-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  libncurses5-dev
0 upgraded, 1 newly installed, 0 to remove and 58 not upgraded.
Need to get 948 B of archives.
After this operation, 6144 B of additional disk space will be used.
Get:1 http://HTTPS///deb.debian.org/debian bullseye/main amd64 libncurses5-dev amd64 6.2+20201114-2+deb11u2 [948 B]
Fetched 948 B in 0s (2172 B/s)          
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
Selecting previously unselected package libncurses5-dev:amd64.
(Reading database ... 46080 files and directories currently installed.)
Preparing to unpack .../libncurses5-dev_6.2+20201114-2+deb11u2_amd64.deb ...
Unpacking libncurses5-dev:amd64 (6.2+20201114-2+deb11u2) ...
Setting up libncurses5-dev:amd64 (6.2+20201114-2+deb11u2) ...
```

```
2023-10-16 21:21:28-04:00 MAKE openssl
tail /home/user/heads/build/x86/log/openssl.log
-----
	apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
	apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
	apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
	apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
	apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
	apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
	apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
	apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
	apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
	apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
	apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
	apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
	apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
	apps/openssl-bin-version.o apps/openssl-bin-x509.o \
	apps/libapps.a -lssl -lcrypto -ldl -pthread 
x86_64-linux-musl-gcc: error: apps/openssl: No such file or directory
make[2]: *** [Makefile:18052: apps/openssl] Error 1
make[2]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make[1]: *** [Makefile:2607: build_programs] Error 2
make[1]: Leaving directory '/home/user/heads/build/x86/openssl-3.0.8'
make: *** [Makefile:473: /home/user/heads/build/x86/openssl-3.0.8/.build] Error 1
```

```
user@heads-tests-deb11min:~/heads$ rm -rf build/x86/openssl-3.0.8/
user@heads-tests-deb11min:~/heads$ make BOARD=nitropad-nv41
[...]
2023-10-16 21:22:14-04:00 CONFIG openssl
2023-10-16 21:22:24-04:00 MAKE openssl
2023-10-16 21:26:04-04:00 DONE openssl
2023-10-16 21:26:04-04:00 WGET https://github.com/tpm2-software/tpm2-tss/releases/download/3.2.0/tpm2-tss-3.2.0.tar.gz
--2023-10-16 21:26:04--  https://github.com/tpm2-software/tpm2-tss/releases/download/3.2.0/tpm2-tss-3.2.0.tar.gz
Resolving github.com (github.com)... 140.82.114.3
Connecting to github.com (github.com)|140.82.114.3|:443... connected.
[...]
```

@123ahaha please replicate:
- sudo apt install libncurses5-dev
- rm -rf build/x86/openssl-3.0.8/
- make BOARD=nitropad-nv41

If successfull for you too, I'll create PR.

123ahaha, 2023-10-17:
libncurses5-dev was already installed on my workstation, 
    rm -rf build/x86/openssl-3.0.8/
    make BOARD=nitropad-nv41
    
Worked for me 

```
2023-10-17 02:34:10-04:00 INSTALL   build/x86/coreboot-nitrokey/nitropad-nv41/coreboot.rom => build/x86/nitropad-nv41/heads-nitropad-nv41-v0.2.0-1809-gbd2a8eb.rom
b35c01bbf39e490176504edbe5445a7a0a927face02e3b2fa85a8a17a8ab678a  build/x86/nitropad-nv41/heads-nitropad-nv41-v0.2.0-1809-gbd2a8eb.rom
33554432:build/x86/nitropad-nv41/heads-nitropad-nv41-v0.2.0-1809-gbd2a8eb.rom
b35c01bbf39e490176504edbe5445a7a0a927face02e3b2fa85a8a17a8ab678a  /home/user/heads/build/x86/nitropad-nv41/heads-nitropad-nv41-v0.2.0-1809-gbd2a8eb.rom
33554432:/home/user/heads/build/x86/nitropad-nv41/heads-nitropad-nv41-v0.2.0-1809-gbd2a8eb.rom
```

closed 2023-10-17 by tlaurion

123ahaha, 2023-10-18:
Let me clarify. It's not the 
```
$ sudo apt install libncurses5-dev
```
That changed something, I already had libncurses5-dev, what changed is the 
``
$ rm -rf build/x86/openssl-3.0.8/
```

mkopec, 2024-02-15:
There seems to be an issue with the generated openssl makefile:

```
apps/openssl: apps/lib/openssl-bin-cmp_mock_srv.o \
              apps/openssl-bin-asn1parse.o apps/openssl-bin-ca.o \
              apps/openssl-bin-ciphers.o apps/openssl-bin-cmp.o \
              apps/openssl-bin-cms.o apps/openssl-bin-crl.o \
              apps/openssl-bin-crl2pkcs7.o apps/openssl-bin-dgst.o \
              apps/openssl-bin-ec.o apps/openssl-bin-ecparam.o \
              apps/openssl-bin-enc.o apps/openssl-bin-engine.o \
              apps/openssl-bin-errstr.o apps/openssl-bin-fipsinstall.o \
              apps/openssl-bin-genpkey.o apps/openssl-bin-genrsa.o \
              apps/openssl-bin-info.o apps/openssl-bin-kdf.o \
              apps/openssl-bin-list.o apps/openssl-bin-mac.o \
              apps/openssl-bin-nseq.o apps/openssl-bin-ocsp.o \
              apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
              apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
              apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
              apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
              apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
              apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
              apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
              apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
              apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
              apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
              apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
              apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
              apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
              apps/openssl-bin-version.o apps/openssl-bin-x509.o \
              apps/libapps.a libssl.so libcrypto.so
	rm -f apps/openssl; \
	$${LDCMD:-$(CC)} $(BIN_CFLAGS) -L. $(BIN_LDFLAGS) \
		-o apps/openssl \
		apps/lib/openssl-bin-cmp_mock_srv.o \
		apps/openssl-bin-asn1parse.o apps/openssl-bin-ca.o \
		apps/openssl-bin-ciphers.o apps/openssl-bin-cmp.o \
		apps/openssl-bin-cms.o apps/openssl-bin-crl.o \
		apps/openssl-bin-crl2pkcs7.o apps/openssl-bin-dgst.o \
		apps/openssl-bin-ec.o apps/openssl-bin-ecparam.o \
		apps/openssl-bin-enc.o apps/openssl-bin-engine.o \
		apps/openssl-bin-errstr.o apps/openssl-bin-fipsinstall.o \
		apps/openssl-bin-genpkey.o apps/openssl-bin-genrsa.o \
		apps/openssl-bin-info.o apps/openssl-bin-kdf.o \
		apps/openssl-bin-list.o apps/openssl-bin-mac.o \
		apps/openssl-bin-nseq.o apps/openssl-bin-ocsp.o \
		apps/openssl-bin-openssl.o apps/openssl-bin-passwd.o \
		apps/openssl-bin-pkcs12.o apps/openssl-bin-pkcs7.o \
		apps/openssl-bin-pkcs8.o apps/openssl-bin-pkey.o \
		apps/openssl-bin-pkeyparam.o apps/openssl-bin-pkeyutl.o \
		apps/openssl-bin-prime.o apps/openssl-bin-progs.o \
		apps/openssl-bin-rand.o apps/openssl-bin-rehash.o \
		apps/openssl-bin-req.o apps/openssl-bin-rsa.o \
		apps/openssl-bin-rsautl.o apps/openssl-bin-s_client.o \
		apps/openssl-bin-s_server.o apps/openssl-bin-s_time.o \
		apps/openssl-bin-sess_id.o apps/openssl-bin-smime.o \
		apps/openssl-bin-speed.o apps/openssl-bin-spkac.o \
		apps/openssl-bin-srp.o apps/openssl-bin-storeutl.o \
		apps/openssl-bin-ts.o apps/openssl-bin-verify.o \
		apps/openssl-bin-version.o apps/openssl-bin-x509.o \
		apps/libapps.a -lssl -lcrypto $(BIN_EX_LIBS)
```

Specifically right here:

```
	$${LDCMD:-$(CC)} $(BIN_CFLAGS) -L. $(BIN_LDFLAGS) \
		-o apps/openssl \
		apps/lib/openssl-bin-cmp_mock_srv.o \
```

BIN_LDFLAGS evaluates to just `-L` so the `-o` preceding `apps/openssl` is ignored, hence why ld can't find `apps/openssl`.

mkopec, 2024-02-15:
The culprit is here: https://github.com/linuxboot/heads/blob/5bff51952d0bf8888b7b171ed14ac052ca33f597/modules/openssl#L10-L11C31

I'll try building without this hack, I'm not sure if it's needed

tlaurion, 2024-02-15:
> The culprit is here: https://github.com/linuxboot/heads/blob/5bff51952d0bf8888b7b171ed14ac052ca33f597/modules/openssl#L10-L11C31
> 
> I'll try building without this hack, TBH I don't quite understand what it's for. @tlaurion can you take a look?
> 
> Edit: Building `nitropad-nv41` on a freshly cloned repo proceeds without errors with this hack removed.

Might be legacy of work prior of changes in global Makefile which deprecated the hack.

tlaurion, 2024-02-15:
> Edit: Building nitropad-nv41 on a freshly cloned repo proceeds without errors with this hack removed.

@mkopec if you are quicker then me, do PR with the hack removed and I'll merge (afk)

mkopec, 2024-02-16:
@tlaurion Sure, submitted a PR: https://github.com/linuxboot/heads/pull/1614

PR #1510 (2023-10-17): CircleCI: fix debian-11 packages dependencies (#1507)
Fixes #1507 by explicitly stating under CircleCI config (list of debian-11 packages explicitly needed to build Heads) that libncurses5-dev is needed so that qubes-os debian-11-minimal, debian-11 install and docker can be used with same apt packages installed to build openssl module, required for boards depending on tpm2 toolstack.
PR #1614 (2024-02-16): modules/openssl: remove libgcc path hack
Passing the path to libgcc in LDFLAGS appears to no longer be needed. Removing this fixes compilation on a freshly cloned repo.

Fixes #1507