linuxboot/heads #1523: Upstreaming eDP/FHD to coreboot 4.22 if quick enough

tlaurion, 2023-11-10:
Hey testers summarized under https://github.com/linuxboot/heads/issues/692 as: x230-fhd/edp variant: @n4ru @computer-user123 (nitro caster board) @Tonux599 @househead @pcm720 (eDP 4.0 board and 1440p display).

If I can come up with a PR for you to test, can one of you test it quick to see if it works as expected?

I would love to drop patches under Heads and the window frame is short.
That would be really appreciated, let me know today so I can do a PR just for that board to test.

pcm720, 2023-11-10:
@tlaurion Sure, I don't mind testing the PR.

tlaurion, 2023-11-10:
Alright, can't promise anything but will try to work on this when I have a coreboot commit to add.

tlaurion, 2023-11-20:
Won't happen in time....

closed 2023-11-20 by tlaurion