linuxboot/heads #1533: reboot from OS not working with Heads

toothlesslizard, 2023-11-18:
step to reproduce : 
1. boot to distro
2. sudo reboot
3. get a black screen (power button stop blinking, looks like a hibernate) and nothing happens
4. i have to hold down the power button and wait for the notebook to turn off

tlaurion, 2023-11-19:
Which board, which os which commit?

toothlesslizard, 2023-11-19:
@tlaurion 

master branch, xx30 board.

tlaurion, 2023-11-19:
> master branch, xx30 board, x230-maximized-fhd_edp\nQubes/Ubuntu both\nInside Heads recovery shell reboot works good.\n\n/home/user/heads/build/x86/x230-maximized-fhd_edp/heads-x230-maximized-fhd_edp-v0.2.0-1914-g1f39d16-dirty.rom

But not on cold boot? @toothlesslizard ?
Correct?

tlaurion, 2023-11-20:
Replicating with https://output.circle-artifacts.com/output/job/672012f5-c08e-4580-b9b2-9c58ce83baeb/artifacts/0/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1914-g1f39d16.zip

Not sure why your rom is "dirty" but I would appreciate if you flashed a clean rom from CircleCI to make sure nothing else is at play but the actual codebase related to commit 1f39d16

1. booting qubesos 4.2. Asking for poweroff from OS gui 
2. Poweroff successful.
3. Booting qubes 4.2
4. Issuing reboot.
5. Reboot successful.

Cannot replicate @toothlesslizard. What is weird is that you talk about the power led flashing, which is EC driven behavior when in S3 suspend on that model, not hibernate which is also a power off with ram content saved to swap partition. 

Maybe clarify all the above and close the issue if that is fixed for you?

closed 2023-11-22 by toothlesslizard