linuxboot/heads #1534: hashes.txt contains no hash for the 12M rom

tlaurion, 2023-11-20:
but the the hash for the bottom rom shows up twice. The same happens also for other boards like the t440p. I did not analyze this, just something I found worth mentioning.

_Originally posted by @gaspar-ilom in https://github.com/linuxboot/heads/pull/1522#issuecomment-1817985671_

tlaurion, 2023-11-20:
Will need to test on clean build.
Latest changes on Makefile to create zip file containing hash for automatic integrity validation seems to interfere with board statements to duplicate hash on build console to hashes.txt.

Last time that was fixed was under https://github.com/linuxboot/heads/commit/ba9235abcbc441e9367803ba674fb7a280c48218 prior of zip file creation for all boards but excluded ones.

tlaurion, 2023-11-20:
Clean build log console output (rebasing #1522 before force push, so new unique commit)

```
2023-11-20 09:44:20-05:00 DONE coreboot-4.19
# Use coreboot.rom, because custom output files might not be processed by cbfstool
"/home/user/heads/build/x86/coreboot-4.19/x230-maximized/cbfstool" "/home/user/heads/build/x86/coreboot-4.19/x230-maximized/coreboot.rom" print
FMAP REGION: COREBOOT
Name                           Offset     Type           Size   Comp
cbfs_master_header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           93288 none
cpu_microcode_blob.bin         0x16d80    microcode       26624 none
fallback/ramstage              0x1d5c0    stage          130643 LZMA (290168 decompressed)
config                         0x3d480    raw              3252 LZMA (10323 decompressed)
revision                       0x3e180    raw               724 none
build_info                     0x3e480    raw               101 none
bootsplash.jpg                 0x3e540    bootsplash      42115 none
fallback/dsdt.aml              0x48a00    raw             14522 none
vbt.bin                        0x4c300    raw              1433 LZMA (4281 decompressed)
cmos_layout.bin                0x4c900    cmos_layout      2012 none
fallback/postcar               0x4d140    stage           39580 none
fallback/payload               0x56c40    simple elf    7026261 none
(empty)                        0x70a300   null          4969060 none
bootblock                      0xbc7580   bootblock       34880 none
2023-11-20 09:44:20-05:00 INSTALL   build/x86/coreboot-4.19/x230-maximized/coreboot.rom => build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833.rom
e96ef0c231f8ca47677bf5cd88ab1d6db6247374a9dab63881161a041bf3184f  build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833.rom
12582912:build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833.rom
2023-11-20 09:44:20-05:00 DD 8MB build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833-bottom.rom
24dae8edbd53b600c55aa0afa43e7efa6d4c5e63685429922a5a313ca7b127aa  /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833-bottom.rom
2023-11-20 09:44:20-05:00 DD 4MB build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833-top.rom
bf7b2423bf8b7e9940dfc2fb140ebb9a6d7e0bc3a4f06bfd73063cfdc14d356f  /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833-top.rom
rm -rf "/home/user/heads/build/x86/x230-maximized/update_pkg"
mkdir -p "/home/user/heads/build/x86/x230-maximized/update_pkg"
cp "/home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833.rom" "/home/user/heads/build/x86/x230-maximized/update_pkg/"
cd "/home/user/heads/build/x86/x230-maximized/update_pkg" && sha256sum "heads-x230-maximized-v0.2.0-1922-g0e42833.rom" >sha256sum.txt
cd "/home/user/heads/build/x86/x230-maximized/update_pkg" && zip -9 "/home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833.zip" "heads-x230-maximized-v0.2.0-1922-g0e42833.rom" sha256sum.txt
  adding: heads-x230-maximized-v0.2.0-1922-g0e42833.rom (deflated 41%)
  adding: sha256sum.txt (deflated 14%)
e96ef0c231f8ca47677bf5cd88ab1d6db6247374a9dab63881161a041bf3184f  /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833.rom
12582912:/home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833.rom
```

output in hashes.txt:
```
user@heads-tests-deb12:~/heads$ cat build/x86/x230-maximized/hashes.txt 
2023-11-20 09:42:36-05:00 0e42833ada7fcf875de4cb74d7ddc0ba12f641e9 clean
0166936f9a95b8f1a2e076c896c811f72550946dcbc5d7120f8aa0f6f24fc4c2  /home/user/heads/build/x86/x230-maximized/bzImage
df96938fd5cdf061aa9e4e07171375e7fbe7948b482bc1a319a3f66c453b5173  /home/user/heads/build/x86/x230-maximized/modules.cpio
-----
b7e7dce347399c789e561a95d892a4810c239cf987512308d026e1e4191c86c4  ./lib/modules/e1000e.ko
5f821c62128cf6a594deb62d9427d3af6f8fc3e5b72b7fde0cd0476696b2a231  ./lib/modules/xhci-pci.ko
59542f937654b5fb8b46e8ba6b7567c13762a9681e6409ffbd2137fd4e0609e2  ./lib/modules/usb-storage.ko
b253c1279ee049031d86b459ca0486f46d711519fdb69125c86fbe6037078da9  ./lib/modules/xhci-hcd.ko
6e2a142a56ba12c1c73810dc542c5d021533fb1939fdbe214c3f255c892babc2  ./lib/modules/ehci-hcd.ko
3f91ea916af2cb834052aa26a2c153df75e5eea09055776dffde84cb14cf40dd  ./lib/modules/ehci-pci.ko
-----
6f9dd72170022897a258904989e4f4dc507acc2ae7edc10e847bf8c69450dd7d  /home/user/heads/build/x86/x230-maximized/tools.cpio
-----
dfd6eed7642c97a7f8a91ba570037c5fbfa70325641360e4636554c3f3d68ab4  ./etc/config
b6abb877a5342eed28b6cf212dab4a547d3c034136fed2421b7d465ac96cc082  ./lib/libksba.so.8
198c74e0fece1f69c96a02bb81ff19c0159bd997c201b30cd67018b6094d4e33  ./lib/libjson-c.so.5
5cee77d4ae3ee96e6a6b6856302449ffad4ff6691414946b53f71fa614e93345  ./lib/libtpm.so
1f77815f340309c1922ce3fcc5769cf05d7283f13a0f9fc268b5199403072455  ./lib/libnpth.so.0
245e115f7ee7f1c2d3bb2a9107e2c8c2b26e3bb3b0de7464c271cdb5d7db22c9  ./lib/libpng16.so.16
1e7cf63adb4c7344d817e9ab5451296593a4ae89a2e3523548347ba0853fc347  ./lib/libassuan.so.0
c8e1da3448591c64c5a924180673e66736ba35a054fcc34e0fd66efa48772c88  ./lib/libgpg-error.so.0
de7490d19736e852e084506ecf273547208b4adcbf62dd8d3665272a218598a1  ./lib/libusb-1.0.so.0
0926dde50a4ded3286f445ed515f3c9e27771022b153fe63cb8c3be597cd0391  ./lib/libcryptsetup.so.12
[... cut ...]
24dae8edbd53b600c55aa0afa43e7efa6d4c5e63685429922a5a313ca7b127aa  /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833-bottom.rom
bf7b2423bf8b7e9940dfc2fb140ebb9a6d7e0bc3a4f06bfd73063cfdc14d356f  /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833-top.rom
e96ef0c231f8ca47677bf5cd88ab1d6db6247374a9dab63881161a041bf3184f  /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-1922-g0e42833.rom
```

@gaspar-ilom is this because you have rebuilt a same commit maybe?

gaspar-ilom, 2023-11-20:
> @gaspar-ilom is this because you have rebuilt a same commit maybe? This is Makefile driven so if 12mb rom exists, its not recreated and therefore hash is not outputted on screen nor on hashes.txt file.

Actually, I did not even build the rom myself but just used the build from circleci. I probably should have mentioned that. To me the hashes.txt in the circleci build still looks the same after the rebase: https://app.circleci.com/pipelines/github/tlaurion/heads/2120/workflows/e15ee75d-f62a-4c1b-9e02-aad2ef20efbd/jobs/36345

tlaurion, 2023-11-21:
I see.

- hashes files but not for the full rom: https://app.circleci.com/pipelines/github/tlaurion/heads/2120/workflows/e15ee75d-f62a-4c1b-9e02-aad2ef20efbd/jobs/36345/parallel-runs/0/steps/0-104

- hashes for full rom in build log : https://app.circleci.com/pipelines/github/tlaurion/heads/2120/workflows/e15ee75d-f62a-4c1b-9e02-aad2ef20efbd/jobs/36345/parallel-runs/0/steps/0-103

I'll check Makefile to understand why output of full rom is under build log but not under hashes.txt/sizes.txt

This is a bug and is reproducible. This bug reappeared between commit above which splits the ROMs for dual spi boards for external flashing and now.

closed 2024-11-05 by tlaurion