linuxboot/heads #1537: Summary window of all secrets after OEM factory reset doesn't display long passphrases in full

UndeadDevel, 2023-11-25:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

Nitropad NV41 with iGPU and Nitrokey 3A Mini using HOTP.

### B. Identify how the board was flashed

Flashed at OEM with Heads 2.1, not updated so far.

### C. Identify the rom related to this bug report

Nitrokey Heads 2.1 maximized.

## Please describe the problem

**Describe the bug**
When doing the OEM factory reset on Heads 2.1 there will be a window that summarizes and displays all the secrets (PINs and passphrases) that were chosen. However, when a very long passphrase (about 60 characters or more) is used, it does not seem to be able to change in size or break the passphrase across multiple lines, but instead the passphrase is shown incompletely (cut off at the window's right side).

**To Reproduce**
Steps to reproduce the behavior:
Do OEM factory reset and choose a very long disk recovery passphrase.

**Expected behavior**
The window should either "grow" to fit the passphrase or break the passphrase across multiple lines.

tlaurion, 2023-11-26:
@UndeadDevel : We could use busybox's "fold" the echoing of the passphrase.

But more then 6-7 words diceware passphrase is considered overkill here (hard to remember, will be forgot and will mostly be lost or written on a paper, that will also be lost) and not be practical at all.

I could still "fold" the passphrases on that screen. @UndeadDevel would be good enough?

UndeadDevel, 2023-11-26:
@tlaurion Sure, sounds good!

Feel free to close.

tlaurion, 2023-11-26:
> @tlaurion Sure, sounds good!
> 
> Feel free to close.

Sorry. Do you mean opsec needs to be adapted or code with fold?

PR #1543 (2023-12-07): Use a wider window to show the secrets
This partially fixes #1537, but while the increased width wouldn't be a problem on the NV41 AFAICT, I don't know about other machines.

I don't know what @tlaurion means with "busybox's folding", which may be a better solution.
commit 85e6f604 (2023-12-07): Update oem-factory-reset - wider window to show secrets

This partially fixes #1537, but while the increased width wouldn't be a problem on the NV41 AFAICT, I don't know about other machines.

I don't know what @tlaurion means with "busybox's folding", which may be a better solution.