linuxboot/heads #1539: Choose stronger encryption by default and/or re-use encryption parameters of LUKS container

UndeadDevel, 2023-11-25:
**Is your feature request related to a problem? Please describe.**
1. Currently re-encrypting the LUKS container from Heads seems to choose default parameters, which are no longer recommended (argon2i with 1 GB memory cost). E.g. the [IRTF recommendation](https://www.rfc-editor.org/rfc/rfc9106.html) states that argon2id with 6 GB of memory cost and parallelity of 4 is recommended for disk encryption; 6 GB might be problematic due to dom0 having only 4 GB of RAM by default, so it would be difficult to change the passphrase from dom0 later in that case (but maybe the user shouldn't do that anyway), but 2 GB would definitely work, or even 3 or perhaps 4, as there is typically 4 GB of swap in dom0 also. 2 GB is recommended in that RFC for general use cases. Other OSs Heads runs on shouldn't have a problem even with 6GB, but see below under "Additional context".
2. Another problem with the current, weaker defaults is the theoretical possibility of compromising the container using an old header or copy of the current volume encryption key encrypted with weaker parameters, even after the user has changed to stronger parameters later in dom0, if Heads is encrypting containers on COW media, such as SSDs, which are widely used these days and even recommended by QubesOS. See e.g. the [cryptseptup FAQ](https://gitlab.com/cryptsetup/cryptsetup/-/wikis/FrequentlyAskedQuestions#5-security-aspects) point 5.19 and others.
3. There's also an [open issue](https://gitlab.com/cryptsetup/cryptsetup/-/issues/785) in the cryptsetup gitlab that is related: does Heads *first* create the new key (for the keyslots, not the volume key) and *only then* re-encrypt (which always creates a new volume key first)? This is the order that is recommended, because otherwise the chance of the old keyslots lying around is greater and the new volume key would first be encrypted with weaker settings; when it is later re-encrypted in the header upon changing the params and keyslots, on COW media, such as SSDs, the old, more weakly encrypted copy of the new volume key may linger.

**Describe the solution you'd like**
I think probably the best solution would be to read out the parameters currently used for that container (`luksDump`) and then either re-use them or prompt the user, telling him the current parameters and whether they are outdated (e.g. pbkdf2 instead of argon2 being used; memory cost too low, e.g. <2 GB if argon2 is used; argon2 being used in weaker variants, i.e. not argon2id). Maybe the user could also be prompted for the iter-time (`-i=milliseconds` parameter; default is `2000`), so that the users can choose their own security / convenience tradeoff, especially since the recommended 6 GB memory are currently not possible with cryptsetup (see "Additional context").

**Describe alternatives you've considered**
An alternative that is still better than the current situation would be to use stronger defaults, at mimimum argon2id with 2 GB memory cost and (and iter time of >=2000, but 2000 is already the default with LUKS2).

**Additional context**
Do note that currently cryptsetup seems to only support a memory cost of up to 4 GB and parallelity of 4. It will also treat those two parameters (`--pbkdf-memory=numBytes` and `--pbkdf-parallel=numThreads`) as a maximum; a parallelity of 4 is used as default by cryptsetup; should the system for some weird reason have e.g. less than 4 GB available while 4 GB are given as the parameter or fewer than 4 cores, then cryptsetup will lower the actual parameters used.

tlaurion, 2023-11-25:
@UndeadDevel first things first, thanks for the good challenge on implementation.

I just want to make sure we understand what is done today.

Two cryptsetup modules exist under heads. Old modules/cryptsetup being 1.75 and newer modules/cryptsetup2 currently being 2.4.0

All boards configs are including modules/cryptsetup2 as of today.

Now, the reencrypting functions we are talking about have limited arguments forced by Heads today: https://github.com/linuxboot/heads/blob/master/initrd/etc/luks-functions#L350 

As pointed at line 350 :

``` 
cryptsetup-reencrypt -B 64 --use-directio "$LUKS" --key-slot 0 --key-file /tmp/luks_current_Disk_Recovery_Key_passphrase
```

This choices were made based on a tradeoff back then where assumptions made by cryptsetup were wrong and where ETA were otherwise exploding. 

Note that the passphrase stored under /tmp/ could be moved to /tmp/secrets to be wiped on recovery shell access even more securely by default codepath or on kexec path as well, but the assumption here again is that tmp is ramfs of initrd and recreated on reboot or overwritten on final os boot in all case, where the owner of the computer is supposed to be behind the computer and alone when doing re-ownership wizard. 

Slot 0 is forced here because the goal is to not add another key, but make sure that the installation key is overwritten in the best way possible. As you pointed out with 5.19 documentation of SSD drives from cryptsetup, there is not much more that can be done but overwrite it and leave physical forensic possible. Using SSD firmware secure erase was debated in other issues and under cryptsetup for years, and SSD doesn't offer better option then using encryption for the blocks. As also present on cryptsetup doc, entropy of passphrase is the important part here. So Heads again does the right thing here if a user decides to add a sealed TPM Disk Unlock Key, which passphrase is then a 128 random characters, sealed into TPM with firmware measurements, luks header measurements and runtime (loaded kernel modules, no recovery shell access) to be possible to be unsealed with a user selected passphrase which can be of lower entropy. That generated passphrase is then unsealed and injected in kexec reconstructed final initrd which is passed to final Os and used to decrypt LUKS container without the user having to type high entropy passphrase. 

Argon is reused if OS created LUKS container with it in version 2.4.0. Heads doesn't interfere with distribution choice as of today. Note, though, that this might need to be revisited (and cryptsetup2 module version bumped) if new platforms not having CPU AES extension are to be added under Heads. If today, OS decides to create LUKS container with Adiantum, then Heads will fail reencrypting it simply because neither Linux kernel nor cryptsetup 2.4.0 supports it. This impacts other architectures then x86/power and X86 i3 older generation CPU and was also discussed in other issues. 

Tldr: cryptsetup-reencryption under heads does not replace OS choices nor user selected settings as of today. Heads attempts to replace in place slot0 to mitigate OEM deployment bad practices and/or limit liability to the moment user receives a laptop and decides to stop trusting the OEM deployed keys from the moment of re-ownership. When a user installs an OS with LUKS settings that might not be supported by current version of cryptsetup deployed under Heads, cryptsetup will bail. 

Todo: there is still one issue you tackled under another issue under nitrokey repo, that is, those luks functions should not create any file under /boot. This is still a bug because not all boards permit TPM sealed Disk Unlock Key. This will need refactoring. 

Note: new LUKS functions have been deployed recently to create LUKS container on USB thumb drive to store gpg key material backup. Those functions hardcode the best practices borrowed from qubesos LUKS container creation. As pointed by cryptsetup documentation, that LUKS container would not be really secure if the passphrase is left to the OEM factory defaults, which refuses GPG Admin PIN which by default is 12345678.
Therefore the same issues are present here then with a "PleaseChangeMe" original passphrase. The OEM factory Reset tries to give power back to the user, but to do so, the user still needs to understand that accepting defaults might not respond to his threat model and therefore, he might want to invest into going the advanced way, which now surveys him into doing the right thing and setup individual passphrases for each secrets.... Hopefully.

UndeadDevel, 2023-11-25:
@tlaurion Thanks for the detailed reply!

> Now, the reencrypting functions we are talking about have limited arguments forced by Heads today.
This choices were made based on a tradeoff back then where assumptions made by cryptsetup were wrong and where ETA were otherwise exploding.

Regarding that line: I don't see a `-B` option in the man pages...if this is block size then that's not applicable for LUKS2 volumes as it was a LUKS1 option. Otherwise it seems that Heads lets the (old) cryptsetup version choose the parameters, which explains why it defaults to argon2i; note that the current version and all prior versions for the last 2 years use argon2id as the default (see, e.g. [here](https://gitlab.com/cryptsetup/cryptsetup/-/commit/db775417909db0f0b07168d07fdf8813e3ca94fe)). I'll also quote that RFC again in their final recommendations:
> The Argon2id variant with t=1 and 2 GiB memory is the FIRST RECOMMENDED option and is suggested as a default setting for all environments. This setting is secure against side-channel attacks and maximizes adversarial costs on dedicated brute-force hardware. The Argon2id variant with t=3 and 64 MiB memory is the SECOND RECOMMENDED option and is suggested as a default setting for memory-constrained environments. 

"t=1" refers to iterations, which cryptsetup binds to 4 as minimum by default, which is even better and exceeds the recommendations; not that it's overkill since, as I've stated, the specific RFC recommendation for disk encryption (different from above general recommendation) is 6 GB memory cost and increasing iterations does *to some degree* allow compensating for lower memory (though if possible, the memory recommendation should be fulfilled *before* increasing iterations, which the RFC also states elsewhere).

So to have argon2id used by Heads either the cryptsetup version needs to be bumped to one from the last 2 years or the pbkdf needs to be supplied with an additional parameter (`--pbkdf=argon2id`). A version bump also makes sense due to CVE-2021-4122 (see cryptsetup FAQ point 5.23), which is fixed in cryptsetup 2.4.3; that version also makes argon2id the default I think.

> As you pointed out with 5.19 documentation of SSD drives from cryptsetup, there is not much more that can be done but overwrite it and leave physical forensic possible with soldering implied.

Honestly I think that this is still not a good approach at all...it would be much better if at least some random, higher entropy password (e.g. 7 words from word list or ~20 random characters) was generated for each user separately by OEM and delivered by mail (physical with notebook / PC or email), but that's a problem for OEM, not Heads itself, per se.
I agree that the feature to have a TPM disk unlock key is a great addition.

> Heads doesn't interfere with OS distribution choices as of today.
[...]
cryptsetup-reencryption under heads does not replace OS choices nor user selected settings as of today

Maybe I'm misunderstanding what you're saying, but Heads does use weaker settings than what the user has set in the OS in some cases; e.g. I had put a new key into slot 0 from dom0 with argon2id and 2 GB memory cost and then rebooted and done Heads LUKS re-encryption. Here Heads created a new key for slot 0 with argon2i (much weaker) and 1 GB instead of 2 GB (weaker) instead of reusing the parameters of the existing container; if cryptsetup were bumped to a more current version (>=2.4.3) then it would have chosen argon2id, at least.
I do agree that if using a high entropy passphrase, either with TPM DUK or simply a good disk recovery passphrase, then the existing config should be good enough. However, I think the recommendations should nevertheless be approached, if possible (i.e. use argon2id and higher memory cost).

> Todo: there is still one issue you tackled under another issue under nitrokey repo, that is, those luks functions should not create any file under /boot. This is still a bug because not all boards permit TPM sealed Disk Unlock Key. This will need refactoring and deserve a seperate issue.
The codebase deals with that "encrypted disk presence file" as if TPM DUK was enforced, which is not true in all cases.

This is what #1536 is about; I simply described it from my user experience; I didn't check if it was due to any files created, though that sounds plausible.

> Note: new LUKS functions have been deployed recently to create LUKS container on USB thumb drive to store gpg key material backup, enabling+forcing authentication for recovery shell and USB boot.

Awesome feature btw., I'm looking forward to using it!

tlaurion, 2023-11-26:
Not time for a PR yet, but testing things under testing branch https://github.com/tlaurion/heads/tree/cryptsetup_version_bump-reencryption_cleanup and trying to document in commits done there.

tlaurion, 2023-11-26:
@UndeadDevel so confirmed that on master, reencrypting passes to argon2i
[image]

tlaurion, 2023-11-26:
Houla @UndeadDevel 

Speeds are way slower under cryptsetup 2.6.1, maybe having to do with new defaults and/or directio
[image]

This is based on https://github.com/tlaurion/heads/commit/61636a5d45249dda4b3afe28c81f7e8fd9bf845d

tlaurion, 2023-11-26:
Same result removing --direct-io on Samsung 860 1TB drive I had lying around with https://github.com/tlaurion/heads/commit/9877d46961f731d64daa31035a3103b6f93ea2cc

[image]

tlaurion, 2023-11-26:
This issue is tricky.

Let me explain. 
- Hardware limitations should impact ciphers decision. Eg: gen3 i3 don't support aes acceleration by CPU and Adendium should be used instead. 
- OS **should** do assessment of what hardware support, align LUKS container and sector size as discussed on QubesOS forum https://forum.qubes-os.org/t/ssd-maximal-performance-native-sector-size-partition-alignment/10189
  - This is complicated.
- So in the reencryption here, since alignment are corrected if they were invalid, reencryption might take longer and create disapointment. But there is not much we can do here then force proper settings?
  -  Should Heads support Adendium, assess with benchmark first and then take decision for the user and maybe break support of the final OS being able to decrypt LUKS from its initrd?

@UndeadDevel those are the thoughts i'm having letting the reencryption of the two devices complete and attempt to reencrypt the first one without --direct-io, and the second one with direct-io. Otherwise, we need to acknowledge that newer versions of cryptsetup are taking radical steps enforcing better defaults that should be applied at OS installation (Newer OSes will follow) but as of now, second laptop (1TB drive) was a test install of Qubes 4.2. 

If Qubes 4.2 has not revised their installation defaults/sector alignments/disk geometry alignements and Heads is having more recent version then OS deploys, we might be better then for our own goods :/

So not sure what to do here but going back to the old thread I opened on QubesOS and try to talk with the experts.
Maybe you can jump into the discussion there as well?

tlaurion, 2023-11-26:
So reencrypting Q4.1 installation resulted into a LUKS container that was still PDBF argon2i. I guess that is good.

But that does not explain speed of Q4.2 reencryption speeds. Stopped reencryption. Will reinstall Q4.2 over MX500 drive I have extended comparative info from the past and stick to that for testing.

I guess --direct-io is still needed. Will re-add that.

tlaurion, 2023-11-26:
Crooslinking https://forum.qubes-os.org/t/ssd-maximal-performance-native-sector-size-partition-alignment/10189/54

UndeadDevel, 2023-11-26:
Regarding the command options: all I know is that I can't find the `-B` parameter in the man pages, so I'm just wondering what it even does; it looks like it might be about block size, but the man page states that his is only for LUKS1. As for `--use-directio` this is also a LUKS1 option only, so it will be ignored when dealing with a LUKS2 volume (which is what happens if creating a new LUKS container). Therefore leaving out those two parameters is not expected to change anything when dealing with LUKS2 volumes, as they will simply be ignored (assuming that man pages are correct). No OEM should ship a LUKS1 volume these days.

Regarding (re-)encryption speed: I don't know why it's that much slower with the new version in your tests, but having the option to re-encrypt still makes sense even if slow, e.g. when receiving the device from OEM and not having a good, secure system to create the installation media on: it would be preferable to have a known good install in that case. Furthermore, the main use case for re-encryption seems to be after receiving the machine from OEM; it's not necessary AFAIU to do this for every factory reset, so I don't think it's a huge issue if it is slower; at least for me personally I much prefer it to be slower than less secure. Bumping the cryptsetup version also has other benefits as we've already discussed here.

Also I'm a bit puzzled, as on my SSD (Samsung 980 Pro 1 TB) it took less than 15 minutes to re-encrypt (with current Heads)...presumably the LUKS container grows as I put more data on it due to the thin provisioning of next layer (LVM)? Even in that case, speed for me would have been > 200 MB/s (I didn't pay attention to it then and I can't re-test right now; late next week I can test re-encryption again).
I've found [this](https://github.com/cloudflare/linux/issues/1#issuecomment-763085915) regarding measures in the Linux kernel, which are supposed to speed up LUKS encryption and are apparently included since 5.10.9.

Lastly, a note on the warning message informing the user of the re-encryption starting (nitpick): 
1. Since it's a warning it should actually warn the user of something, e.g. not to do a hard reboot; if it's not a warning, then it shouldn't state "Warning". This becomes relevant with:
2. "current passphrase" is a slight semantic miss (it's possible to take that to mean the old passphrase, especially when combined with the "Warning" statement, as the process engaged in by the user involves changing the passphrase, so proper, non-ambiguous terminology would be, e.g., "new" and "old"); also, the object is incorrectly phrased: "Recovery Disk Key passphrase" suggests that there is a "Recovery Disk", which requires a key and the passphrase is to protect that key.

IMHO the proper phrasing would be "new Disk Recovery Key passphrase" instead of "current Recovery Disk Key passphrase".

tlaurion, 2023-11-27:
@UndeadDevel similar comments from https://forum.qubes-os.org/t/ssd-maximal-performance-native-sector-size-partition-alignment/10189/55

As per latest tests:
> `cryptsetup reencrypt --force-offline-reencrypt --disable-locks "$LUKS" --debug --key-slot 0 --key-file /tmp/luks_current_Disk_Recovery_Key_passphrase`

Results in reencryption still in online mode of 31 MiB/s. Will open issue on cryptsetup gitlab soon with proper output log. This is annoying.

tlaurion, 2023-11-27:
> 've found [this](https://github.com/cloudflare/linux/issues/1#issuecomment-763085915) regarding measures in the Linux kernel, which are supposed to speed up LUKS encryption and are apparently [...]

I will look up on those in next steps. Thanks!

tlaurion, 2023-11-27:
>I've found this regarding measures in the Linux kernel, which are supposed to speed up LUKS encryption and are apparently included since 5.10.9

Will dig that down and might explain things since kernel on tested laptop (as for most others outside of librems and nv41/ns50) are based on 5.10.5

tlaurion, 2023-11-27:
As for -B or --block-size=MiB, or -b or --size=SECTORS, those are still present under cryptsetup 2.6.1.

Recommendations are to check smatctl -i output which goes striahgt back about how to deal with the problem as reported to qubes forum

[user@dom0 ~]$ sudo smartctl -i /dev/sda
```
smartctl 7.2 2021-01-17 r5171 [x86_64-linux-6.1.43-1.qubes.fc32.x86_64] (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Crucial/Micron Client SSDs
Device Model:     CT500MX500SSD1
Serial Number:    1904E1E577CA
LU WWN Device Id: 5 00a075 1e1e577ca
Firmware Version: M3CR023
User Capacity:    500,107,862,016 bytes [500 GB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    Solid State Device
Form Factor:      2.5 inches
TRIM Command:     Available
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-3 T13/2161-D revision 5
SATA Version is:  SATA 3.3, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Mon Nov 27 00:26:09 2023 EST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
```

So we have physical sector size of 4096 where logical is 512 and where 2.6.1 is by default using 512 again.
I'm getting lost and tired of this rabbit hole. Will reground and try to fiture out how to question this correctly to cryptsetup devels.

UndeadDevel, 2023-11-27:
That command doesn't even show me sector sizes on my nvme, but `sudo fdisk -l` (from dom0) says:
```
Disk model: Samsung SSD 980 PRO 1TB                 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
```
Might help explain why I've had much better performance...

tlaurion, 2023-11-27:
> Regarding the command options: all I know is that I can't find the `-B` parameter in the man pages, so I'm just wondering what it even does; it looks like it might be about block size, but the man page states that this is only for LUKS1. As for `--use-directio` this is also a LUKS1 option only, so it will be ignored when dealing with a LUKS2 volume (which is what happens if creating a new LUKS container). Therefore leaving out those two parameters is not expected to change anything when dealing with LUKS2 volumes, as they will simply be ignored (assuming that man pages are correct). No OEM should ship a LUKS1 volume these days.

Absolutely right there. The only parameters relevant to 2.6.1 are --sector-size=bytes, where 2.7.0 will reintegrate equivalent of -B --block-size=MiB through --data-unit-size=bytes. --use-directio also only applies to luks1.

When looking at debug output and disk activity, chunks rewritten to disk are way slower to write to disk than before. This makes me think that something is wrong with configured options. Some internals changed, including sse aes support which is now optional. cryptsetup guidelines say that cpu should not impact much since normally reencrypting is tugging on IO more then CPU, but this is clearly not the case here, at least on 3rd gen cpu like for the x230.

Enabling all kernel options for benchmarking cryptsetup as well as adding sse acceleration for aes on argon internal library. Note once again that cryptsetup is compiled to use kernel crypto backend, that internal argon library is used instead of external one (for firmware size footprint) and that libgcrypt is considered slower then the kernel if cpu acceleration is enabled. Also note that cryptsetup uses /dev/urandom here (unless /dev/random specified) where again hw crytpo backed acceleration is activated in kernel config for a while now.

Testing new install of Q4.2rc5 and redoing with latest roms with configure option and all interesting crypto in kernel.
> 
> Regarding (re-)encryption speed: I don't know why it's that much slower with the new version in your tests, but having the option to re-encrypt still makes sense even if slow, e.g. when receiving the device from OEM and not having a good, secure system to create the installation media on: it would be preferable to have a known good install in that case. Furthermore, the main use case for re-encryption seems to be after receiving the machine from OEM; it's not necessary AFAIU to do this for every factory reset, so I don't think it's a huge issue if it is slower; at least for me personally I much prefer it to be slower than less secure. Bumping the cryptsetup version also has other benefits as we've already discussed here.
> 
> Also I'm a bit puzzled, as on my SSD (Samsung 980 Pro 1 TB) it took less than 15 minutes to re-encrypt (with Nitrokey Heads 2.1)...presumably the LUKS container grows as I put more data on it due to the thin provisioning of next layer (LVM)? Even in that case, speed for me would have been > 200 MB/s (I didn't pay attention to it then and I can't re-test right now; late next week I can test re-encryption again). I've found [this](https://github.com/cloudflare/linux/issues/1#issuecomment-763085915) regarding measures in the Linux kernel, which are supposed to speed up LUKS encryption and are apparently included since 5.10.9.
> 
> Lastly, a note on the warning message informing the user of the re-encryption starting (nitpick):
> 
>     1. Since it's a warning it should actually warn the user of something, e.g. not to do a hard reboot; if it's not a warning, then it shouldn't state "Warning". This becomes relevant with:
> 
>     2. "current passphrase" is a slight semantic miss (it's possible to take that to mean the old passphrase, especially when combined with the "Warning" statement, as the process engaged in by the user involves changing the passphrase, so proper, non-ambiguous terminology would be, e.g., "new" and "old"); also, the object is incorrectly phrased: "Recovery Disk Key passphrase" suggests that there is a "Recovery Disk", which requires a key and the passphrase is to protect that key.
> 
> 
> IMHO the proper phrasing would be "new Disk Recovery Key passphrase" instead of "current Recovery Disk Key passphrase".

tlaurion, 2023-11-28:
@UndeadDevel Attempts to reencryption speed and discussions on that specific issue under #1541

tlaurion, 2023-11-29:
Fell on https://github.com/cloudflare/linux/issues/6 pointing to https://github.com/torvalds/linux/commit/39d42fa96ba1b7d2544db3f8ed5da8fb0d5cb877 last night. Resulting testing works way faster and stabler.

tlaurion, 2024-04-07:
It was decided that Heads should not interfere with OS choices, where pushing for better defaults should unfortunately be done upstream in the OS installer hardcoded options.

If for whatever reason, you believe Heads should force better encryption options down at Re-Ownership reencryption time, please reopen.

Note though that some versions of old hardware don't support hardware accelerated encryption and this is why this choice is not currently enforced. One such example is the x230i. Therefore, detection of hardware specs and hw acceleration features should be done under OSes. At some point I feel I have to draw a line into maintaining things that might break down not properly thought impacts of such improvement requests. I think OSes are doing a proper job at following best practices generally in that matter, where Heads should focus on permitting re-ownership flawlessly, and that enough is becoming complicated.

UndeadDevel, 2024-04-09:
@tlaurion The QubesOS 4.2 installer does result in the PBKDF argon2id to be used, however; the problem is that Heads, when doing a container re-encryption, re-encrypts with argon2i due to using an outdated cryptsetup version, though I see that you've added commits to the PR, which I haven't reviewed yet (see my comment there).

I understand that Heads maintainers won't necessarily have the resources to be better than OSs like QubesOS (where I've contributed [this comment](https://github.com/QubesOS/qubes-issues/issues/8600#issuecomment-1823571562) to argue for improved defaults), but I do think that it's a necessity for Heads to provide a more current cryptsetup version to not use outdated defaults of an older version (which, again, may be fixed in the PR I haven't reviewed yet).

-------------------

It might also make sense to provide the user the opportunity to choose the memory cost and iteration time; i.e. after the user chooses re-encryption in the whiptail menu or during factory reset to a `read` for memory cost (but force a minimum of 1 GB; this would be fed into the cryptsetup parameter `--pbkdf-memory` in kilobytes) as well as a `read` for the iteration time (force minimum of 2 seconds; this would be fed into the cryptsetup parameter `-i` in milliseconds). That way the user can increase the encryption strength if desired, which would also somewhat increase Heads encryption robustness in case cryptsetup bumps the defaults for those parameters in a future version, where Heads would be under less pressure to upgrade in that case.

If you agree with this I could see about providing a PR to implement it. This, together with a cryptsetup version bump in Heads, and considering that point 3 of the OP was implemented already in #1542, would indeed fix this issue in a sort of second alternative solution.

tlaurion, 2024-04-09:
Thanks @UndeadDevel .

Version bump to 2.6.1 occurs in PR as of now.
As for changing settings in Metadata with `--persistent`, I'm still unsure.

UndeadDevel, 2024-04-18:
Okay great, @tlaurion, so that (cryptsetup version bump) fixes the most pressing issue!
I'll post your relevant comment portion from the PR thread here to discuss further the encryption parameters:
> @UndeadDevel the code under kexec-seal-key is adding a luks key. Therefore, we could enforce better defaults there, if desired. Let's see what is being done by default from cryptetup 2.6.1 here. As stated before, I would prefer staying away and relaying on what cryptsetup enforces as the best defaults there. When comes to DUK, lets remember that the DUK "passphrase" here, more a key itself really, is a 128 chars randomly generated (with now HW backed randomization if available) and therefore I strill believe this to be irrelevant since that keyslot cannot even be filled with weak "passphrase". Nobody will type 128 chars here and no collision is theoritically possible, even?

Okay, so I totally agree about not bumping up to 2.7.1.

Regarding `--persistent`: I don't think any metadata stored with `--persistent` would be an issue as that affects affects how the volume in mounted regarding configuration normally stored in crypttab. Right now the two slots will almost certainly have different memory and time costs anyway and there's no issue; they can even have different PBKDFs (I tested this).

The thing is, the official recommendations of the people who created Argon2 as well as the IRTF recommendations differ from the personal opinion of the lead cryptsetup maintainer:
According to the former, first parallelity should be increased as much as possible, then memory and then iterations, all the while keeping the time cost acceptable to the user.
What cryptsetup actually does, however, is:
1. use as many threads as cores, up to 4 (this is okay, but not great...CPUs with more than 4 cores are becoming more and more common, so hopefully a future cryptsetup version allows more than 4)
2. memory: here cryptsetup will behave differently depending on whether `--pbkdf-memory` was given as a parameter:
   - if yes, then use up to the provided value as long as at least 4 iterations are done and the time cost (iteration time) is not exceeded; the maximum possible memory cost is 4 GB (not great at all, as the Argon2 authors recommended 6 GB already 7 years ago for disk encryption use cases (!), but changing this requires a substantial rewrite of certain cryptsetup code IIUC)
   - if no, then use up to 1 GB only, again as long as at least 4 iterations are done and the time cost (iteration time) is not exceeded

So cryptsetup deviates from the recommendations of the people who probably know better by prioritizing iterations over memory, which should not be done for Argon2, especially argon2id and argon2d (even for argon2i it's not recommended, but less bad; overall this doesn't mean we should switch to argon2i, however, as argon2id is still more robust).

Also, the hash function SHA512 is recommended by yet other experts for better quantum resistance for `aes-xts-plain64` ciphers (used in LUKS2). I'm not sure that cryptsetup even supports that, but it's certainly not the default (which is SHA256).

So really the problem is with the cryptsetup people, who stubbornly oppose the official recommendations. Ideally they would wise up and change their defaults + implement the possibility of more than 4 threads and more than 4 GB (+SHA512 as the hash function if not already supported).

I can understand if you want to say "upstream issue, not going to touch it", but IMHO it would be nice if the IMO most severe limitation of current cryptsetup defaults (memory cost capped at 1 GB in default benchmark mode) would be (partially) mitigated by allowing the user to choose the memory cost, which we would feed into the `--pbkdf-memory` parameter to at least "unlock" the current 4 GB maximum for the memory cost.

And yes, I agree that this isn't really an issue for the TPM DUK (here the question is rather how securely it's stored in the TPM), but it _is_ an issue for the LUKS DRK, which Heads also changes during OEM Factory Reset. The user will not be remembering 256 random characters for the LUKS DRK, after all.

tlaurion, 2024-08-17:
@UndeadDevel Everythig needed under #1541 ?

closed 2024-10-30 by tlaurion