linuxboot/heads #1544: x230t - maximazied (v0.2.0-1916-gf2f0831) - can't find the boot partion

madbehaviorus, 2023-12-03:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

x230t (maximized)

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [x] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [x] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [x] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [ ] Other

**5. Are you using the PGP key to provide HOTP verification?**
- [ ] Yes
- [x ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ x] First-time flash
- [ ] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [x] External flashing

**3. How was Heads initially flashed**
- [x] External flashing
- [ ] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [x] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [] Yes
- [x] No - used "stock" builded libreboot before
- [] Don't know 

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [x] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [x] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*

## Please describe the problem

**Describe the bug**
I've tested the following image for the x230t.
heads-x230-maximized-v0.2.0-1916-gf2f0831.rom

I can configure the TPM but it is not possible, to set or change the boot path for my actually os  in the heads configuration menu.
Every time I try it, it cannot find any OS.
I tested it with an usb-stick and hdd/sdd also.
I Try all usb ports without any effect.
The operating systems I've used are Debian 11 and a fresh installed Lubuntu 22.04.

In reference to this issue, I can also confirm the keyboard bug is also represent.
https://github.com/linuxboot/heads/issues/3306
And in Addition, the three buttons on the screen are also unusable.

**To Reproduce**
Steps to reproduce the behavior:
1. download  x230-maximized-v0.2.0-1916-gf2f0831.rom
2. flash it externally on the x230t
3. try to set or change the boot partition from the heads configuration menu

I hope anyone have an idea what could be the cause?
A specific commit had change and brake something?
A generally problem with the x230t? But that can be the cause, the bios and libreboot images are for all x230* models and working without issues, so far I know.

tlaurion, 2023-12-23:
You followed https://osresearch.net/InstallingOS/#compatibility?

tlaurion, 2024-04-16:
No feedback closing

closed 2024-04-16 by tlaurion