linuxboot/heads #1545: unlock passphrase doesn't work with Qubes kernel 6.6.2

duncancmt, 2023-12-04:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

Novacustom NV41 (Nitrokey branded)

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [x] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [x] Nitrokey
- [ ] Purism
- [ ] Other provider
- [ ] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [ ] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [x] Other - Nitrokey 3A Mini

**5. Are you using the PGP key to provide HOTP verification?**
- [x] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

I deleted this section because I purchased this machine from Nitrokey

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [ ] I downloaded it
- [ ] I built it
- [x] System came pre-flashed

**2. If you downloaded your rom, where did you get it from?**
- [ ] Heads CircleCi
- [ ] Purism
- [x] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*

Nitrokey heads v2.2

**3. If you built your rom, which repository:branch did you use?**
- [ ] Heads:Master
- [x] Other (please identify) [Nitrokey/heads:v2.2](https://github.com/Nitrokey/heads/releases/tag/v2.2)

**4. What version of coreboot did you use in building?**
- [ ] 4.8.1 (current default in heads:master)
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] Other (please specify)
- [x] I don't know

**5. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [x] I don't know

## Please describe the problem

**Describe the bug**

I did a dom0 update in Qubes and now the disk unlock passphrase doesn't work anymore. Kernel 6.5.8 works fine, but kernel 6.6.2 won't boot with the unlock passphrase, only the recovery passphrase. I presume something changed with dracut or the kernel boot process preventing concatenated cpios from overriding each other, but I have no idea how I would begin to go about debugging that.

**To Reproduce**
Steps to reproduce the behavior:

1. Update Qubes dom0, installing kernel 6.6.2
2. Attempt to boot using the disk unlock passphrase
3. Observe that Plymouth still prompts for the recovery passphrase
4. Observe further that in the initramfs `/etc/crypttab` has not been patched and that `/secret.key` is missing

**Expected behavior**

Booting Qubes with the disk unlock passphrase does not prompt for the recovery passphrase. Also I would expect `/etc/crypttab` to be patched and `/secret.key` to be present in the initramfs if/when it drops into the OS recovery shell.

tlaurion, 2023-12-10:
https://github.com/Nitrokey/heads/issues/30

tlaurion, 2023-12-11:
Not Nitrokey fork related. Can replicate on newest Q4.2 kernel (6.6.2.1) as can be seen at https://github.com/Nitrokey/heads/issues/30#issuecomment-1850309877

tlaurion, 2023-12-11:
Crosslinking https://github.com/QubesOS/qubes-issues/issues/8763

tlaurion, 2023-12-17:
Fixed. See details https://github.com/QubesOS/qubes-issues/issues/8763

closed 2023-12-17 by tlaurion