linuxboot/heads #1546: Consider flashprog

tlaurion, 2023-12-06:
Flashrom stable for older boards

tlaurion, 2024-08-30:
We arrive to the point of prioritizing this https://github.com/linuxboot/heads/pull/1753#issuecomment-2315733932

i-c-o-n, 2024-08-31:
- [x] Intel support: Up to and including Raptor Lake ([Meteor Lake patch](https://review.sourcearcade.org/c/flashprog/+/189) pending, tested reading so far)
- [x] AMD support: Up to and including Zen 4 (AMD is successively locking things down, so YMMV)
- [ ] WP support: [New CLI](https://review.sourcearcade.org/c/flashprog/+/72991) under review. Needs UX feedback, current differences:
      * `flashrom --wp-status -p ...` vs. `flashprog wp --status -p ...`
      * *flashprog* only allows one command per call to avoid confusion about the execution order.
- [ ] Check for / port WP related work that landed in *flashrom* and is missing in *flashprog*.
- [ ] Check [Dasharo chip list](https://github.com/Dasharo/flashrom/blob/bbab5484c1cb64e3b641b4793b88a0179de05ca3/Documentation/chips.md) for missing, low hanging fruits.

closed 2024-10-29 by tlaurion