linuxboot/heads #1548: Build Error on Raspbian x64

sparton175, 2023-12-10:
## Context of the Build

**1. What board are you trying to build?**
t430-hotp-maximized
**2. What repository:branch are you using to build from?**
- [x] Heads:Master
- [ ] Other (please specify)

**3. What version of coreboot are you trying to build**
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] 4.17
- [x] Other (please specify) 4.19 based on the build folder

**4. In building the rom where did you get the blobs?**
- [x] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [x] No

**6. What operating system are you using**

    Kernel: Linux 6.1.0-rpi7-rpi-2712
    Architecture: arm64

## Please describe the problem

**Describe the bug**
Make errors during build of the ROM for the T430.

**To Reproduce**
Steps to reproduce the behavior:
1. Run "make BOARD=t430-hotp-maximized
2. After building several things, the process errors out
4. See error

**Expected behavior**
A successful build of the ROM.

**Screenshots**
xxxxxxxx@raspberrypi:~/heads-master $ make BOARD=t430-hotp-maximized
fatal: not a git repository (or any of the parent directories): .git
warning: Not a git repository. Use --no-index to compare two paths outside a working tree
usage: git diff --no-index [<options>] <path> <path>

Diff output format options
    -p, --patch           generate patch
    -s, --no-patch        suppress diff output
    -u                    generate patch
    -U, --unified[=<n>]   generate diffs with <n> lines context
    -W, --function-context
                          generate diffs with <n> lines context
    --raw                 generate the diff in raw format
    --patch-with-raw      synonym for '-p --raw'
    --patch-with-stat     synonym for '-p --stat'
    --numstat             machine friendly --stat
    --shortstat           output only the last line of --stat
    -X, --dirstat[=<param1,param2>...]
                          output the distribution of relative amount of changes for each sub-directory
    --cumulative          synonym for --dirstat=cumulative
    --dirstat-by-file[=<param1,param2>...]
                          synonym for --dirstat=files,param1,param2...
    --check               warn if changes introduce conflict markers or whitespace errors
[... 114 lines cut ...]
-----
         You might want to install the Texinfo package:
         <http://www.gnu.org/software/texinfo/>
         The spurious makeinfo call might also be the consequence of
         using a buggy 'make' (AIX, DU, IRIX), in which case you might
         want to install GNU make:
         <http://www.gnu.org/software/make/>
make[7]: *** [Makefile:541: bfd.info] Error 127
make[7]: Leaving directory '/home/xxxxxxxx/heads-master/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/build/local/x86_64-linux-musl/obj_binutils/bfd/doc'
make[6]: *** [Makefile:1641: info-recursive] Error 1
make[6]: Leaving directory '/home/xxxxxxxx/heads-master/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/build/local/x86_64-linux-musl/obj_binutils/bfd'
make[5]: *** [Makefile:2691: all-bfd] Error 2
make[5]: Leaving directory '/home/xxxxxxxx/heads-master/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/build/local/x86_64-linux-musl/obj_binutils'
make[4]: *** [Makefile:850: all] Error 2
make[4]: Leaving directory '/home/xxxxxxxx/heads-master/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/build/local/x86_64-linux-musl/obj_binutils'
make[3]: *** [Makefile:204: obj_binutils/.lc_built] Error 2
make[3]: Leaving directory '/home/xxxxxxxx/heads-master/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/build/local/x86_64-linux-musl'
make[2]: *** [Makefile:170: install] Error 2
make[2]: Leaving directory '/home/xxxxxxxx/heads-master/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1'
make[1]: *** [Makefile:174: musl-target] Error 2
make[1]: Leaving directory '/home/xxxxxxxx/heads-master/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1'
make: *** [Makefile:500: /home/xxxxxxxx/heads-master/build/x86/musl-cross-38e52db8358c043ae82b346a2e6e66bc86a53bc1/.build] Error 1

**Additional context**
From my understanding the blobs required to build a T430 ROM are already provided.

sparton175, 2023-12-19:
Made it further this time but ran into another error:

xxxxxxx@raspberrypi:~/heads-master $ make BOARD=t430-hotp-maximized
fatal: not a git repository (or any of the parent directories): .git
warning: Not a git repository. Use --no-index to compare two paths outside a working tree
usage: git diff --no-index [<options>] <path> <path>

Diff output format options
    -p, --patch           generate patch
    -s, --no-patch        suppress diff output
    -u                    generate patch
    -U, --unified[=<n>]   generate diffs with <n> lines context
    -W, --function-context
                          generate diffs with <n> lines context
    --raw                 generate the diff in raw format
    --patch-with-raw      synonym for '-p --raw'
    --patch-with-stat     synonym for '-p --stat'
    --numstat             machine friendly --stat
    --shortstat           output only the last line of --stat
    -X, --dirstat[=<param1,param2>...]
                          output the distribution of relative amount of changes for each sub-directory
    --cumulative          synonym for --dirstat=cumulative
    --dirstat-by-file[=<param1,param2>...]
                          synonym for --dirstat=files,param1,param2...
    --check               warn if changes introduce conflict markers or whitespace errors
    --summary             condensed summary such as creations, renames and mode changes
    --name-only           show only names of changed files
    --name-status         show only names and status of changed files
    --stat[=<width>[,<name-width>[,<count>]]]
                          generate diffstat
    --stat-width <width>  generate diffstat with a given width
    --stat-name-width <width>
                          generate diffstat with a given name width
    --stat-graph-width <width>
                          generate diffstat with a given graph width
    --stat-count <count>  generate diffstat with limited lines
    --compact-summary     generate compact summary in diffstat
    --binary              output a binary diff that can be applied
    --full-index          show full pre- and post-image object names on the "index" lines
    --color[=<when>]      show colored diff
    --ws-error-highlight <kind>
                          highlight whitespace errors in the 'context', 'old' or 'new' lines in the diff
    -z                    do not munge pathnames and use NULs as output field terminators in --raw or --numstat
    --abbrev[=<n>]        use <n> digits to display object names
    --src-prefix <prefix>
                          show the given source prefix instead of "a/"
    --dst-prefix <prefix>
                          show the given destination prefix instead of "b/"
    --line-prefix <prefix>
                          prepend an additional prefix to every line of output
    --no-prefix           do not show any source or destination prefix
    --inter-hunk-context <n>
                          show context between diff hunks up to the specified number of lines
    --output-indicator-new <char>
                          specify the character to indicate a new line instead of '+'
    --output-indicator-old <char>
                          specify the character to indicate an old line instead of '-'
    --output-indicator-context <char>
                          specify the character to indicate a context instead of ' '

Diff rename options
    -B, --break-rewrites[=<n>[/<m>]]
                          break complete rewrite changes into pairs of delete and create
    -M, --find-renames[=<n>]
                          detect renames
    -D, --irreversible-delete
                          omit the preimage for deletes
    -C, --find-copies[=<n>]
                          detect copies
    --find-copies-harder  use unmodified files as source to find copies
[... 57 lines cut ...]
    --find-object <object-id>
                          look for differences that change the number of occurrences of the specified object
    --diff-filter [(A|C|D|M|R|T|U|X|B)...[*]]
                          select files by diff type
    --output <file>       output to a specific file

fatal: not a git repository (or any of the parent directories): .git
Makefile:201: warning: overriding recipe for target 'all'
/home/xxxxxxx/heads-master/boards/t430-hotp-maximized/t430-hotp-maximized.config:86: warning: ignoring old recipe for target 'all'
2023-12-18 23:23:39-07:00 INSTALL-MODULE drivers/net/ethernet/intel/e1000e/e1000e.ko
2023-12-18 23:23:39-07:00 INSTALL-MODULE drivers/usb/host/ehci-hcd.ko
2023-12-18 23:23:39-07:00 INSTALL-MODULE drivers/usb/host/ehci-pci.ko
2023-12-18 23:23:39-07:00 INSTALL-MODULE drivers/usb/host/xhci-hcd.ko
2023-12-18 23:23:39-07:00 INSTALL-MODULE drivers/usb/host/xhci-pci.ko
2023-12-18 23:23:39-07:00 INSTALL-MODULE drivers/usb/storage/usb-storage.ko
2023-12-18 23:23:39-07:00 SYMLINK bin/busybox
2023-12-18 23:23:41-07:00 INSTALL-BIN build/x86/kexec-tools-2.0.26/build/sbin/kexec
2023-12-18 23:23:41-07:00 INSTALL-BIN build/x86/tpmtotp-4d63d21c8b7db2e92ddb393057f168aead147f47/totp
2023-12-18 23:23:41-07:00 INSTALL-BIN build/x86/tpmtotp-4d63d21c8b7db2e92ddb393057f168aead147f47/hotp
2023-12-18 23:23:41-07:00 INSTALL-BIN build/x86/tpmtotp-4d63d21c8b7db2e92ddb393057f168aead147f47/qrenc
2023-12-18 23:23:41-07:00 INSTALL-BIN build/x86/tpmtotp-4d63d21c8b7db2e92ddb393057f168aead147f47/util/tpm
2023-12-18 23:23:41-07:00 INSTALL-BIN build/x86/pciutils-3.5.4/lspci
2023-12-18 23:23:41-07:00 INSTALL-BIN build/x86/flashrom-1776bb46ba6ea3d1ab2ec3f0cd88158aabed7400/flashrom
2023-12-18 23:23:41-07:00 CONFIG popt
make: *** [Makefile:500: /home/xxxxxxx/heads-master/build/x86/popt-1.16/.configured] Error 1

tlaurion, 2023-12-23:
Please redo make BOARD=xyz V=1 and share full console output. I don't get the git error nor can make sense of the error. You replicate debian dependencies exposed by apt calls under Circleci config.

tlaurion, 2024-05-25:
We switched to nix develop creatd docker images under #1269

closed 2024-05-25 by tlaurion