linuxboot/heads #1550: Stop unpacking & repacking the initramfs

DemiMarie, 2023-12-12:
**Is your feature request related to a problem? Please describe.**
Unpacking & repacking the initramfs is quite fragile and is not a very elegant solution.  More importantly, it is incompatible with Unified Kernel Images (UKIs) that combine the hypervisor, kernel, and initramfs into a single signed binary.

**Describe the solution you'd like**
Instead of Heads injecting the secret into the initramfs, Heads should leave the secret in the TPM for use by the kexec’d OS.  The OS is then responsible for using the secret to unlock the drive and for deleting the secret when no longer needed.

**Describe alternatives you've considered**
None.

**Additional context**
This might prevent #1545.  It also allows the OS to use other types of encrypted storage, such as encrypted ZFS or bcachefs volumes.

tlaurion, 2023-12-17:
Related discussions happened at 
- https://github.com/QubesOS/qubes-issues/issues/8763#issuecomment-1858909201
- https://github.com/linuxboot/heads/issues/1499#issuecomment-1858939270

As of now this suggestion is also highly impractical.

Note that https://github.com/QubesOS/qubes-issues/issues/8763 and associated QSB https://github.com/QubesOS/qubes-secpack/blob/master/QSBs/qsb-098-2023.txt resolved the issue. For now.

tlaurion, 2023-12-17:
@DemiMarie please reopen issue if you think it was closed too soon and justify why it should be prioritized.

closed 2023-12-17 by tlaurion

tlaurion, 2023-12-17:
> This might prevent https://github.com/linuxboot/heads/issues/1545. It also allows the OS to use other types of encrypted storage, such as encrypted ZFS or bcachefs volumes.

As for this specifically, since we are under linuxboot context here, and Heads **should as well be considered as a trusted recovery shell environement**, Heads participates in upstream discussions to define **what the kernel should support in its really limited SPI storage space**. So supporting ZFS or bcachefs should be different issues under Heads to be discussed accordingly. As of today, **Heads support what QubesOS supports as its defaults** which are LUKS+LVM2 pools in the goal of providing its users capabilites to unlock LUKS volumes and inspect LVM volumes, mounting them if necessary (ro by default). 

**Supporting ZFS+bcachefs are not even planned as to be supported under QubesOS** as of now, where proper support for BRTFS and comparative to defaults, or changing defaults to BRTFS is not even a thing today under QubesOS. When QubesOS will change, or when Heads users show need to have support for such filesystems in recovry environement, then those subects can be reaised in seperate issues. Otherwise we are shelving dreams here, dreams not put practical in action. Hear me out though, Ubuntu is pushing for ZFS. But heads user base is not really Ubuntu as of today.

I opened issues https://github.com/Nitrokey/ubuntu-oem/issues/11 and similar downstream in projects tolerated per QubesOS, just like https://github.com/Nitrokey/qubes-oem/issues/20, so that unexperienced users not able to follow https://osresearch.net/InstallingOS/#compatibility have something to trust booting which would do the right thing.

TLDR: once those things exist, Heads will adapt. Until then, bear with me. Doing all of that is far above my pay grade. Which is another story https://github.com/sponsors/tlaurion