linuxboot/heads #1557: HDMI output usually not available after cold boot

marmarek, 2023-12-19:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

x230

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [x] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [x] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [x] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [ ] Other

**5. Are you using the PGP key to provide HOTP verification?**
- [x] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ ] First-time flash
- [ ] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ] Using the Heads GUI
- [x] Flashrom via the Recovery Shell
- [ ] External flashing

**3. How was Heads initially flashed**
- [ ] External flashing
- [x] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [x] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [ ] Yes
- [ ] No
- [ ] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [x] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [x] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*

https://circleci.com/gh/linuxboot/heads/14178 ( x230-hotp-maximized_usb-kb of https://github.com/linuxboot/heads/commit/4a57c615e972149eefd52d95ba919ff54d53bb0a)

## Please describe the problem

**Describe the bug**

HDMI output is not available on cold boot (no signal). Pressing ctrl+alt+delete usually fixes it. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect HDMI output to the laptop (via miniDP->HDMI converter)
2. Press power button

**Expected behavior**

Heads output on both internal screen and the external one.

**Additional context**

It wasn't a problem on a (much) older version - the one used for initial x230 Qubes OS certification.

I've tested also commit just before merging https://github.com/linuxboot/heads/pull/1522 and the issue happens there too.

tlaurion, 2023-12-19:
@marmarek Do you see cbmem -1/dmesg differences between coldboot/warmboot? 
Definitely a coreboot issue here but if we can see something different there then a proper upstream linked bug report might lead to resolving the issue here (again coreboot upstream is 4.22 and version used here 4.19 because fhd patchset not merged upstream and used downstream... but that is different story)

tlaurion, 2023-12-22:
@marmarek ping

tlaurion, 2023-12-22:
@marmarek ping

marmarek, 2023-12-22:
Logs here.

[cbmem-cold.txt](https://github.com/linuxboot/heads/files/13756919/cbmem-cold.txt)
[cbmem-warm.txt](https://github.com/linuxboot/heads/files/13756920/cbmem-warm.txt)
[dmesg-cold.txt](https://github.com/linuxboot/heads/files/13756921/dmesg-cold.txt)
[dmesg-warm.txt](https://github.com/linuxboot/heads/files/13756922/dmesg-warm.txt)

I don't see much difference besides indeed coreboot log says HDMI is disabled on cold boot. BTW, after warm reboot Linux says `input: AT Translated Set 2`, while after cold boot it's `input: AT Raw Set 2 keyboard` - related to https://github.com/linuxboot/heads/issues/958 ? Note, I stopped at Heads here (collected logs from recovery shell), Qubes (or any other OS) was not started between reboots.

bahrom04, 2026-09-14:
this issue also persist on novacustom v56 V560TNx laptop.

tlaurion, 2026-09-14:
> this issue also persist on novacustom v56 V560TNx laptop.

@bahrom04 
- v560TU (intel iGPU) is supported: https://docs.dasharo.com/variants/novacustom_v560tu/hardware-matrix/
   - Has a Dasharo+Heads release https://docs.dasharo.com/variants/novacustom_v560tu/releases_heads/
-  V560TNx doesn't have a Dasharo+Heads release https://docs.dasharo.com/variants/novacustom_v560tnx/releases/ and is not Heads tested.

Sorry, but you will have to create issue under Dasharo, this is irrelevant here.

tlaurion, 2026-09-14:
@marmarek I'm sorry about this and didn't come back to this issue and not sure How Heads is supposed to handle upstream issues, like the AT translated/AT raw issue: this needs to be coreboot related, and still unclear how this is supposed to be fixed. 

You seem to have a workaround though, which is to reset/ctrl-alt-delete?
An issue would need to be opened under https://ticket.coreboot.org/issues/ and linked here for resolution.

Best I could do here is to bump coreboot to 26.06/26.09 and have you test again, so issue could be filed under https://ticket.coreboot.org/issues/

You up for testing this @marmarek? I know this must be related to OpenQA testing and HDMI output, I can help but won't resolve this, and totally unsure why this is different then other payload since it should not be payload related.

marmarek, 2026-09-14:
> You seem to have a workaround though, which is to reset/ctrl-alt-delete?

Yup, I have this workaround, and while not perfect, it's good enough for me. That said, The issue appears to be fixed already. Looking at some select openQA logs. It was broken in a [test 5 months ago](https://openqa.qubes-os.org/tests/171255#step/install_startup/1) (doesn't mean Heads version from there, likely substantially older, but unfortunately logs are purged already), but with more recent test [it works fine](https://openqa.qubes-os.org/tests/187688#step/startup/1) (Heads-v0.2.1-3087-g6ebc734) already.

> this issue also persist on novacustom v56 V560TNx laptop.

This is very likely completely different issue. On those laptops the HDMI output is connected to Nvidia GPU, not Intel one, and I believe the driver for it is not included in Heads.

closed 2026-09-14 by marmarek

tlaurion, 2026-09-15:
> > this issue also persist on novacustom v56 V560TNx laptop.
> 
> This is very likely completely different issue. On those laptops the HDMI output is connected to Nvidia GPU, not Intel one, and I believe the driver for it is not included in Heads.

@bahrom04 V560TNx laptop could have  Heads port. @marmarek was it QubesOS certified? Otherwise @bahrom04 maybe ask Nitrokey/Novacustom to fund port if their is interest?

bahrom04, 2026-09-15:
> [@bahrom04](https://github.com/bahrom04) V560TNx laptop could have Heads port. [@marmarek](https://github.com/marmarek) was it QubesOS certified? Otherwise [@bahrom04](https://github.com/bahrom04) maybe ask Nitrokey/Novacustom to fund port if their is interest?

So found whatʻs causing this issue. It turns out gnome-shell uses 2MB of of nvidia memory on background even i enabled nvidia prime offload (uses igpu but wakes up dgpu when needed). I did disabled this 2MB memory consumption in order to let my dgpu sleep in d3cold mode but hdmi connection not worked.

```nix
  services.udev.extraRules = ''
  SUBSYSTEM=="drm", DRIVERS=="nvidia", TAG+="mutter-device-ignore"
  '';
```

I found this code snipped on [gitlab issues](https://gitlab.gnome.org/GNOME/mutter/-/work_items/2310) but disabled after because i needed hdmi connection after all. Thankʻs for quick reply 👍️

marmarek, 2026-09-15:
> > 
> 
> [@bahrom04](https://github.com/bahrom04) V560TNx laptop could have Heads port. [@marmarek](https://github.com/marmarek) was it QubesOS certified?

No, the TN models are not currently Qubes OS certified.