linuxboot/heads #1560: X230 FHD-eDP internal screen blank

jmlzr, 2023-12-21:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

x230-maximized-fhd_edp

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [x] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [x] Self-installed

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [x] First-time flash
- [ ] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [x] External flashing

**3. How was Heads initially flashed**
- [x] External flashing
- [ ] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [x] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [ ] Yes
- [ ] No
- [x] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [x] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [x] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

v0.2.0-1952-g25d7b06-bottom.rom
https://app.circleci.com/pipelines/github/linuxboot/heads/715/workflows/10754d2a-aa0f-419b-834f-044986e6f9ef/jobs/14382/artifacts

## Please describe the problem

**Describe the bug**

The internal (eDP) screen stays blank in heads. Heads will show up and work on a external display connected to the MiniDP port though.
Once a "full" Linux is booted (tested the Arch install media) the internal eDP screen works fine.

**To Reproduce**

1. Externally flash heads
2. Turn on Computer

**Expected behavior**

Heads uses the internal display.

**Additional context**

X230 with FHD/eDP Mod (not nitrocasters but similar: same DP lanes used; no MCU; BL_EN and BL_CTRL directly connected; no EDID EEPROM).

tlaurion, 2023-12-22:
https://github.com/linuxboot/heads/pull/967

See comments there. Seems like your particular mod is not supported by referred patchset applied on top of coreboot 4.19

tlaurion, 2023-12-22:
https://github.com/linuxboot/heads/pull/967

See comments there. Seems like your particular mod is not supported by referred patchset applied on top of coreboot 4.19

Discussions should happen at https://review.coreboot.org/c/coreboot/+/28950

jmlzr, 2023-12-22:
Thanks, I flew over that thread but couldn't find much information regarding to different mods...
I must add that I designed my mod myself, it's supposed to be a minimalist and cheap alternative designed to be used with modified firmware (so no MCU and fake EDID on board). I'm going to put it on my github soon as I want it to be open hardware. I thought it should work with the same patch though.

I know the coreboot thread and will join that discussion soon. Sadly that thread is a bit of a mess.
I tried that coreboot patch directly with an edk2 payload at first but hat the same issue (i.e. it's booting but no display output in firmware stage). There also seems to be 0 documentation in this thread (or anywhere else online) on how they modified the VBT.

Since unmodified (Arch) Linux has no problem working with my mod without any special settings and heads uses Linux I was hopeful I might have more luck with it. And heads is also probably the coolest firmware for the X230.

tlaurion, 2023-12-24:
> Could be the that heads uses the framebuffer while Linux distros use the full blown i915 driver which might be able to reconfigure properly on it's own?

Well, edp needs to be exposed correctly for efifb to be exposed, where yes, i915+drm would be more agressive on init path to get it properly.

Butnif coreboot was doing the right thing in libgfxinit then efifb would be able to drive the panel. Something is different for your mod and needs to be applied there for libgfxinit and efifb to pick and drive it up.

closed 2026-03-05 by tlaurion