linuxboot/heads #1565: Error while trying to store LUKS disk encryption keys in TPM with X230-hotp-maximized CircleCI

copyvar, 2023-12-27:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**
X230

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] Don't know

**3. Who installed Heads on this computer?**
- [ ] Self-installed

**4. What PGP key is being used?**
Nitrokey Pro 2

**5. Are you using the PGP key to provide HOTP verification?**
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ ] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell

**3. How was Heads initially flashed**
- [ ] External flashing

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [ ] Maximized

**5. If Heads was externally flashed, was IFD unlocked?**
- [ ] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [ ] I downloaded it

**2. If you downloaded your rom, where did you get it from?**
- [ ] Heads CircleCi

*Please provide the release number or otherwise identify the rom downloaded*
https://app.circleci.com/pipelines/github/linuxboot/heads/715/workflows/ff6c0f29-2c3b-4327-8155-8bff9907b1d5/jobs/14347

**3. If you built your rom, which repository:branch did you use?**
- [ ] I don't know

**4. What version of coreboot did you use in building?**
- [ ] I don't know

**5. In building the rom where did you get the blobs?**
- [ ] Extracted from a backup rom taken from this device 
- [ ] I don't know

## Please describe the problem

**Describe the bug**
After installing Qubes 4.2 on my X230 I want to store the LUKS disk encryption key in the TPM for easier usage. 

Steps I did:
- flash heads to the newest version from CircleCI
- install Qubes 4.2 with default setup options
- reboot (ignoring all warnings and running unsafe mode to finish Qubes installation)
- update Qubes 4.2
- reboot
- starting heads recommended oem factory reset/re-ownership, everything works
- reboot
- try to make Qubes 4.2 default and seal LUKS keys in TPM

The following is a shorter version/summary of the terminal output, hopefully I included all relevant information:
HOTP code is correct 
found verified kexec boot params
good gpg signature
verfied boot hashes

do you wish to add a disk encryption key to the TPM -> yes
no encrypted lvms found
single encrypted disk found at /dev/sda2
enter disk recovery key/passphrase -> here I entered the passphrase I used when installing Qubes
new tpm disk unlock key -> new passphrase
repeat tpm disk unlock key -> again

generating new randomized...
removing old key slot 1
keyslot 1 is not active
warning removal of key in slot 1 failed: might not exist.
adding key to slot 1
new value of pcr6
error illegal index from NV_WriteValue
tpm owner password -> the one which i set up in reownership process
got error error "authentication failed (incorrect password)" from TPM_NV_DefineSpace2().
warning: unable to define TPM NVRAM space: trying anyway
error illegal index from NV_WriteValue
!!! error: unable to write sealed secret to TPM NVRAM !!!
!!! error: unable to write TPM disk unlock key to NVRAM !!!
!!! error: failed to save and generate TPM Disk Unlock Key !!!
!!! error: failed to save the  TPM Disk Unlock Key !!!
failed to save defaults
head: invalid number "/tmp/kexec/kexec_menu.txt"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

tlaurion, 2023-12-29:
@copyvar Please confirm that ROM produced by #1566 fixes the issue

closed 2023-12-29 by tlaurion

copyvar, 2023-12-30:
@tlaurion It works, perfect!

PR #1566 (2023-12-29): tpmr: fix TPM Disk Unlock Key which was not using proper cached TPM owner passphrase.
- change TPM DUK sealing which needs to be on cached TPM owner passphrase
- Add debugging that was needed to spot the issue

Fixes #1565