linuxboot/heads #1569: t420-hotp-maximized

h2dden, 2024-01-01:
## Context of the Build
fresh debian 12, building t420-hotp-maximized
**1. What board are you trying to build?**
t420-hotp-maximized
**2. What repository:branch are you using to build from?**
- [X ] Heads:Master

**3. What version of coreboot are you trying to build**
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] 4.17
- [X ] Other (please specify) seems like 4.19

**4. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [X ] I don't know: followed the instructions at: https://osresearch.net/general-building/

**5. If using the automated tools to get the blobs did you run the relevant scripts in the blobs directory**
- [ ] Yes
- [ ] No

**6. What operating system are you using**
debian 12
*please specify*

## Please describe the problem
build process fails with:
    AR         bootblock/external/vboot_reference-bootblock/vboot_fw.a
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/12/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 12.2.0-14' --with-bugurl=file:///usr/share/doc/gcc-12/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-12 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-12-bTRWOB/gcc-12-12.2.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-12-bTRWOB/gcc-12-12.2.0/debian/tmp-gcn/usr --enable-offload-defaulted --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.0 (Debian 12.2.0-14) 
MAKEFLAGS=rRsw -j8 -l16  --no-print-directory -- CFLAGS_x86_64=-fdebug-prefix-map=/home/user/heads=heads\ -gno-record-gcc-switches\ -Wno-error=packed-not-aligned\ -Wno-address-of-packed-member CFLAGS_x86_32=-fdebug-prefix-map=/home/user/heads=heads\ -gno-record-gcc-switches\ -Wno-error=packed-not-aligned\ -Wno-address-of-packed-member BUILD_TIMELESS=1 XGCCPATH=/home/user/heads/build/x86/coreboot-4.19/util/crossgcc/xgcc/bin/ DOTCONFIG=/home/user/heads/build/x86/coreboot-4.19/t420-hotp-maximized/.config obj=/home/user/heads/build/x86/coreboot-4.19/t420-hotp-maximized BOARD=t420-hotp-maximized
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/12/:/usr/lib/gcc/x86_64-linux-gnu/12/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/12/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/12/:/usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/12/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/12/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 't420-hotp-maximized/util/cbfstool/cbfstool' '-mtune=generic' '-march=x86-64' '-dumpdir' 't420-hotp-maximized/util/cbfstool/cbfstool.'
 /usr/lib/gcc/x86_64-linux-gnu/12/collect2 -plugin /usr/lib/gcc/x86_64-linux-gnu/12/liblto_plugin.so -plugin-opt=/usr/lib/gcc/x86_64-linux-gnu/12/lto-wrapper -plugin-opt=-fresolution=/tmp/ccwPHNhD.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -dynamic-linker /lib64/ld-linux-x86-64.so.2 -pie -o t420-hotp-maximized/util/cbfstool/cbfstool /usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/Scrt1.o /usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/12/crtbeginS.o -L/usr/lib/gcc/x86_64-linux-gnu/12 -L/usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/12/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/12/../../.. t420-hotp-maximized/util/cbfstool/cbfstool.o t420-hotp-maximized/util/cbfstool/common.o t420-hotp-maximized/util/cbfstool/cbfs_image.o t420-hotp-maximized/util/cbfstool/cbfs-mkstage.o t420-hotp-maximized/util/cbfstool/cbfs-mkpayload.o t420-hotp-maximized/util/cbfstool/elfheaders.o t420-hotp-maximized/util/cbfstool/rmodule.o t420-hotp-maximized/util/cbfstool/xdr.o t420-hotp-maximized/util/cbfstool/partitioned_file.o t420-hotp-maximized/util/cbfstool/platform_fixups.o t420-hotp-maximized/util/cbfstool/cbfs_private.o t420-hotp-maximized/util/cbfstool/fsp_relocate.o t420-hotp-maximized/util/cbfstool/fmap.o t420-hotp-maximized/util/cbfstool/kv_pair.o t420-hotp-maximized/util/cbfstool/valstr.o t420-hotp-maximized/util/cbfstool/linux_trampoline.o t420-hotp-maximized/util/cbfstool/cbfs-payload-linux.o t420-hotp-maximized/util/cbfstool/compress.o t420-hotp-maximized/util/cbfstool/lz4.o t420-hotp-maximized/util/cbfstool/lz4hc.o t420-hotp-maximized/util/cbfstool/lz4frame.o t420-hotp-maximized/util/cbfstool/xxhash.o t420-hotp-maximized/util/cbfstool/lz4_wrapper.o t420-hotp-maximized/util/cbfstool/lzma.o t420-hotp-maximized/util/cbfstool/LzFind.o t420-hotp-maximized/util/cbfstool/LzmaDec.o t420-hotp-maximized/util/cbfstool/LzmaEnc.o /home/user/heads/build/x86/coreboot-4.19/t420-hotp-maximized/util/vboot_lib/libvboot_host.a -lgcc --push-state --as-needed -lgcc_s --pop-state -lc -lgcc --push-state --as-needed -lgcc_s --pop-state /usr/lib/gcc/x86_64-linux-gnu/12/crtendS.o /usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/crtn.o
/usr/bin/ld: /home/user/heads/build/x86/coreboot-4.19/t420-hotp-maximized/util/vboot_lib/libvboot_host.a(2crypto.o): warning: relocation in read-only section `.rodata.vb2_sig_names'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
COLLECT_GCC_OPTIONS='-v' '-o' 't420-hotp-maximized/util/cbfstool/cbfstool' '-mtune=generic' '-march=x86-64' '-dumpdir' 't420-hotp-maximized/util/cbfstool/cbfstool.'
make[1]: Leaving directory '/home/user/heads/build/x86/coreboot-4.19'
make: *** [Makefile:511: /home/user/heads/build/x86/coreboot-4.19/t420-hotp-maximized/.build] Error 1

what am I doing wrong?

tlaurion, 2024-01-02:
Duplicate of https://github.com/linuxboot/heads/issues/1414

closed 2024-01-02 by tlaurion

tlaurion, 2024-01-02:
```
user@heads-tests-deb12:~/heads$ grep -Rn apt .circleci/config.yml 
17:            apt update
18:            apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg gawk iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract sudo libssl-dev device-tree-compiler u-boot-tools sharutils e2fsprogs parted curl unzip imagemagick libncurses5-dev zip
53:            apt update
54:            apt install -y build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg gawk iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract sudo imagemagick libncurses5-dev
user@heads-tests-deb12:~/heads$ rm -rf build/x86/t420-* build/x86/coreboot-4.19/t420*
user@heads-tests-deb12:~/heads$ make BOARD=t420-hotp-maximized
2024-01-02 12:31:19-05:00 CONFIG coreboot-4.19
2024-01-02 12:31:21-05:00 INSTALL   build/x86/linux-5.10.5/linux-x230-maximized//arch/x86/boot/bzImage => build/x86/t420-hotp-maximized/bzImage
0166936f9a95b8f1a2e076c896c811f72550946dcbc5d7120f8aa0f6f24fc4c2  build/x86/t420-hotp-maximized/bzImage
 2488800:build/x86/t420-hotp-maximized/bzImage
0166936f9a95b8f1a2e076c896c811f72550946dcbc5d7120f8aa0f6f24fc4c2  /home/user/heads/build/x86/t420-hotp-maximized/bzImage
 2488800:/home/user/heads/build/x86/t420-hotp-maximized/bzImage
2024-01-02 12:31:21-05:00 INSTALL-MODULE drivers/net/ethernet/intel/e1000e/e1000e.ko
2024-01-02 12:31:21-05:00 INSTALL-MODULE drivers/usb/host/ehci-hcd.ko
2024-01-02 12:31:21-05:00 INSTALL-MODULE drivers/usb/host/ehci-pci.ko
2024-01-02 12:31:21-05:00 INSTALL-MODULE drivers/usb/host/xhci-hcd.ko
2024-01-02 12:31:21-05:00 INSTALL-MODULE drivers/usb/host/xhci-pci.ko
2024-01-02 12:31:21-05:00 INSTALL-MODULE drivers/usb/storage/usb-storage.ko
2024-01-02 12:31:21-05:00 CPIO      build/x86/t420-hotp-maximized/modules.cpio
df96938fd5cdf061aa9e4e07171375e7fbe7948b482bc1a319a3f66c453b5173  /home/user/heads/build/x86/t420-hotp-maximized/modules.cpio
  685056:/home/user/heads/build/x86/t420-hotp-maximized/modules.cpio
2024-01-02 12:31:21-05:00 HASHES    build/x86/t420-hotp-maximized/modules.cpio
2024-01-02 12:31:21-05:00 SIZES     build/x86/t420-hotp-maximized/modules.cpio
2024-01-02 12:31:21-05:00 SYMLINK bin/busybox
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/kexec-tools-2.0.26/build/sbin/kexec
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/tpmtotp-4d63d21c8b7db2e92ddb393057f168aead147f47/totp
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/tpmtotp-4d63d21c8b7db2e92ddb393057f168aead147f47/hotp
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/tpmtotp-4d63d21c8b7db2e92ddb393057f168aead147f47/qrenc
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/tpmtotp-4d63d21c8b7db2e92ddb393057f168aead147f47/util/tpm
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/pciutils-3.5.4/lspci
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/flashrom-1776bb46ba6ea3d1ab2ec3f0cd88158aabed7400/flashrom
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/cryptsetup-2.3.3/.libs/cryptsetup
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/cryptsetup-2.3.3/.libs/cryptsetup-reencrypt
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/cryptsetup-2.3.3/.libs/veritysetup
2024-01-02 12:31:23-05:00 INSTALL-BIN build/x86/gnupg-2.4.0/g10/gpg
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/gnupg-2.4.0/agent/gpg-agent
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/gnupg-2.4.0/scd/scdaemon
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/pinentry-1.1.0/tty/pinentry-tty
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/lvm2.2.02.168/tools/dmsetup
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/lvm2.2.02.168/tools/lvm
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/flashtools-d1e6f12568cb23387144a4b7a6535fe1bc1e79b1/flashtool
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/flashtools-d1e6f12568cb23387144a4b7a6535fe1bc1e79b1/peek
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/flashtools-d1e6f12568cb23387144a4b7a6535fe1bc1e79b1/poke
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/flashtools-d1e6f12568cb23387144a4b7a6535fe1bc1e79b1/cbfs
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/flashtools-d1e6f12568cb23387144a4b7a6535fe1bc1e79b1/uefi
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/fbwhiptail-1.3/fbwhiptail
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/fbwhiptail-1.3/whiptail
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/hotp-verification-b69bb20119d3cea5ec5c13d11b213dd80dfd8334/hotp_verification
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/hotp-verification-b69bb20119d3cea5ec5c13d11b213dd80dfd8334/hotp_initialize
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/bash-5.1.16/bash
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/io386-fc73fcf8e51a70638679c3e9b0ada10527f8a7c1/io386
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/zstd-1.5.5/programs/zstd-decompress
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/e2fsprogs-1.47.0/misc/mke2fs
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/exfatprogs-1.2.1/fsck/fsck.exfat
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/exfatprogs-1.2.1/mkfs/mkfs.exfat
2024-01-02 12:31:24-05:00 INSTALL-BIN build/x86/nvmutil-tools-20220815/nvm
2024-01-02 12:31:24-05:00 MAKE cbmem
[... cut ...]
cd "/home/user/heads/build/x86/t420-hotp-maximized/update_pkg" && zip -9 "/home/user/heads/build/x86/t420-hotp-maximized/heads-t420-hotp-maximized-v0.2.0-1959-g8701d14.zip" "heads-t420-hotp-maximized-v0.2.0-1959-g8701d14.rom" sha256sum.txt
  adding: heads-t420-hotp-maximized-v0.2.0-1959-g8701d14.rom (deflated 12%)
  adding: sha256sum.txt (deflated 15%)
7b461d373ccc6eecbab15e12e08acadf3570b8e38da2e23e4002bec77404da7c  /home/user/heads/build/x86/t420-hotp-maximized/heads-t420-hotp-maximized-v0.2.0-1959-g8701d14.rom
 8388608:/home/user/heads/build/x86/t420-hotp-maximized/heads-t420-hotp-maximized-v0.2.0-1959-g8701d14.rom

```

tlaurion, 2024-01-02:
TLDR: CircleCI (continuous environement) is on debian-11 docker image. Builds roms and is the supported way for the moment.