linuxboot/heads #1603: OS Reboot Hangs - x230 fhd_edp

lethedata, 2024-02-03:
## Basic Details

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**
[x230-maximized-fhd_edp](https://github.com/linuxboot/heads/tree/master/boards/x230-maximized-fhd_edp)

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [x] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [x] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [ ] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [x] Yubikey
- [ ] Other

**5. Are you using the PGP key to provide HOTP verification?**
- [ ] Yes
- [x] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ ] First-time flash
- [x] Updating heads

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [x] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [ ] External flashing

**3. How was Heads initially flashed**
- [x] External flashing
- [ ] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [x] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [x] Yes
- [ ] No
- [ ] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [x] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [x] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*
heads-x230-maximized-fhd_edp-v0.2.0-2022-g5bff519

## Problem

**Describe the bug**
After updating from an older version (early Feb 2023 I think) of Heads rebooting from OS doesn't seem to work. All the systems lights turn off (USB still blinks if plugged in) but the system hangs in that state appearing to be powered off. Holding the power button leads to the "pop" sound as the system is forced shutdown. I experienced this in both the latest Fedora and QubesOS.

**To Reproduce**
Steps to reproduce the behavior:
1. Boot OS
2. Login
3. Reboot device

**Expected behavior**
Successful reboot of system

**Additional context**
Rebooting from heads itself works without issue. Non-fhd_edp  x230 appears to works without issue.

lethedata, 2024-02-03:
Not really sure of any other troubleshooting method except building out some of the older versions till one works. Any ideas?

tlaurion, 2024-02-03:
The edp patch changed last time at https://github.com/linuxboot/heads/commit/6300dd178a35cbe68bc6587fdf48a210f1437607 so around 1year ago.

This patch was never merged upstream under coreboot as referred per commit reference to upstream unmerged https://review.coreboot.org/c/coreboot/+/28950

First time this issue is reported. Per https://github.com/linuxboot/heads/issues/692 edp known board owners: x230-fhd/edp variant: @n4ru @computer-user123 (nitro caster board) @Tonux599 @househead @pcm720 (eDP 4.0 board and 1440p display)

Can you replicate this issue?

pcm720, 2024-02-03:
Cannot replicate on the latest Linux kernel currently available in Arch Linux (Linux 6.7.3-arch1-1) and heads-x230-maximized-fhd_edp-v0.2.0-2022-g5bff519

tlaurion, 2024-02-03:
> owners/willingntesters: x230-fhd/edp variant: @n4ru @computer-user123 (nitro caster board) @Tonux599 @househead @pcm720 (eDP 4.0 board and 1440p display)

As one can read upstream, eg nitrocaster mod doesn't require the mod? Other mods doesn't require patched vbt or edp to be enabled to drive Fhd? Basically, each board owner here should state clearly there what is the mod they use and report upstream about funcional/not functional state and _only then_ upstream final patchset should state what kits are covered to be working with modified vbt and board variant. As stated there, using edp board variant might not work depending of the kit used and this is most probably why this patchset was never merged usptream yet.

pcm720, 2024-02-03:
@tlaurion, like I mentioned in some previous issue or MR, I don't know what kind of participation is expected from us to get this patch merged. It has always worked perfectly for me.  

What this patch actually does is disabling the internal LCD and setting external display as internal so that OS only sees one internal display, it has nothing to do with the actual brightness control or driving external display since it's connected to DisplayPort on the docking connector. All mods work fine without the patch, but without it the OS sees two displays: one internal (that's no longer connected) and external (the one actually installed).
 
In the comments that are supposed to be resolved before this patch can be accepted on coreboot's Gerrit, I don't see anything actually blocking and requiring any rewrites. As far as I'm aware, all variations of this mod is working fine with that patch if they were installed as intended by their creators.

In my opinion, everything else (like bridging VCC3P as suggested by some X220 patches) should be considered unsupported and YMMV unless someone decides to implement it in another patchset.

lethedata, 2024-02-03:
I'd agree with @pcm720 on the outstanding comments preventing getting it merged upstream. The VCC3P should be a separate future patch but shouldn't prevent the current patch from being merged.

I'm currently in the process of building c4b964c the parent patch to 6300dd1 tho my hunch is that this issue is unrelated to the edp specifically as heads has no issues. The next commit I will try is 5c7148f which I think is the version I had before updating.

In the meantime trying a few different things as it builds. I installed the latest Arch Linux and the system is now acting slightly differently leaving the mute lights. The battery indicator has remained on no matter what OS is installed which I just noticed.

tlaurion, 2024-02-04:
> I'd agree with @pcm720 on the outstanding comments preventing getting it merged upstream. The VCC3P should be a separate future patch but shouldn't prevent the current patch from being merged.
> 
> I'm currently in the process of building c4b964c the parent patch to 6300dd1 tho my hunch is that this issue is unrelated to the edp specifically as heads has no issues. The next commit I will try is 5c7148f which I think is the version I had before updating.
> 
> In the meantime trying a few different things as it builds. I installed the latest Arch Linux and the system is now acting slightly differently leaving the mute lights. The battery indicator has remained on no matter what OS is installed which I just noticed. -- Didn't end up getting anywhere with this.
> 
> Just had a bad flash by my own doing and don't have access to tools to externally flash atm so have to put further investigations on hold.

@lethedata if you keep Linux logs in your os, you can grep journalctl logs for Heads coreboot/linux and find past version used.

What I meant prior with my commit traces was to show that the only changes there were bumping coreboot 4.13 to 4.19 with edp patch having changed a little between those. One would have to change that patch against 4.19 or 4.22 (master) to be accepted upstream, pushing the idea that any other mod requiring soldering or anything not covered in current patchset will need to be done subsequently. My understanding of the patchset not getting merged is because of the scope bing too large, and the kits covered not being enough specific in the commit messages.

As @pcm720 said prior, he cannot reproduce the reboot issue. I am not sure why OS does what it needs to do to reboot while the power lines are kept alive in your machine. To be honest I didn't understand your comment @lethedata stating that the mute button stays in some undesired state either; all of which behaviors are driven by EC firmware+hosy kernel, not coreboot nor the patch and where Heads job stops when the OS is kexec'ed so I have no insight here but pushing back to: what EC version and what is the Edp kit used to obtain a different and non-reproducible behavior just between you too. That is why the patchset is not yet merged uostream. At least that is my understanding and seems to be the understanding of coreboot devs being reviewers not having edp mod, and just like me, being a bit too confused about this since 2018 to make things go forward.

lethedata, 2024-02-04:
> @lethedata if you keep Linux logs in your os, you can grep journalctl logs for Heads coreboot/linux and find past version used.
> 

I wiped the drive to install something new before I realized there was an issue unfortunately. Noted for next time tho.

> What I meant prior with my commit traces was to show that the only changes there were bumping coreboot 4.13 to 4.19 with edp patch having changed a little between those. One would have to change [...]

That makes sense, to many boards floating around there's not enough info to tell what works and what doesn't. I'd say there is also a knowledge gap with the people running the mods. Get it running and just leave it making keeping things up to date and tested much harder. 

> As @pcm720 said prior, he cannot reproduce the reboot issue. I am not sure why OS does what it needs to do to reboot while the power lines are kept alive in your machine. To be honest I didn't understand your comment @lethedata stating that the mute button stays in some undesired state either; all of which behaviors are driven by EC firmware+OS kernel, not coreboot nor the patch and where Heads job stops when the OS is kexec'ed so I have no insight here but pushing back to: what EC version and what is the Edp kit used to obtain a different and non-reproducible behavior just between you too. As you noted, Heads can poweroff/reboot (which uses sysrq magic but at the end, all the same).
> 

The comment on the mute buttons and such are unrelated, its just what I'm seeing that shows that the system is keeping power for whatever reason.

Hopefully I'll have time later this week to tear the device open, re-flash externally, and get more information about what Edp kit is installed. It was already modded when I received it so didn't really look to deeply into things.

tlaurion, 2024-02-04:
@lethedata

> It was already modded when I received it so didn't really look to deeply into things.

I think this is exactly why the patch is not merged: the fear of causing more harm then good. If you flash this board config with a different kit, you get a brick. Coreboot won't merge this until this is clarified. I hope we have an understanding that this is upstream work needing to happen where Heads cannot replace coreboot discussions here. Heads is a mere user of coreboot patchsets, which this one was picked at a time where it was thought to be merged in coreboot 4.14 which never happened. I reworked and interacted with coreboot before 4.20 release, hoping this would be the last time this patchset was going to be maintained by me. And then I attempted again before 4.22 release and gave up.

**This is a lot of time and as said in other tickets, if this patchset doesn't get merged into coreboot, I will bump coreboot version soon to 4.22 and edp variant will go unsupported under Heads until situation is fixed upstream. **

pcm720, 2024-02-04:
@tlaurion 
> 
> I think this is exactly why the patch is not merged: the fear of causing more harm then good. If you flash this board config with a different kit, you get a brick. 

I still don't believe that this patch can cause a brick.  
Like I mentioned in my previous comment, it doesn't do anything except disabling internal display and changing the external port on dock connector as internally connected.

> Coreboot won't merge this until this is clarified. I hope we have an understanding that this is upstream work needing to happen where Heads cannot replace coreboot discussions here.

Let's work on upstreaming it together then.

While I'm interested in getting this patch merged, I don't have any eDP mods other than eDP 4.0 and I'm not willing to get another X230 and spend ~120 USD to mod it.  
How long is the merge window before 4.22 is released? Since this mod is pretty niche, I doubt we'll be able to collect enough information from other users, so we'll have to work with what we already have.

Apart from eDP 4.0 mod, we know at least two users with nitrocaster's mod that reportedly have no issues running Heads on their X230s. It doesn't support anything higher than 1920x1080 due to how works (it connects only two DisplayPort lanes instead of four).  
All of these reports are pretty old though...  

We also know that a.gain mod works with the patch as well since Xue Yao (XY Tech) has been selling pre-modded X230 and [recommended this patch](https://www.xyte.ch/support/x330-support/x330-bios/) to customers.

So that's at least three mods working with the patch we can mention in the merge request.  
If we need any more details on mods, I guess we'll need to ping anyone else we know that has this mod installed and ask them to chip in.

tlaurion, 2024-02-05:
> How long is the merge window before 4.22 is released?

4.22 is released. Meaning if this patch, rebased on coreboot master/4.22 works then heads can use 4.22 under modules/coreboot which is endgoal here.

pcm720, 2024-02-05:
> 4.22 is released. Meaning if this patch, rebased on coreboot master/4.22 works then heads can use 4.22 under modules/coreboot which is endgoal here.

Well, okay then, I guess I overlooked it.  
I also see that you've already updated the patchset for 4.22 around three months ago. So how can we test if this patch works?  
There was some activity back in November (#1524), but it got nowhere because CI build was failing for some reason.  
Are there any updates on this?

computer-user123, 2024-02-05:
> I'll try rebuilding clean on top of master after updating that PR real
> quick.

@tlaurion - I will test anything we need to test, especially pertaining to
x230-FHD-eDP (nitro caster)
If it’s not too much trouble for you, please send the url to circle ci
artifact, and I will download and test tonight !
Thanks man!
Appreciate your hard work and this project a lot!

tlaurion, 2024-02-05:
@pcm720 @computer-user123 build passing all boards to 4.22.01 coreboot release + edp/fhd patch in as per last update upstream at https://app.circleci.com/pipelines/github/tlaurion/heads/2268/workflows/4a50ca50-0cba-4ddd-9599-f590e66be755

pcm720, 2024-02-05:
> build passing all boards to 4.22.01 coreboot release + edp/fhd patch in as per last update upstream at https://app.circleci.com/pipelines/github/tlaurion/heads/2268/workflows/4a50ca50-0cba-4ddd-9599-f590e66be755 which build https://github.com/linuxboot/heads/pull/1604 PR

Works perfectly. No issues with rebooting either.  
I'll add a comment in coreboot's Gerrit.

> You must have things unclean under ./install for which flashrom attempts to link against

Did a fresh clone, still the same error. Not sure why, I'll try building on Debian 11 next since that's what I used to build Heads before. Had to reinstall the OS on my Heads VM after hardware change because it started crashing for no reason.

computer-user123, 2024-02-10:
Tested

   -
   build/x86/x230-maximized-fhd_edp/heads-x230-maximized-fhd_edp-v0.2.0-2023-g1a6aacb.rom
   <https://output.circle-artifacts.com/output/job/303870e5-aad2-4012-96c7-77870f5ce3c7/artifacts/0/build/x86/x230-maximized-fhd_edp/heads-x230-maximized-fhd_edp-v0.2.0-2023-g1a6aacb.rom>

…
1-installed qubes R4.2.0
2-boot and own TPM
3-test totp-works.
<x230 3320m 8gb>

Works fine..  x230 w/nitrocaster 1080p .
Thx @Tlaurion

lethedata, 2024-02-13:
Doesn't look like this issue is related to the eDP patches. 6300dd1 worked without issue and trying the latest non-fhd version on the modded system had issues.

Next thing I'm going to try is going month by month starting from 1f39d16 as it was the earliest build I have tested so far that failed.

As a side note, I'm like 95% sure this system is one of XY Tech's as it's an extremely clean install with the mod covered by the black protecting film, had to cut a corner to get the board version number. It also has all the additional adders he mentions on his website suck has the mSATA and internal USB.

tlaurion, 2024-02-13:
One thing heads did along the commits you pointed out before is that on 4.13, edp was enabled on top of Heads packing i915, and not enabling libgfx for bootsplash if I recall well. 

On 4.19, patches for libgfx exposing graphics through efifb replaced i915/i915 drm. That was not merged under coreboot at that time. Heads pushed for this and tested it and generalized it's usage for Intel based iGPU.

Coreboot 4.20+ had libgfxinit trampoline to expose iGPU through efifb. Unfortunately, Heads got stuck to 4.19 because of edp/fhd.

Recent tests showed that 4.21 broke cbmem for measured boot. The history depicted here shows that Heads uses codepaths under coreboot that nobody else uses and therefore, the patch needs to be merged upstream for Heads to bump to newer coreboot versions, at least testing releases as well otherwise things bitrot.

@lethedata i would keep this issue open and would suggest pointing to it while opening an upstream issue to track this correctly. The more traction upstream the more chances this is functional all together.

There is interest from coreboot to permit to build Heads as payload. But we are not there yet. Meanwhile, Heads needs to thoroughly test coreboot releases otherwise more work is needed later on to fix broken things. On my side I learned my lesson: if things are not merged upstream, boards having unmerged upstream patches on next release will move to UNMAINTAINED_ boards variants. I will do that for linuxboot boards in tree as well, since they are more then untested: they don't build and low interest in them led to their current state until that changes as well.

tlaurion, 2024-02-18:
@lethedata edp/fhd was merged upstream for what will be coreboot 4.23 release. As of now cbmem TPM event log is broken but fix is not required (tpmr under heads is able to replay measurements and sealing/unsealing operations work as expected if we switch to TCPA event log as opposed to coreboot custom event log format as we used to before since new format is now available.)

Tldr: we will be able to use edp/fhd under coreboot 4.23. Tendency is to bump to 4.22 as of now so patch for edp/fhd might stay under heads until we switch to coreboot 4.24 release. TBD

lethedata, 2024-02-18:
Awesome and sounds good!

> @lethedata i would keep this issue open and would suggest pointing to it while opening an upstream issue to track this correctly until resolution. The more traction upstream the more chances this is functional all together.

Yup, going to be working with upstream to get it sorted out. I can still test heads if needed but it's probably better not to until I get this resolved.

tlaurion, 2024-02-19:
> * [6300dd1](https://github.com/linuxboot/heads/commit/6300dd178a35cbe68bc6587fdf48a210f1437607) : Functional

@lethedata I'm confused on the above. Any insights here?
The "functional" patch was https://github.com/linuxboot/heads/blob/6300dd178a35cbe68bc6587fdf48a210f1437607/patches/coreboot-4.19/0001-x230-fhd-variant.patch

Upstream merged fhd/edp and is now under https://github.com/linuxboot/heads/pull/1604 which I consider ready for review
which now looks like https://github.com/linuxboot/heads/blob/6b61f61e6e209192ebf0b9a413744f7daaab06e8/patches/coreboot-4.22.01/0001-x230-fhd-variant.patch

tlaurion, 2024-03-24:
@lethedata tracking of discussions upstream?

PR #1604 (2024-03-26): 4.22.01 fhd patch merged upstream (4.24) + bump all 4.19 boards to 4.22.01
Revisiting 4.22 coreboot bump with fhd patch (superseeds and replaces #1524), Might fix #1603)

    - patches/coreboot-4.22.01/0001-x230-fhd-variant.patch created per
      - git fetch https://review.coreboot.org/coreboot refs/changes/50/28950/27 && git format-patch -1 --stdout FETCH_HEAD > ~/heads/patches/coreboot-4.22.01/0001-x230-fhd-variant.patch
    - all boards configs bumped with:
      - sed 's/4.19/4.22.01/g' boards/*/*.config -i
      - grep -Rn 4.22 boards/ | awk -F "/" {'print $2'}| while read line; do make  [... cut]
closed 2024-03-26 by tlaurion