linuxboot/heads #1605: Manjaro and Garuda won't boot automatically

JonathonHall-Purism, 2024-02-13:
Heads can't boot Manjaro or Garuda ISOs.  Both of these use variables in the GRUB configuration that Heads doesn't understand.

In this menu item, Heads would need to know the content of "$title" and that it will become "$2" in the menu entry.  (This becomes part of the kernel file name and some important command line parameters.  `$kopt` is not critical, it boots without it.)

```
    title=""
    for kopt in x86_64 $kopts  misobasedir=garuda root=miso:LABEL=GARUDA_DR460NIZED_RAPTOR quiet systemd.show_status=1 ibt=off; do
        if [ -n "$title" ] ; then
            title="$title $kopt";
        else
            title="$kopt";
        fi;
    done
    menuentry "Boot with open source drivers" --class=Garuda.x86_64 "$title" {# set arguments above with the editor
        linux /boot/vmlinuz-$2 driver=free nouveau.modeset=1 i915.modeset=1 radeon.modeset=1
        initrd /boot/amd_ucode.img /boot/intel_ucode.img /boot/initramfs-x86_64.img
    }
```

JonathonHall-Purism, 2024-02-13:
Handling the `$2` seems possible if `$title` wasn't broken in the first place, but handling the loop that sets `$title` isn't trivial.  In this case it would work if executed in a shell, but I'm not sure if that's always true.

This was reported on the Librem 11 and it's possible to boot these manually from the recovery shell as a workaround: https://forums.puri.sm/t/librem-11-seabios-coreboot-does-not-recognize-keyboard-pureboot-does-not-boot-usb/22433/19

tlaurion, 2024-02-13:
Non-directly related #1499

tlaurion, 2026-08-16:
@JonathonHall-Purism still current past https://github.com/linuxboot/heads/pull/2186 ?