linuxboot/heads #1607: W541 dGPU unable to boot Qubes

TrustExecutor, 2024-02-13:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

I have the W540 and use the w541-hotp-maximized
The model# is 20BH-A01YMS

**2. Does your computer have a dGPU or is it iGPU-only?**
- [X] dGPU
- [ ] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [X] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [ ] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [X] Nitrokey 3A NFC

**5. Are you using the PGP key to provide HOTP verification?**
- [X] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ ] First-time flash
- [X] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [X] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [ ] External flashing

**3. How was Heads initially flashed**
- [X] External flashing
- [ ] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [X] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [X] Yes
- [ ] No
- [ ] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [ ] I downloaded it
- [X] I built it

**3. If you built your rom, which repository:branch did you use?**
- [X] Heads:Master
- [ ] Other (please identify)

**4. What version of coreboot did you use in building?**
- [X] 4.8.1 (current default in heads:master)  (I use master, but coreboot config says 4.19 ?)
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] Other (please specify)
- [ ] I don't know

**5. In building the rom where did you get the blobs?**
- [X] No blobs required (downloaded during build)
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [X] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

## Please describe the problem

See PR: https://github.com/linuxboot/heads/pull/1606

**Describe the bug**
I built setting `CONFIG_USE_OPTION_TABLE=y` in the coreboot config file and then used `nvramtool -w enable_dual_graphics=Enable` and the card showed up in lspci with Debian 12 without any vbios blobs. But Qubes get a kernel panic at boot. If I set it to disable then Qubes boot normally.

I inspected the backup rom using UEFITool and found two Intel VGA bios with PCI ID 8086,0406 and a bunch of Nvidia VGA bios.
The Nvidia VGA bios I found two that had the same PCI  ID (10de,11fc) as printed in lspci using stock firmware. Intel stock PCI ID were 8086,0416 with the stock firmware.

I then created another board and included vbios binaries for both iGPU and dGPU in the coreboot config.

I also tried to change the script in the w530/t530 board targets, that download and extract vbios blobs from a Lenovo update .exe using `VbiosFinder`, `innoextract`, and `rom-parser`. 
See the script in the PR. The script works and extracts two verified blobs. The hashes of the output files differ from the ones I extracted from the backup rom with UEFITool.

The blobs has the generic 8086,0406 PCI ID. i have tried using `rom-fixer` to change PCI ID to 8086,0416 as `lspci` on Debian reports, without luck.

When

tlaurion, 2024-02-16:
@TrustExecutor 

Reworked your branch at https://github.com/tlaurion/heads/tree/w541-dgpu so that CI provides artifacts to download and test so we remove possibilities of host tools issues.

Also see comment on your PR https://github.com/linuxboot/heads/pull/1606#issuecomment-1946627492

Direct download links for CI produced rom images for commit 20df526 and related changes https://github.com/linuxboot/heads/compare/master...tlaurion:heads:w541-dgpu : 

- https://output.circle-artifacts.com/output/job/26b4ee44-4463-45e5-a802-89efed482a91/artifacts/0/build/x86/UNTESTED_w541-dgpu-k2100m-maximized/heads-UNTESTED_w541-dgpu-k2100m-maximized-v0.2.0-2026-g20df526.zip
- https://output.circle-artifacts.com/output/job/b2d54a08-3774-4c2e-9938-b6d0b3841d5a/artifacts/0/build/x86/UNTESTED_w541-dgpu-k2100m-hotp-maximized/heads-UNTESTED_w541-dgpu-k2100m-hotp-maximized-v0.2.0-2026-g20df526.zip

tlaurion, 2024-02-16:
Just stumbled upon https://doc.coreboot.org/northbridge/intel/haswell/known-issues.html#known-issues-with-haswell

Seems like https://review.coreboot.org/c/30456 might be needed?

tlaurion, 2024-02-16:
Just stumbled upon https://doc.coreboot.org/northbridge/intel/haswell/known-issues.html#known-issues-with-haswell

Seems like https://review.coreboot.org/c/30456 might be needed? 

Edit: or ditch mrc binaries altogether with NRI

TrustExecutor, 2024-02-18:
@tlaurion Thanks for cleaning up the branch and for your input.
What is really interesting is that Debian 12 lists both the iGPU and the dGPU in ´lspci´ with no vbios blobs and libgfxinit (booted from heads). This is after setting `nvramtool -w enable_dual_graphics=Enable`
It is only in Qubes this setting makes the system non-bootable.

This makes me think it may instead be a noveau driver issue in Qubes and not something mrc/coreboot related.
I also found a youtube video where the same setup is made and he got the same result in debian where both GPUs show up.
https://youtu.be/abDv1SAGNSA?t=135

TrustExecutor, 2024-02-18:
Here is also some information about running the dGPU under coreboot on the T440p, which is also Haswell.
https://crayphish.github.io/posts/t440p-adventures/

tlaurion, 2024-02-18:
> Here is also some information about running the dGPU under coreboot on the T440p, which is also Haswell.
> https://crayphish.github.io/posts/t440p-adventures/

@TrustExecutor : 

>The current coreboot code for the T440p has the PCI Express Graphics (PEG) device that handles the dGPU disabled. Thanks to nullenvk, you can restore dGPU functions by applying this commit, which adds the “Enable dedicated graphics (experimental)” option to nconfig. Enabling this will re-enable the PEG device in your next coreboot build. You need to include the VBIOS for both the iGPU and dGPU. You can extract both using VBiosFinder. If you’re building coreboot from master branch, the current SeaBIOS payload will result in a bootloop. Use SeaBIOS 1.15 or TianoCore (UEFI OS only).

Hmm. Well, I could add the patch to coreboot for you to test outcome rom if you commit to testing? Unfortunately, that means that if patch doesn't apply directly, Heads would need to maintain that patch across coreboot version bump which would be a burden on myself and on which I cannot commit from the moment the patch doesn't apply directly. A similar story happened for fhd/edp which finally got merged under coreboot, where this patch referred here in thread and in this article is abandoned and would need input from interested board owners and committed testers. Tldr: Heads intends to be a user of coreboot, not a maintainer of it.

@TrustExecutor can you commit in testing and bridging the gap upstream by commenting on the coreboot patch to try to upstream it by at least making the bridge between that PR to be created and the CB patchset upstream? If yes I can try to adapt the patchset here and have you comment on it once it builds. Note that the rom extracter is using obsolete gems and also needs to be worked on in upstream project, otherwise Heads also commits in maintaining those tools and that is not planned. Tldr on that: it won't work on debian-12 nor nix buildsystem that heads plans to switch to in the mid-term.

tlaurion, 2024-07-02:
NRI was revived in June 2024 https://github.com/linuxboot/heads/issues/1711

tlaurion, 2024-07-02:
@TrustExecutor : don't hesitate pinging me here with your testing progress for dgpu+nri progress, you would have my attention.

tlaurion, 2026-09-17:
@optimist555 same user as @TrustExecutor? Still the case? if such, please reopen (2024 is really old)

closed 2026-09-17 by tlaurion