linuxboot/heads #1608: cbmem broken under coreboot 4.21+

tlaurion, 2024-02-13:
> @JonathonHall-Purism 

>Well, TBH I'm not sure a lot of other use so much measured boot outside of Heads @JonathonHall-Purism ?
This shows its not tested a lot, or as you said, since cbmem is built by musl-cross-make related doubts, which is compiler buillding everything tools.cpio related under Heads, including cbmem here), that might as well be an undesired artifact of those? (modified quote)

> I took the time to use rom produced here (4.22.01 tag release codebase built with measured boot and coreboot custom event log format) and test cbmem built on top of ubuntu 23.10/debian trixie OS (to remove musl-cross-make doubts).

> Interestingly enough:
> - `sudo ~/Downloada/coreboot-4.22.01/util/cbmem/cbmem -L` :  `Unknown TPM log specification:` : FAIL
> - `sudo ~/Downloada/coreboot-4.19/util/cbmem/cbmem -L` : `coreboot TPM log: [...]` : OK

> So now, let's test cbmem behavior in between those release versions:
> - `sudo ~/Downloada/coreboot-4.21/util/cbmem/cbmem -L` :  `Unknown TPM log specification:` : FAIL
> - `sudo ~/Downloada/coreboot-4.20.1/util/cbmem/cbmem -L` : `coreboot TPM log: [...]` : OK

> **So `cbmem -L` broke somewhere between 4.20.1 and 4.21.**

> ---
> Raised awareness under #coreboot channel at https://matrix.to/#/!EhaGFZyYcbyhdSgStq:matrix.org/$DJW_mF5d1PfhJIvoM-_HvX8s3BpzNvtNu77auhvKBo0?via=matrix.org&via=sibnsk.net&via=datanauten.de. @miczyg1 said he would take a look at https://matrix.to/#/!EhaGFZyYcbyhdSgStq:matrix.org/$34SHe96Ef5Qrv2pIXtmi2lqpSwnh1GsKXlZCR9I7PpY?via=matrix.org&via=sibnsk.net&via=datanauten.de

_Originally posted by @tlaurion in https://github.com/linuxboot/heads/issues/1604#issuecomment-1942087004_

miczyg1, 2024-02-14:
Issued a PR that should address the issue: https://github.com/linuxboot/heads/pull/1609

Why `cbmem -L` does not work with the standard TPM log per specifications is a separate matter (I suspect the first event in the event log is not SpecID event, but why? not sure yet).

tlaurion, 2024-02-14:
@miczyg1 @JonathonHall-Purism theoritically, tpm1.2 vs coreboot log format should not matter since from my understanding, they use PCR2 for log. 

For our interest here and where the bug was discovered (all ivy/sandy/haswell boards were based on 4.19 since edp/fhd patch was not merged upstream), boards stayed on 4.19. But since https://review.coreboot.org/c/coreboot/+/28950 got merged yesterday as https://github.com/coreboot/coreboot/commit/a88dd4b6fbac8748a6f4580ad6363da47f5b3bc2, I'll prepare a PR switching all boards to that commit and be able to give traces, switching boards to TCG TPM1.2 TPM Event log format and go from there.

tlaurion, 2024-02-15:
Logs provided and analysis at https://github.com/linuxboot/heads/pull/1609#issuecomment-1944461330

tlaurion, 2024-02-19:
Discussion at https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$CQDLxAckANKmXAu8ZDtcDIHUn1dgvgfzyifZidHr7SA?via=matrix.org&via=nitro.chat&via=fairydust.space

tlaurion, 2024-02-19:
Ok, so to trace this better

- https://github.com/tlaurion/heads/tree/422_fhd at commit https://github.com/tlaurion/heads/commit/4bb831720e323cb75c29a400d005561d4016e3ca used here to test on x230 (coreboot custom event log format)
  - https://output.circle-artifacts.com/output/job/cfdd3b68-2cd1-401f-8758-0ef2715e1cce/artifacts/0/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-2026-g4bb8317.zip
    -  cbmem -L "Unknown TPM log specification"
    - cbmem -1: Event log reported in coreboot logs fine
    - [cbmem_list_custom_coreboot_format.log](https://github.com/linuxboot/heads/files/14335605/cbmem_list_custom_coreboot_format.log)
    - [cbmem_54435041_coreboot_format.log](https://github.com/linuxboot/heads/files/14335602/cbmem_54435041_coreboot_format.log)

tlaurion, 2024-02-19:
Ok, so to trace this better

- https://github.com/tlaurion/heads/tree/422_fhd at commit https://github.com/tlaurion/heads/commit/6b61f61e6e209192ebf0b9a413744f7daaab06e8 used here to test on x230 (TCG event log format)
  - locally built x230-maximized from Talos II while CircleCI prepares equivalent rom (will update with replacement link, same commit)
    - cbmem -L:  good until log entry 12 (corrupted)
      - [cbmem_L_TCG_event_log.log](https://github.com/linuxboot/heads/files/14336053/cbmem_L_TCG_event_log.log)
    - cbmem -1: Event log reported in coreboot logs fine
    - [cbmem_list_TCG_format.log](https://github.com/linuxboot/heads/files/14336054/cbmem_list_TCG_format.log)
    - [cbmem_54445041_TCG_format.log](https://github.com/linuxboot/heads/files/14336071/cbmem_54445041_TCG_format.log)
    - [cbmem_434f4e53_TCG_console.log](https://github.com/linuxboot/heads/files/14336074/cbmem_434f4e53_TCG_console.log)

tlaurion, 2024-02-19:
> Issued a PR that should address the issue: #1609

@miczyg1 #1609 closed as explained under https://github.com/linuxboot/heads/issues/1608#issuecomment-1952968828 : switching to coreboot custom format doesn't fix anything at all.

tlaurion, 2024-02-22:
Blocker for #1568

tlaurion, 2024-04-22:
@miczyg1 ping?

tlaurion, 2024-08-15:
This is related to coreboot changes for TCPA->TPM event log

tlaurion, 2024-08-15:
Opened https://github.com/Dasharo/dasharo-issues/issues/1004

Closing here

miczyg1, 2024-10-30:
https://review.coreboot.org/c/coreboot/+/84926

Fixed the cbmem utility side, but it would also be good to clear that log area on creation.

miczyg1, 2024-10-30:
https://review.coreboot.org/c/coreboot/+/84927

Fixed the log area creation for TPM1.2. Not sure why but the TPM2.0 log area was completely cleared, thus the issue did not appear when parsing TPM2.0 logs...

PR #1824 (2024-10-30): patches/coreboot-*: fix tpm1 cbmem crash on 'cbmem -L'
fix #1608 for Dasharo, 24.02.01 and Purism coreboot forks

@JonathonHall-Purism please review
closed 2024-10-30 by tlaurion