linuxboot/heads #1615: Provide OS integration to sign /boot and/or root during updates

JonathonHall-Purism, 2024-02-20:
A common paint point in Heads is that when any changes are made, it is difficult to tell if the changes are intentional or due to tampering.

In my opinion, the best solution to this is to eliminate the question.  When intentional changes are made, sign them then as part of the update process.

It might be possible for Heads to include or offer integrations for common OSes.  E.g. we could build packages for Debian-based, Fedora-based, etc. distributions that would know how to re-sign /boot after an update occurs.  Potentially, this could be extended to root also for the root-hashing feature.

For Debian, it is possible to install apt/dpkg hooks that occur before or after dpkg invocations, package installation, etc. (one example: https://unix.stackexchange.com/questions/76294/hook-a-script-to-apt-get).  Assuming other package managers offer similar functionality, we could probably export a Heads script and hook it at the appropriate place to integrate signing into the update process.

Alternative to #1599

tlaurion, 2024-05-20:
@DemiMarie this is the only real solution to the problem until unified signed kernels that kexec could verify signature against. 

How would those hook script look like for qubesos dom0? Ideas on how to get there practically would be beneficial from the OS side.

DemiMarie, 2024-05-21:
@tlaurion what makes this more desirable than #1620?

One approach might be:

1. Qubes OS provides a directory containing scripts that are run during each kernel update.  The scripts are passed the kernel and initramfs as arguments.
2. A script provided by Heads signs the kernel & initramfs using a key in the TPM.  The key has policies that make it useable only when a verified kernel + initramfs combo has been booted.
3. Either the signature or the code being signed contains a version number.
4. The version number is compared against a TPM NV index, preventing downgrade attacks.

marmarek, 2024-12-20:
Posting my message from today discussion:

> I have an idea (feel free to ignore ;) ) to improve the message about /boot files changes - add a diagram to that error message, explaining which step went wrong, kinda like cloudflare error messages which show nice and easy where the problem is. It could be something like Nitrokey -> Heads -> boot config -> boot files (or "kernel" or "operating system"), and mark which verification failed. I guess it could be even ascii-art (with colors? if whiptail supports that) for simpler implementation / space saving. This should avoid confusion about "Nitrokey blinks green but Heads says something is wrong".

It's not a solution to the original proposal here (which would be much better), but it's a band-aid for the current situation.

tlaurion, 2024-12-20:
> Posting my message from today discussion:
>
>
> It's not a solution to the original proposal here (which would be much better), but it's a band-aid for the current situation.

Or @marmarek @nestire @daringer, if need is for hotp_verification to permit a red only toogle to fix the problem since UI to me is already pretty clear, maybe open an issue on hotp-verification side @marmarek ?

[Per matrix discussion under qubesos-public channel](https://matrix.to/#/!WtRrlYUTHOQjqGcSnn:invisiblethingslab.com/$e68K67VOQasvsV-1XHrWvk7AiVpAacpWvfcL86yR3ko?via=matrix.org&via=nitro.chat&via=envs.net), which happens all the time:

Not sure this fix would help if UX expects hotp security dongle led to flash led here

Image extracted from referred discussion:

[image]