linuxboot/heads #1617: [Purism 14] Feb. 29th messed up clock

Rot127, 2024-03-01:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

Purism 14 v1

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [x] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [x] Purism
- [ ] Other provider
- [ ] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [x] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [ ] Other

**5. Are you using the PGP key to provide HOTP verification?**
- [x] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [x] First-time flash
- [ ] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [ ] External flashing

**3. How was Heads initially flashed**
- [ ] External flashing
- [ ] Internal-only / 1vyrain
- [x] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [ ] Maximized
- [ ] Non-maximized / legacy
- [x] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [ ] Yes
- [ ] No
- [x] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [ ] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [ ] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*

**3. If you built your rom, which repository:branch did you use?**
- [ ] Heads:Master
- [x] Purism Firmware 24

**4. What version of coreboot did you use in building?**
- [ ] 4.8.1 (current default in heads:master)
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [x] 4.18 Purism-1
- [ ] I don't know

**5. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [x] I don't know

## Please describe the problem

**Describe the bug**

On February the 29th the date is in the future (either in the year 2070 or 2069).
This happened the first time when the laptop came out of standby!
So I suspect it is a problem with the hardware clock of the Purism 14 or Coreboot? Since Heads was no longer running.

This time problem messes of course with pretty much everything.
The problem persisted over reboots.

Setting it to the correct date via `date -s "<date time>" && hwclock -w` does not survive reboots.

I changed the CMOS battery, so this is not the problem.
Also everything works magically today again.

**To Reproduce**

A potential test could be to set the clock to Feb. 29th in the recovery shell.
Unfortunately, don't have time to test it though.

**Expected behavior**

Either the clock in the laptop or Heads/Coreboot can handle the gap year.

**Screenshots**
none

**Additional context**
Add any other context about the problem here.

daringer, 2024-03-01:
Yes, this seemingly was due to a leap-day/year bug, most [likely in coreboot](https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/57F4LNJQ6MBRT5VAJAKWHU4QHW5EFPVN/) - the temporary workaround was like you already tried it, but with the date being set one day into the future:
```
date -s 2024-03-01
hwclock -w
reboot
```
That's also the reason why this is working today

Rot127, 2024-03-17:
Since it was Coreboot I close this.

closed 2024-03-17 by Rot127