linuxboot/heads #1618: Replace GnuPG with Sequoia

DemiMarie, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
GnuPG is a large amount of legacy C code that operates on untrusted input.

**Describe the solution you'd like**
Use Sequoia instead.  Only signature verification is needed.

**Describe alternatives you've considered**
Use a different tool for verifying signatures, such as signify or ssh-keygen.

**Additional context**
GnuPG has known bugs and will decompress data in the signature, creating extra attack surface.

tlaurion, 2024-03-27:
@DemiMarie 

> Describe alternatives you've considered
> Use a different tool for verifying signatures, such as signify or ssh-keygen.

Neither support smartcard, don't they?

> Additional context
> GnuPG has known bugs and will decompress data in the signature, creating extra attack surface.

https://github.com/linuxboot/heads/blob/05289c0989b4ded7accd197be53b123b5467d959/modules/gpg2#L29-L49

Not aware of any decompression being possible in currently configured/compiled gpg2. 
Did I missed something you found/tested?

> Is your feature request related to a problem? Please describe.
> GnuPG is a large amount of legacy C code that operates on untrusted input.

Where/How is it used under GUI ops under Heads?
Heads uses gpgv (wrapper for verify only ops) on daily ops, and uses gpg detache-sign calls otherwise upon request from user, which is followed by a reboot and then a clean state again. Did I miss something?

-----
### Discussion

- Is sequoia smartcard support production ready now?
- Is the firmware footprint advantageably lower then gpg toolstack today? Can most of the features be deactivates as they are for gpg toolstack? What is the size comparison of the toolstacks today?

tlaurion, 2024-04-16:
> Is sequoia smartcard support production ready now?

It is though smartcard cradle.
That would be a Big refactoring. If footprint is not out of scope.