linuxboot/heads #1619: Set vm.panic_on_oom=1 in bootloader

DemiMarie, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
Shell scripts are not very good at handling unexpected errors.  The most likely cause of an unexpected error during boot is running out of memory, which should never happen.

**Describe the solution you'd like**
Panic the system if an unexpected out of memory happens during boot.

**Describe alternatives you've considered**
Fix all of the missing error checks.  I expect this to be difficult.

**Additional context**
Out-of-memory can cause any non-builtin command to fail without crashing the script, and shell scripts usually are not written to handle this.

JonathonHall-Purism, 2024-03-14:
Agree 100%.  I'm planning to jump on a bunch of PB/Heads work in a few weeks, if nobody beats me to it I'll test it.

tlaurion, 2024-03-27:
Understood that desired implementation here is to add `vm.panic_on_oom=1` in all coreboot config's kernel command line.

PR #1626 (2024-04-04): Enable panic_on_oom and requirements checks/warnings
Closes  #1619

- `CONFIG_SYSCTL` and `CONFIG_PROC_SYSCTL` added under all linux configs
- `CONFIG_TMPFS` removed from all linux configs (to allign with purism config. Other configs need to be alligned across board configs as we already know which is #1628)
- Applies panic_on_oom through procfs if CONFIG_PROC_SYSCTL not disabled, otherwise warns during early boot

Output in case config missing at https://github.com/linuxboot/heads/pull/1626#issuecomment-2030545581

-----

OLD: no, we cannot apply this directly through coreboot-> linux(heads p [... cut]
closed 2024-04-04 by JonathonHall-Purism