linuxboot/heads #1620: dm-verity signed `/boot`

DemiMarie, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
Verifying signatures on individual files in `/boot` is error-prone and slow.

**Describe the solution you'd like**
Use dm-verity and multiple possible `/boot` partitions.  This allows atomic updates of `/boot` by replacing the entire partition.  Heads will automatically choose the newest version.

**Describe alternatives you've considered**
#1599

**Additional context**
dm-verity is used by most other verified boot schemes.

tlaurion, 2024-03-27:
@DemiMarie 

Community is pushing the other way around, wanting more output on files being verified as #1369 
As per past discussions, the ideal here would be for OSes to sign their kernel/initrd/UKI, and once there is progress there, have Heads validate those as an additional security mechanism first, then switch to that.

Recap on what Heads does, here and was improved in the past years in terms of boot integrity validation/authentication
- tree of /boot dirs/files tree to list existing dirs/files and be able to discover (to diff new files which otehrwise cannot be verified against only with hashes): /boot/kexec_tree.txt
- hash of /boot files that are checked against prior of booting default boot option and warns user if changed: /boot/kexec_hashes.txt
- default boot params part of trusted kexec call per default signed op: /boot/kexec_default.1.txt
- default boot files hashes per detached signed op: /boot/kexec_default_hashes.txt
- default boot+TPM DUK setup created known LUKS devices: /boot/kexec_key_devices.txt
- TPM2 primary handle hash per TPM ownership/last TOTP/HOTP sealed op: /boot/kexec_primhdl_hash.txt
- default+TPM DUK setup initrd crypptab override (to pass secret.key to OS) : /boot/kexec_initrd_crypptab_overrides.txt
- detached signed signature of everything `/boot/kexec_*.txt` verified against each default boot: /boot/kexec.sig

---

And then as described under qubes forum: https://forum.qubes-os.org/t/verifying-installation/11739/4.
From dom0:
``sha256sum $(find /boot/kexec*.txt) | gpg --verify /boot/kexec.sig - && cd /boot && sudo sha256sum -c /boot/kexec_hashes.txt``

Output without dom0 having public key:
```
(2)$ sha256sum $(find /boot/kexec*.txt) | gpg --verify /boot/kexec.sig -
gpg: Signature made Tue 26 Mar 2024 04:03:48 PM EDT
gpg:                using RSA key 575F80D1599EA6D2C70AA9A19A53E1BB3FF00461
gpg: Can't check signature: No public key
```

----
Q: What would be the advantages of your proposed solution as opposed to #1599 or #1615 ?
#1615 is the real solution, but would require QubesOS to attack the problem and not defer to Fedora and/or to contribute fix to Fedora. I looked into UKI as also implied under https://github.com/linuxboot/heads/issues/1550#issue-2036902092 but no distro is actually enforcing it today, deferring to the end user the task of signing UKI, which would also better fix this issue then proposed solution. 

I would also defer the problem to having /boot encrypted under LUKS but then again, using dm-verity would come with a lot of other issues since opening /boot is not even always done in R/O and touching FS metadata will also make dm-verity fail. So what is the added value for the end user at this point?

> dm-verity is used by most other verified boot schemes
Who does it? Where can I "most other verified boot schemes" using dm-verity?

QubesOS still need to seperate dom0 rootfs writing, which AFAIK depends on 
- Externalizing logging to sys-log (https://forum.qubes-os.org/t/working-guide-for-creating-a-logging-qube-i-e-sys-log-wip/6587/2)
- Having dom0 stop writing to rootfs, see https://github.com/QubesOS/qubes-issues/issues/4371#issuecomment-761863958
- Switch from fedora to silverblue for dom0 layer?

DemiMarie, 2024-03-27:
> Who does it? Where can I "most other verified boot schemes" using dm-verity?

dm-verity is used by Android, ChromeOS, and I believe many embedded systems.  Apple uses something similar (Sealed System Image) in macOS and iOS.

That said, if you think this should be closed in favor of UKI (which is what I think Qubes OS is moving to for secure boot support) and automatic signing on updates, that’s fine.

tlaurion, 2024-03-27:
> > Who does it? Where can I "most other verified boot schemes" using dm-verity?
> 
> dm-verity is used by Android, ChromeOS, and I believe many embedded systems.  

Android also seperates SYSTEM partition from config and deploys this with things embedded in firmware with VBOOT to enforce this. Doing so, they also permit to "Factory Reset" user partition, which goes along the lines of my suggestion of going Silverblue way. 

> Apple uses something similar (Sealed System Image) in macOS and iOS.

Have not followed this, but pretty sure the idea is for OS to be closely embedded in firmware doing a similar thing than Android above. 

> 
> That said, if you think this should be closed in favor of UKI (which is what I think Qubes OS is moving to for secure boot support) and automatic signing on updates, that’s fine.

It's not important what I think here outside of implementing something here that doesn't lock down users of their system because the OS doesn't comply with what is attempted in the bootloader. 

As said above, as long as OSes do not implement something that can be leveraged and enforced in firmware, I just wanted to restate once again that those discussions should happen upstream and be referenced downstream (here). Heads tries to implement what is readily available and enforceable, today. 

None of the suggested solutions, outside of the rejected git repo under /boot, can actually tell what changed, and even less say if those changes are as prescribed by the OS.

As opposed to Android or Apple perspectives referred here, which controls not only the firmware but OS states, implementing dm-verity as suggested would be a regression, not an improvement over current state.

tlaurion, 2024-04-23:
Original discussion https://groups.google.com/g/qubes-devel/c/hG93VcwWtRY

tlaurion, 2024-05-02:
Related https://github.com/QubesOS/qubes-issues/issues/9146

marmarek, 2024-05-25:
I'm not sure about other OSes supported by Heads, but currently there are no plans for having separate /boot (dm-verity protected or not) partitions for different kernel versions in Qubes OS. So, unless some other OS used with Heads would support such feature, IMO having this in Heads makes no sense.

DemiMarie, 2024-05-25:
I’ll close this issue, then.  It’s ultimately a hack anyway, and UKIs (with a hardened FAT filesystem parser) would be a much better choice.

closed 2024-05-25 by DemiMarie