linuxboot/heads #1624: Problem with flashing T430

heartbl3ed, 2024-03-23:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**
T430 hotp maximized

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [ ] iGPU-only
- [X] I'm not sure, from quick look in bios, i think there's a mention of dGPU.

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [X] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [ ] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [X] Nitrokey 3

**5. Are you using the PGP key to provide HOTP verification?**
- [X] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [X] First-time flash
- [ ] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ] Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [ ] External flashing

**3. How was Heads initially flashed**
- [X] External flashing
- [ ] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [X] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [ ] Yes
- [ ] No
- [X] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [X] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [X] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*

**3. If you built your rom, which repository:branch did you use?**
- [X] Heads:Master
- [ ] Other (please identify)

**4. What version of coreboot did you use in building?**
- [X] 4.8.1 (current default in heads:master)
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] Other (please specify)
- [ ] I don't know

**5. In building the rom where did you get the blobs?**
- [X] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

## Please describe the problem
Greetings everyone,

This is my second time flashing heads rom, the first one with x230 that i dont have problem.
and recently i bought t430 and just doing flash without having backup top rom.

I can read both chip, also can write top but not with bottom chip.

i have try prebuilt rom rom from circle ci, it was from latest commit :

    UNTESTED_t430-legacy-flash
    Since there's no bottom rom, so i assume that it will work but nope, after powering up, just after 1 second, fan stopping, power led up, and its like the pc just shutting down.

    UNTESTED_t430-legacy
    Havent try it since it was 12MB, so i assume i cannot use this rom.

    UNTESTED_t430-hotp-legacy
    Havent try it since it was 12MB, so i assume i cannot use this rom.

    t430-hotp-maximized
    I can succesfully write top rom, but not with bottom rom, also have try using all the chip* but no success. (havent try to boot up but i assume flashing top only would likely fail?)

    t430-maximized
    I can succesfully write top rom, but not with bottom rom, also have try using all the chip* but no success. (havent try to boot up but i assume flashing top only would likely fail?)

Chip found for top is :
Found Macronix flash chip "MX25L3205(A)" (4096 kB, SPI) on ch341a_spi.
Found Macronix flash chip "MX25L3205D/MX25L3208D" (4096 kB, SPI) on ch341a_spi.
Found Macronix flash chip "MX25L3206E/MX25L3208E" (4096 kB, SPI) on ch341a_spi.
Found Macronix flash chip "MX25L3233F/MX25L3273E" (4096 kB, SPI) on ch341a_spi.

And i was using MX25L3206E/MX25L3208E for all rom flash.

Chip found for bottom is :
Found Macronix flash chip "MX25L6405" (8192 kB, SPI) on ch341a_spi.
Found Macronix flash chip "MX25L6405D" (8192 kB, SPI) on ch341a_spi.
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on ch341a_spi.
Found Macronix flash chip "MX25L6436E/MX25L6445E/MX25L6465E/MX25L6473E/MX25L6473F" (8192 kB, SPI) on ch341a_spi.

There's a problem with this, sometimes it was found this chip, and sometime it will changed, like top chip, its weird, sometime it not found, so i suspect the flashing fail its because the bottom chip problem ? I can assure the ch341 is fine, because this was i used to flash x230 too. for about 100 times reading bottom chip 70% is goes to above chip and the rest is this chip:

Found Unknown flash chip "SFDP-capable chip" (8192 kB, SPI) on ch341a_spi.
Found Macronix flash chip "MX25L4005(A/C)/MX25L4006E" (512 kB, SPI) on ch341a_spi.

I use all the chip available but no success (everything that comes with MX25L64*).

```
root@debian:/home/user/Downloads# flashrom --programmer ch341a_spi -c MX25L6406E/MX25L6408E -w heads-t430-hotp-maximized-v0.2.0-2038-gebdcc85-bottom.rom 
flashrom unknown on Linux 6.1.0-18-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Macronix flash chip "MX25L6406E/MX25L6408E" (8192 kB, SPI) on ch341a_spi.
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
Reading old flash chip contents... done.
Erasing and writing flash chip... FAILED at 0x00000010! Expected=0xff, Found=0xfb, failed byte count from 0x00000000-0x00000fff: 0x8d
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x0000001b! Expected=0xff, Found=0xfb, failed byte count from 0x00000000-0x0000ffff: 0x9db
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x00000000! Expected=0xff, Found=0xbf, failed byte count from 0x00000000-0x0000ffff: 0x9f4
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x00000002! Expected=0xff, Found=0xbf, failed byte count from 0x00000000-0x007fffff: 0x51f27
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
FAILED at 0x0000000c! Expected=0xff, Found=0xbf, failed byte count from 0x00000000-0x007fffff: 0x511f3
ERASE FAILED!
Reading current flash chip contents... done. Looking for another erase function.
Looking for another erase function.
Looking for another erase function.
No usable erase functions left.
FAILED!
Uh oh. Erase/write failed. Checking if anything has changed.
Reading current flash chip contents... done.
Apparently at least some data has changed.
Your flash chip is in an unknown state.
Please report this to the mailing list at flashrom@flashrom.org or
on IRC (see https://www.flashrom.org/Contact for details), thanks!
```

So i hope there's a hint from you guys or, please provide the stock top rom, so i can try save this laptop.

tlaurion, 2024-03-23:
> I use all the chip available but no success (everything that comes with MX25L64*).

> root@debian:/home/user/Downloads# flashrom --programmer ch341a_spi -c MX25L6406E/MX25L6408E -w heads-t430-hotp-maximized-v0.2.0-2038-gebdcc85-bottom.rom flashrom unknown on Linux 6.1.0-18-amd64 [...]

> So i hope there's a hint from you guys or, please provide the stock top rom, so i can try save this laptop.

The behavior you see is typical of a clip wearing out: the clamp slowly slides out and eventually looses grip where erase function fails and connection being loose fails to detect correctly and different erase functions are attenpted; same happening with detection where if I recall well, generic chips are stated last.

I would recommend changing your clip, I doubt your ch341a is at fault.

You indeed have to flash both top and bottom images which are spitted from the whole 12mb image to be flashed externally. Flashing top is indeed insufficient.

Closing as I can't fix this: this is a mechanical problem you are facing that needs a clip replacement. You do not see the same behavior on the top chip simply because the loosening of the clip doesn't happen in the smaller time lapse needed to flash 4mb as opposed to 8mb on a slow link.

tlaurion, 2024-03-23:
Oh.

And as documentation specify, you want maximized version. Choice here is between dgpu or not, hotp or not.

If that is unclear from doc, please open issue under heads-wiki repo, this is not code related but would be documentation related therefore heads-wiki related.

closed 2024-03-23 by tlaurion

heartbl3ed, 2024-03-23:
will using stock top rom will fix the problem, do you have it as backup ? can i try ? i just want to make sure the board isnt faulty

tlaurion, 2024-03-24:
> will using stock top rom will fix the problem, do you have it as backup ? can i try ? i just want to make sure the board isnt faulty

I don't. Guide specifies to backup first. Change clip and flash bottom rom.