linuxboot/heads #1627: Addition of "Commercial Support" Section to Documentation

pietrushnic, 2024-03-31:
**Is your feature request related to a problem? Please describe.**

The current problem is the lack of easily accessible information for users and businesses interested in obtaining commercial support for the Heads. As a company interested in deploying Heads across various systems, the challenge lies in identifying reputable companies or individuals who specialize in supporting, customizing, or deploying Heads, which may limit adoption.

**Describe the solution you'd like**

I would like to add a new section within the Heads documentation dedicated explicitly to listing companies and consultants that provide commercial support for the Heads. This section should be easily accessible from the main documentation page and include:

- A brief introduction explaining the purpose of the section.
- A list of companies and individual consultants, along with a short description of the services they offer related to Heads (e.g., customization, deployment, support).
- Contact information or links to their respective websites for further details.

This information could be modeled after the structure found in the coreboot commercial support page (https://coreboot.org/consulting.html), the LVFS consulting page (https://fwupd.org/lvfs/docs/consulting), and the eLinux experts page (https://elinux.org/Experts), which effectively provide similar information for their ecosystems.

**Describe alternatives you've considered**

The section in README?

**Additional context**

Adding a commercial support section would not only aid businesses and individuals looking for professional assistance but also potentially increase the adoption of Heads. It can also drive a professional ecosystem around the project, contributing to its sustainability and growth.

JonathonHall-Purism, 2024-04-01:
Agree @pietrushnic , commerical support is an important part of the ecosystem, IMO.

tlaurion, 2024-10-22:
@pietrushnic @JonathonHall-Purism : https://github.com/linuxboot/heads-wiki/pull/164 satisfying as a base?
@JonathonHall-Purism you want to add Purism as part of that PR?