linuxboot/heads #1633: coreboot (rom stitching) not reproducible

tlaurion, 2024-04-04:
```
user@heads-tests-deb12-clean:~/heads$ wget -q https://output.circle-artifacts.com/output/job/b7236410-3202-441e-aa15-fedf09627fb6/artifacts/0/build/x86/librem_15v4/hashes.txt -O hashes.txt
user@heads-tests-deb12-clean:~/heads$ rm -f ./bad_hashes ./identical_hashes ./output_non_reproducible && grep / hashes.txt | awk -F " " '{print $1}'| while read hash; do grep -q $hash ~/heads/build/x86/librem_15v4/hashes.txt; if [ $? == 0 ]; then echo $hash >> ./identical_hashes; else echo $hash >> ./bad_hashes;fi;done; cat ./bad_hashes| while read bad; do grep $bad hashes.txt >> ./output_non_reproducible; done;
user@heads-tests-deb12-clean:~/heads$ cat output_non_reproducible 
77d95bf358a3fede66dded0a77e4279eae9dbb6490133cf761ead12984b2ade3  /root/heads/build/x86/librem_15v4/heads-librem_15v4-v0.2.0-2050-g8208c86.rom
```

Ok opening another issue on coreboot reproducibility. My insight is that we modify coreboot build options compared to what coreboot tries to do reproducibily, keeping maybe not needed generally applied customizations that should only exist per board configs for old version (coreboot 4.11 artifacts maybe).

@JonathonHall-Purism this PR is ready for review, other traces above not relevant.

_Originally posted by @tlaurion in https://github.com/linuxboot/heads/issues/1630#issuecomment-2038101225_

tlaurion, 2024-04-04:
Only insight of the moment is that we hardcode flags passed down to coreboot when rom is built:
`EXTRA_FLAGS ?= -fdebug-prefix-map=$(pwd)=heads -gno-record-gcc-switches -Wno-error=packed-not-aligned -Wno-address-of-packed-member`

tlaurion, 2024-04-04:
Diffoscope doesn't help :(

```
user@heads-tests-deb12-clean:~/heads$ docker run --rm -t -w $(pwd) -v $(pwd):$(pwd):ro registry.salsa.debian.org/reproducible-builds/diffoscope heads-x230-hotp-maximized-v0.2.0-2050-g8208c86.rom build/x86/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-2050-g8208c86.rom 
Unable to find image 'registry.salsa.debian.org/reproducible-builds/diffoscope:latest' locally
latest: Pulling from reproducible-builds/diffoscope
5fd0f220e694: Pull complete 
1812f988493a: Pull complete 
8876f5223738: Pull complete 
4f4fb700ef54: Pull complete 
Digest: sha256:415e41f9f20f31a28543a5b068ce2598ed61db38b177bbbb9670c91a7a474e0f
Status: Downloaded newer image for registry.salsa.debian.org/reproducible-builds/diffoscope:latest
--- heads-x230-hotp-maximized-v0.2.0-2050-g8208c86.rom
+++ build/x86/x230-hotp-maximized/heads-x230-hotp-maximized-v0.2.0-2050-g8208c86.rom
│┄ Command `cbfstool heads-x230-hotp-maximized-v0.2.0-2050-g8208c86.rom extract -n fallback/romstage -f «TEMP»/diffoscope_8r8y6tsx_pmaximized/tmpdog07tsu_CbfsContainer/romstage` failed with exit code 1. (No output)
@@ -28642,228 +28642,228 @@
 0006fe10: 1d37 5956 da86 60fd d095 588d 1a4a 9b78  .7YV..`...X..J.x
 0006fe20: cc3d c32a 9674 92b6 6ad3 8f65 76ef 11b9  .=.*.t..j..ev...
 0006fe30: 252c c935 2ff3 5497 a9c5 1537 d1e2 0537  %,.5/.T....7...7
 0006fe40: 57a5 c451 ffff ffff ffff ffff ffff ffff  W..Q............
 0006fe50: ffff ffff ffff ffff ffff ffff ffff ffff  ................
 0006fe60: ffff ffff ffff ffff ffff ffff ffff ffff  ................
 0006fe70: ffff ffff ffff ffff ffff ffff ffff ffff  ................
-0006fe80: 4c41 5243 4849 5645 0000 0d25 0000 0050  LARCHIVE...%...P
+0006fe80: 4c41 5243 4849 5645 0000 0d2c 0000 0050  LARCHIVE...,...P
 0006fe90: 0000 0020 0000 0030 636f 6e66 6967 0000  ... ...0config..
-0006fea0: 4243 5a4c 0000 0010 0000 0001 0000 29c0  BCZL..........).
-0006feb0: 0100 3000 00c0 2900 0000 0000 0000 1188  ..0...).........
+0006fea0: 4243 5a4c 0000 0010 0000 0001 0000 29de  BCZL..........).
+0006feb0: 0100 3000 00de 2900 0000 0000 0000 1188  ..0...).........
 0006fec0: 8a95 ee5b 5f2d de0a e00f 3bdc bda4 1e19  ...[_-....;.....
 0006fed0: be28 21a5 4c09 3966 25c0 f57a 548f 9848  .(!.L.9f%..zT..H
 0006fee0: 625d dd04 8f62 6d70 57ce 73fe a205 9cfb  b]...bmpW.s.....
 0006fef0: 90d0 6835 82c8 1680 d919 d778 c339 bdc9  ..h5.......x.9..
 0006ff00: cfc9 d76d ea91 6b6d b9ad dc0c dfb6 e16b  ...m..km.......k
 0006ff10: 2d56 d380 b58b b03f 6548 0695 7c35 eb66  -V.....?eH..|5.f
 0006ff20: 1647 eaeb 6f4a 85ad 6cad 58ec 9c72 e519  .G..oJ..l.X..r..
 0006ff30: 2e80 a370 82d0 12a7 42f3 b41a 08ff ded2  ...p....B.......
 0006ff40: 0481 45ce d63c 0be9 64be cd86 1244 537f  ..E..<..d....DS.
 0006ff50: 11dc 7ff5 5218 e768 6a08 239f d1a7 74c0  ....R..hj.#...t.
-0006ff60: 6ce3 9e39 4f50 ee8f 6ea5 2902 18ae 61c5  l..9OP..n.)...a.
-0006ff70: 710a d929 50a8 59b5 25fc d59c 1095 5256  q..)P.Y.%.....RV
-0006ff80: eb44 0ee1 8773 fa14 284b 7fd6 7a6a 1e1a  .D...s..(K..zj..
-0006ff90: 739f 06a2 005a d4d4 b0e0 e37e 3d65 5282  s....Z.....~=eR.
-0006ffa0: 2d06 2faf 29bd e3c0 b07c eecf 0dbf 055e  -./.)....|.....^
-0006ffb0: b4d4 2ce8 5659 44b1 9d1e 9dfd 5a31 15f7  ..,.VYD.....Z1..
-0006ffc0: afed 8a9e 0912 f1e0 c394 1a08 b4cf e0ab  ................
-0006ffd0: c929 e101 316f 1c42 21c8 996e 7ae9 a155  .)..1o.B!..nz..U
-0006ffe0: d4c9 d804 73c4 7a33 2429 95fd 91e9 223d  ....s.z3$)...."=
-0006fff0: 76df 8edb 4f19 a889 8f4e 60db 7934 7676  v...O....N`.y4vv
-00070000: 39ba b980 fb2c df58 2b2f e5e0 6356 563e  9....,.X+/..cVV>
-00070010: 268d 49f8 79cb 2118 e354 efbf a1c9 e342  &.I.y.!..T.....B
-00070020: 5b5c f14a 30c1 be08 f0ed cac2 9500 34b2  [\.J0.........4.
-00070030: e695 b0e2 d2eb 4ff5 e1fc 4856 5fff 418c  ......O...HV_.A.
-00070040: b9d9 532a f768 fa8b 2cbe 72fe 7220 9b4f  ..S*.h..,.r.r .O
-00070050: c5ff 4a1a 1a20 393e 7f29 d3a7 9997 21bc  ..J.. 9>.)....!.
-00070060: 4e96 f143 f9fc d240 7656 577d 812e 2313  N..C...@vVW}..#.
-00070070: 9c01 d30d 9750 60d4 4b2a 2217 0401 6008  .....P`.K*"...`.
-00070080: d2d4 930c ea23 63a8 511f fa04 5729 f9c2  .....#c.Q...W)..
-00070090: ec72 287d 32b0 b859 2c32 8a4a cd30 23c3  .r(}2..Y,2.J.0#.
-000700a0: 5ed5 2aad 693e 7837 b258 5093 ba42 f894  ^.*.i>x7.XP..B..
-000700b0: 5167 9006 1e9c 4d9f e6ae 53f1 017d 0c30  Qg....M...S..}.0
-000700c0: 15d9 5e85 c8f3 2421 e0d7 ed3b 6c50 e6b6  ..^...$!...;lP..
-000700d0: c95f fefb ef10 e155 4d54 d15d 76b7 c55d  ._.....UMT.]v..]
-000700e0: 62b7 9de9 4f5c 11b7 cf9c 075a 724d 149b  b...O\.....ZrM..
-000700f0: 7842 77d7 3c58 60b4 f020 09dc 0d4a 9680  xBw.<X`.. ...J..
-00070100: 3fd6 f22f c956 980f 6126 1f12 4dfc b73e  ?../.V..a&..M..>
-00070110: 8591 fc60 93b3 a5a0 fe63 3c9d 4bb9 1df5  ...`.....c<.K...
-00070120: c4c9 2809 b50f d2be 0206 9ad5 a7fd 175b  ..(............[
-00070130: 18ae 0454 6c55 e72a d032 9cfd b31d 53d1  ...TlU.*.2....S.
[... 353 lines cut ...]
+00070ad0: b82e cd05 d43d eebf b045 0ebb 0abd 9c1e  .....=...E......
+00070ae0: 32db a7c4 23a4 1830 24f2 6658 45ff 3cb5  2...#..0$.fXE.<.
+00070af0: 8919 780d bb9e 4446 7c67 a08a a5d7 f1d0  ..x...DF|g......
+00070b00: 4783 c8cc a447 a695 7d79 e652 dbee da54  G....G..}y.R...T
+00070b10: 5abd 50c0 93ed 41fa 1c58 4504 2b9e 4518  Z.P...A..XE.+.E.
+00070b20: 36e8 fbaa 4c84 b098 04d0 eb2f 3795 6980  6...L....../7.i.
+00070b30: 9673 dd11 4694 838f 2a8a f02a 1588 3b11  .s..F...*..*..;.
+00070b40: b958 cdda 87cd 0ecf 2a87 5add 7592 960d  .X......*.Z.u...
[... cut ...]
+00070bd0: c3ed cc02 362c 27b8 88c1 b64e ffff ffff  ....6,'....N....
 00070be0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
 00070bf0: ffff ffff ffff ffff ffff ffff ffff ffff  ................
 00070c00: 4c41 5243 4849 5645 0000 02d4 0000 0050  LARCHIVE.......P
 00070c10: 0000 0000 0000 0024 7265 7669 7369 6f6e  .......$revision
 00070c20: 0000 0000 2f2a 2062 7569 6c64 2073 7973  ..../* build sys
 00070c30: 7465 6d20 6465 6669 6e69 7469 6f6e 7320  tem definitions 
 00070c40: 2861 7574 6f67 656e 6572 6174 6564 2920  (autogenerated)
```

tlaurion, 2024-04-04:
But points to romstage being different which is still a start

JonathonHall-Purism, 2024-04-08:
@tlaurion TL;DR - CI seems to lack the `v5.0.1` tag (and possibly others), causing `git describe --tags --abbrev=7 --dirty` to produce a different version.

If you pick apart the coreboot stages, that appears to be the only difference (with #1630).  E.g.:
```
$ diff -u <(xxd ci/cbfs/fallback/romstage) <(xxd local_clean/cbfs/fallback/romstage)
--- /dev/fd/63  2024-04-08 13:50:28.844606505 -0400
+++ /dev/fd/62  2024-04-08 13:50:28.844606505 -0400
@@ -2067,8 +2067,8 @@
 00008120: dc2e 02ff 382d 02ff 0000 00b0 5468 7520  ....8-......Thu 
 00008130: 4a61 6e20 3031 2030 303a 3030 3a30 3020  Jan 01 00:00:00 
 00008140: 5554 4320 3139 3730 0066 9090 2d48 6561  UTC 1970.f..-Hea
-00008150: 6473 2d76 302e 322e 302d 3230 3530 2d67  ds-v0.2.0-2050-g
-00008160: 3832 3038 6338 3600 2d54 494d 454c 4553  8208c86.-TIMELES
+00008150: 6473 2d76 352e 302e 312d 3939 382d 6738  ds-v5.0.1-998-g8
+00008160: 3230 3863 3836 0090 2d54 494d 454c 4553  208c86..-TIMELES
 00008170: 532d 2d4c 4553 5354 494d 452d 0066 9090  S--LESSTIME-.f..
 00008180: 5145 4d55 0066 9090 0000 1000 0000 2000  QEMU.f........ .
 00008190: 0000 8000 6690 6690 b896 0100 0040 0000  ....f.f......@..
```

There is also still a difference in Busybox affecting the initrd also, have not looked into that.

One option could be to get CircleCI to sync the tags if that's possible.  Or we could stop setting a version based on tags - Heads isn't tagging any releases anyway, but I do tag downstream PureBoot releases and might still want the tag in the version downstream.

There could be a middle ground where we only use the tag if it's pointing to this commit (never tag+commit height), but that doesn't seem to be something git-describe will readily do, we'd have to do it ourselves.

tlaurion, 2024-04-09:
Let's revisit once #1630 is merged.

JonathonHall-Purism, 2024-04-09:
Deleting the v5.0.1 tag from our local repos and the tlaurion fork fixed these issues - @tlaurion I think we can close this

closed 2024-04-09 by tlaurion