linuxboot/heads #1639: Have flashrom EC firmware flashing

tlaurion, 2024-04-16:
flashrom needs to support ite_ec programmer to be able to flash ec firmware internally on nv41

SRC: https://docs.dasharo.com/unified/novacustom/initial-deployment/#installing-dasharo

tlaurion, 2024-04-16:
- if goal forward is to be able to also flash firmware through fwupd, then qubesos would also need to fork and provide their own flashrom outside of the one provided by fc37
- firmware package upgrades (zip) will need to provide also ec firmware
  - not sure how to do that. Is Heads supposed to also build EC firmare to provide it?
  - If so, a board config will also need to keep track of EC firmware that matches what is expected from firmware
  
@marmarek @pietrushnic am I missing something on the strategy level?

macpijan, 2024-04-16:
> flashrom needs to support ite_ec programmer to be able to flash ec firmware internally on nv41
> SRC: https://docs.dasharo.com/unified/novacustom/initial-deployment/#installing-dasharo

Unfortunately it is more convoluted. The flasrom `ite_ec` programer works with proprietary EC firmware only. So it allows us to transition from proprietary to open-source EC firmware. But from this point on, flashrom `ite_ec` plugin is pretty much useless unfortunately, as the system76ec we base on uses different update protocol than the proprietary one.

macpijan, 2024-04-16:
We have a bit different proposal from @mkopec here: https://github.com/Dasharo/coreboot/pull/481

The EC firmware to be updated shall be stored in the coreboot binary. Then, the coreboot itself will sync (update) the firmware from main flash into EC flash. So no need for `ite_ec` or any other external tools, as the EC update protocol is implemented in coreboot.

tlaurion, 2024-04-16:
Rwnamed issue on goal, not on its strategy

tlaurion, 2024-04-22:
@macpijan needed under https://github.com/linuxboot/heads/issues/1647

fhvyhjriur, 2024-05-05:
The topic here looks general about EC update. Is this also about thinkpad EC? There its known for people to realize at some point, that new OEM batteries are not really available and they need to patch the EC for the replacement battery they got.
Also there are issues if you have not updated the lenovo bios ever before and the outdated EC from the first lenovo bios releases is known to make problems with coreboot.
The most common way to make a thinkpad ready for flashing coreboot is this project here: https://github.com/hamishcoleman/thinkpad-ec
If you missed this step before flashing coreboot, in general you are stuck at reflashing the lenovo OEM bios file and then lock your SPI again. And then you have again to open up your device once again and redo everything or you have to learn how to remove the write protection in the OEM Lenovo image before writing it to the SPI and this is really terrible documented.

To be able to flash inside heads the EC of the Thinkpad would be some great improvement. If you for example decide ever later to use a x20 keyboard in a x30 Thinkpad you can just change the EC with heads.

tlaurion, 2024-05-05:
> The topic here looks general about EC update. Is this also about thinkpad EC? There its known for people to realize at some point, that new OEM batteries are not really available and they need to patch the EC for the replacement battery they got.

Right. Unfortunately, this doesn't apply to Lenovo. I changed the title. 

> Also there are issues if you have not updated the lenovo bios ever before and the outdated EC from the first lenovo bios releases is known to make problems with coreboot.
> The most common way to make [...]

1vyprep is another way to do this AFAIK. Should be in wiki. 

> If you missed this step before flashing coreboot, in general you are stuck at reflashing the lenovo OEM bios file and then lock your SPI again. And then you have again to open up your device once [...]