linuxboot/heads #1644: Setting a TPM DUK on Ubuntu 24.04 fails at unpacking initramfs

tlaurion, 2024-04-20:
[image]

Otherwise boots on x230 on master so #1641 specific to nv41 @alex-nitrokey @daringer @nestire

tlaurion, 2024-04-20:
@JonathonHall-Purism what on hell is `type 02214c18b266` will be subject to another PR down the path...

JonathonHall-Purism, 2024-04-22:
It's LZ4.  https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/lib/decompress.c#n58

We'd need to ship an LZ4 decompressor (if we don't already), then add it to unpack_initramfs.sh: https://github.com/linuxboot/heads/blob/master/initrd/bin/unpack_initramfs.sh#L67.

tlaurion, 2024-11-22:
Irrelevant under 24.10+ since they switched to zstd which was already supported. Added xz though, needed by debian derivatives (tested tails) on upcoming PR that will close this issue.

PR #1860 (2024-11-22): initrd/bin/unpack_initramfs.sh: add xz unpacking support.
Closes https://github.com/linuxboot/heads/issues/1644 : was preventing to setup TPM DUK key because initrd couldn't be unpacked and repacked on the fly to be passed at kexec call to contain unsealed DUK key for final OS to pick up and use to unlock LUKS on boot.

- reformat with tabs and unify

[95c6eb5](https://github.com/linuxboot/heads/pull/1860/commits/95c6eb5c498bebc028cd92d62c83e33c2be3ed2e):
- adds xz unpacking that was missing and needed for debian derivatives.
- other format added in comments so easy to test when testing candid [... cut]
closed 2024-11-22 by tlaurion