linuxboot/heads #1647: Add ability to configure some important coreboot config settings

UndeadDevel, 2024-04-21:
**Is your feature request related to a problem? Please describe.**
See https://github.com/Dasharo/dasharo-issues/issues/285 (coreboot allows configuring battery charging limits both from the BIOS as well as the OS); this used to work in Heads 2.1 for me as well on my NV41.

See https://github.com/Nitrokey/heads/issues/35 (Setting battery charging limits worked in NK Heads 2.1, but not anymore in NK Heads 2.4 and 2.4.1). In that issue @daringer says that it's an upstream issue with Heads.

**Describe the solution you'd like**
Allow configuring battery charging limits (and possibly other coreboot configurable settings) from within Heads or the OS.

**Describe alternatives you've considered**
Unfortunately right now there doesn't seem to be an alternative other than removing Heads and using Coreboot (or going back to Heads 2.1). The usually suggested solutions online, e.g. `tlp` or `smbios-utils` all require that the firmware make the relevant system files, usually `/sys/class/power_supply/BAT0/charge_control_start_threshold` and `/sys/class/power_supply/BAT0/charge_control_end_threshold`, available to the OS, as that is how those utilities set the new limits.

**Additional context**
The focus is on the battery charging limits, as that's not just a convenience feature, but also important to help preserve battery life for laptop users.

tlaurion, 2024-04-22:
@UndeadDevel im confused.

If I understand this correctly, the feature is not implemented by coreboot, nor ed2k, but the EC firmware. Coreboot is responsible to expose acpi tables of EC, and then the system76_acpi is modprobed and if exposed acpi tables are exposed, exposes the additional thresholds under BAT0 which tlp can interact with at runtime?

tlaurion, 2024-04-22:
Most probable solution https://github.com/Dasharo/coreboot/pull/481 and then integration under heads:

- include functional system76_acpi kernel module able to interact with EC
- create dynamic menus, probably on presence of kernel module packed under modules.cpio or by export in board config

@UndeadDevel thanks for opening issues upstream here and linking to downstram issues.

tlaurion, 2024-04-24:
That's https://github.com/Dasharo/dasharo-issues/issues/820