linuxboot/heads #1648: HOTP mismatch when booting computer without HOTP USB Security dongle 5 times in a row

tlaurion, 2024-04-21:
An old customer recontacted me today again for this issue which is kinda cryptic and doesn't make sense.

The hotp counter is not related in any way to the TPM counter, is a file under /boot which is not part of the kexec.sig file because kexec* not being kexec*.txt (normal since it changes every boot) and its purpose is unclear.

I would have swore there was an issue opened for that before. But couldn't find it.

Was even countacted a while back for a security report since that counter could be changed (+1/-1) and decreased to be falsified without notice.

What is the purpose of that counter today? Should we remove it?
I remember vaguely a discussion with @JonathonHall-Purism on that, probably off-channel.

But if there is a purpose to that counter, then it should be accompanied with a really big warning to the user reminding him that if the computer is booted more then 5 times in a row, the computer will be confused, since local increments not matching counter on HOTP dongle anymore.

Users are then confused on what to do.

I also vaguely recall that in most OEM setups/rebrandings, the TPMTOTP secret is not shown as QR code to be scanned by end users. So if thsi is the case and then HOTP counter falls out of sync, what is the end user supposed to do? Reset TPM since he cannot validate against his phone?

The only suggestion to those users today (not having TPMTOTP Qr code scanned) is to flash a clean firmware image and  regenerated HOTP from option menus.

But that is a mitgation to an implementation flaw. I think I also remember a discussions with @JonathonHall-Purism about only incrementing the counter when HOTP is actually verified, and not on every boot? Otherwise the UX is missing the big fat warning resulting is way too many questions about that.

@jan23 @daringer @JonathonHall-Purism @wessel-novacustom: what SHOULD be the good behavior here? My take is increment HOTP only when HOTP dongle is connected and continue to do what we do today, limiting user support requests, or discuss on what wwe throught this counter did, what we want it to do in 2024 and make it do what it should do.

tlaurion, 2024-04-22:
@JonathonHall-Purism problem is that since we closed that issue, we now offer the possibility of nitnshowing TPMTOTP qrcode for second channel verification, where HOTP is supposed to be used and connected at each boot. I user doesn't want to use it for one boot bale cause it's too far away, fine.

But all things considered I agree with
> @kylerankin said:
> > we do warn the user when they boot without a dongle inserted now. If that needs to be amended to make it clear that it will eventually throw off the HOTP counter I'm ok with that as long as the warning message is clear
> 
> and
> 
> > we warn at every boot that the key isn't inserted, but the user can dismiss the warning and boot anyway. But that warning doesn't go into detail about the consequences of that decision (ie. that ultimately HOTP counter will fall out of sync)
> 
_Originally posted by @tlaurion in https://github.com/linuxboot/heads/issues/1266#issuecomment-1589905676_

JonathonHall-Purism, 2024-04-22:
@tlaurion I think we should stop incrementing the counter if the HOTP dongle is not present.  I do not think it is an effective measure of detecting that someone tried to boot the system for the reasons already listed.  It's not secure, and even when it does appear it can't be reasonably differentiated from actual tampering.

Since it's not secure or reliable, it's IMO an implementation problem, which doesn't merit a warning when we can actually fix it.  (Why warn about "sorry our implementation is bad, if you do this 5+ times something will go wrong" when we could eliminate the problem and warning entirely?)

tlaurion, 2024-04-22:
> @tlaurion I think we should stop incrementing the counter if the HOTP dongle is not present. I do not think it is an effective measure of detecting that someone tried to boot the system for the [...]

Is #1650

wessel-novacustom, 2024-04-28:
I agree with @JonathonHall-Purism