linuxboot/heads #1651: Make TOTP (+ HOTP) verification possible **without OS** installed

tlaurion, 2024-04-22:
It would be nice if we could have TOTP (+ HOTP) verification possible **without OS** by going through the OEM Factory Reset / Re-Ownership options. It would allow us to have solid root-of-trust from us (OEM) to the end user, even if no OS was chosen. A lot of users order Heads without OS.

From our side, credentials are then separately delivered via different channels (physical paper, email and Signal). Ideally, credentials should be randomly generated.

_Originally posted by @wessel-novacustom in https://github.com/linuxboot/heads/discussions/1521#discussioncomment-9083965_

tlaurion, 2024-04-22:
This is currently limited by several factors, one being that HOTP is stored under /boot as recently reviewed under #1650 

@JonathonHall-Purism shared some ideas off-channel.

Might want to chip them in in the referred discussion at https://github.com/linuxboot/heads/discussions/1521#discussioncomment-9083965

JonathonHall-Purism, 2024-04-23:
The idea I shared was #1480 - that is, replace reverse HOTP with a TPM signature on the PCR state (=firmware state) and a nonce from the token.  That would eliminate the HOTP counter, which is currently stored in /boot and is the reason HOTP doesn't work without an installed OS.

The token would need to support such an operation though.

As a stopgap we could find another place to store the HOTP counter.  This would be compatible with existing tokens but loses another benefit of #1480 that the attestation secret never leaves the TPM.
* Could we use a TPM counter?  I think this was considered long ago and seems to have been rejected, I'm not sure why
* Could we use flash space?  E.g. coreboot smmstore or something specific to this counter.  We would want to consider how often sectors get erased so we don't wear out flash.  In principle if we dedicated a whole sector to this counter, we should get thousands of counts per erase at least.
* Could we store it on the HOTP token itself somewhere?

tlaurion, 2024-04-24:
@daringer @jans23 on previous @JonathonHall-Purism comment?

@JonathonHall-Purism looking at git blame on currently commented code under seal-hotp gives insights on past and reverted implementation. I only Remeber vaguely but that didn't work on TPM1. Will dig later

tlaurion, 2024-04-24:
@JonathonHall-Purism

https://github.com/linuxboot/heads/commit/fe34aba719073ef3710b25627b0acbbb5236815d

gaspar-ilom, 2026-10-05:
I would like to throw a new idea into the ring:

We keep using reverse TOTP/HOTP but make the secret non-extractable:

1. Generate the HOTP secret in Heads.
2. Store it in Nitrokey.
3. Import it to the TPM with `-a 'fixedtpm|fixedparent|sign|userwithauth` to ensure it cannot be extracted/duplicated. The secret should be accessible under the same PCR states as currently.
4. Directly compute the HMAC-SHA1 in the TPM using the respective counter for HOTP/TOTP. No need really to do this in Heads.

I would propose to store the HOTP counter in the NV storage and use NV_increment.
It seems that this would not work with the TPM 1.2 boards as per https://github.com/linuxboot/heads/commit/fe34aba719073ef3710b25627b0acbbb5236815d
If we really need multiple counter increments, I think the best option for these boards is keeping the counter on /boot but move it to the TPM for TPM2 boards.

> TPM monotonic counter was causing conflicts with the Heads configuration counter as TPM 1.2 can only increment one counter per reboot.

Unlike #1480 this approach does not depend on an update of the Nitrokey firmware as we just keep using reverse-HOTP, which is already supported, but we make it significantly harder to extract the secret and accidental leaks compared to the current situation. 

The way I understand the current situation:
* the only thing that prevents an attacker from extracting the HOTP secret are the TPM PCR states. So they cannot change the firmware or do anything from a booted OS, but have to work with what Heads allows them to do
  * this seems brittle, because as soon as an attacker finds a way to get a shell or run their code without having the PCR extended, they can extract the secret
  * omitting extending the pcr in some codepath is a very much conceivable bug.
* once you have the HOTP secret you replace the entire chain of trust
  * flash a different firmware
  * reset the TPM and reseal the secret
  * use the secret and the public readable counter to make HOTP verification pass
  * replace the public key with an attacker controlled one etc. 
* This would only be detected once the user manually verifies signatures using the Nitrokey.