linuxboot/heads #1653: Improve TPM DUK resealing UX

UndeadDevel, 2024-04-23:
**Is your feature request related to a problem? Please describe.**
After certain dom0 updates it becomes necessary to reseal the TPM DUK (e.g. new kernel, so boot option list changed). The following secrets must be entered in the following order to make this happen:
1. LUKS DRK passphrase
2. new TPM DUK passphrase (with validation check, so needs to be entered exactly the same way twice)
3. TPM Ownership passphrase
4. User GPG PIN

AFAICT if any one of those is not entered correctly (not sure about the last one, but this is definitely true for 1-3) then the entire process will fail and has to be started over, which is pretty bad UX as we're talking about a lot of complex secrets.

**Describe the solution you'd like**
Similarly to what was implemented in https://github.com/linuxboot/heads/pull/1595, i.e. allow multiple tries, at least for the LUKS DRK, new TPM DUK and User GPG PIN (ideally for TPM Ownership passphrase as well, but that may be problematic regarding rate-limiting by the TPM).

**Describe alternatives you've considered**
At the very least my suggestion in the other review should be implemented (move the validation of the new TPM DUK passphrase up to occur before the LUKS DRK passphrase entry). But really 1, 2 and 4 in above list should allow multiple tries.

tlaurion, 2024-04-24:
@UndeadDevel should this be part of the PR? XN you propose changes? I'm stalled under too many tasks. I would cherry pick your commits.

UndeadDevel, 2024-04-24:
@tlaurion Maybe this weekend or next week, but can't promise anything.

UndeadDevel, 2024-05-17:
@tlaurion PR #1678 is a partial fix for this.

What's left is allowing multiple tries for incorrect TPM Ownership passphrase and GPG User PIN. If this is already provided for in the new Heads version (current WiP) then this issue would be complete after merging #1678.