linuxboot/heads #1656: Can’t scan QR code with Authenticator app

PeterLacknase, 2024-04-27:
## Please identify some basic details to help process the report

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

NovaCustom NV41

**2. Does your computer have a dGPU or is it iGPU-only?**
- [ ] dGPU
- [X] iGPU-only

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [X] Other provider: NocaCustom
- [ ] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [ ] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [X] Other: Nitrokey 3A mini

**5. Are you using the PGP key to provide HOTP verification?**
- [X] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ ] First-time flash
- [ ] Updating heads 
- [X] Neither (AFAICS)

## Please describe the problem

**Describe the bug**
A clear and concise description of what the bug is.

Choosing -> Options -> TPM/TOTP/HOTP Options -> Generate new TOTP/HOTP secret

the shown QR code can‘t be scanned with the otherwise and anywhere perfectly functioning Authenticator app from Matt Rubin. Cf.:

https://github.zz/mattrubin/Authenticator

Adding the code manually works fine, but is quite tedious.

**To Reproduce**

See above.

**Expected behavior**

Point - click - done.

**Screenshots**

None.

**Additional context**

None.

tlaurion, 2024-04-28:
> Choosing -> Options -> TPM/TOTP/HOTP Options -> Generate new TOTP/HOTP secret\n\nthe shown QR code can‘t be scanned with the otherwise and anywhere perfectly functioning Authenticator app from Matt Rubin. Cf.:\n\nhttps://github.zz/mattrubin/Authenticator\n\nAdding the code manually works fine, but is quite tedious.

@PeterLacknase :
The format used inside of the qrcode is a qrencode direct string rendering of the text shown below the qrcode. It includes the board name as well, which in other app is translated in title of the TOTP entry when the qrcode is scanned and it's content is interpreted correctly. 

This works in Aegis, Google authenticator, keepassdx and a lot of other apps. This is the first time someone reports an app not functioning, leading to think that the qrcode in question includes content that the Authenticator app you are using doesn't support the Url format used in the qrcode to easily create named entries under other apps. 

What you are reporting here seems to need to be reported as an issue at https://github.com/mattrubin/Authenticator, not here, giving them a clear example to reproduce and support fully the standard of what works in other apps but not theirs?

Here is a TPMTOTP example to be referred without privacy/security implications that can be used as a provided example:
[image]

tlaurion, 2024-04-28:
There are other TOTP apps out there as well.
Unless something is proved wrong in URL itself, encoded in qrcode, the qrcode content is fine and entering the secret manually also proved to work.

Closing this issue as unplanned since more compliant and more maintained TOTP applications exist for IOS ecosystem. This one doesn't seem maintained with no commit for a year.

I would suggest to use an alternative app that will work.

If you disagree with this decision, feel free to tag me and this issue will be reopened.

closed 2024-04-28 by tlaurion