linuxboot/heads #1663: Add support for Acer Chromebook 515 Plus (CB515-2H) / Google Omnigul Platform

mdrobnak, 2024-05-05:
Add support for the Acer Chromebook 515 Plus - CB515-2H, also known as the Google Omnigul / Brya.

- Already runs Coreboot
- Intel iGPU "UHD Graphics"
- 12th Gen Intel Core i3-1215U
- Fixed 8 GB RAM
- Intel AX211 Wi-Fi 6E
- 32MB Winbond chip, not sure of the model at the second.

Work already being done in https://github.com/linuxboot/heads/pull/1658.

mdrobnak, 2024-05-15:
[Lifecycle Extension Guide](https://global-download.acer.com/GDFiles/Document/Lifecycle%20Ext.%20Guide/Lifecycle%20Ext.%20Guide_Acer_1.0_A_A.pdf?acerid=638320940119709511&Step1=&Step2=&Step3=CB515-2H&OS=ALL&LC=en&BC=ACER&SC=PA_6) from Acer - Shows how to disassemble parts of the laptop. [

Flash chip is on underside of board.

WSON8 6x8 mm probe necessary to unlock flash descriptor.

flashrom in Ubuntu 23.10 can reach chip using external programmer.

ifdtool used to unlock flash descriptor and enable HAP bit.

flashroom used to flash modified descriptor and machine boots fine.

tlaurion, 2024-11-16:
@mdrobnak can you Tldr what is your understanding if ehst is kidding for tpm compatibility which is the last blocker here?