linuxboot/heads #1664: Coreboot and Libreboot newly supported HP EliteBook 820 G2 sometimes soon in heads?

fhvyhjriur, 2024-05-05:
There are few people who do not like the screen of the X230 and are also not happy with the CPU speed and other things. Some people have build because of this the eDP screen coreboot images for the x230 to use modern screens. Also some people soldered a quadcore CPU into the X230. 
The X240 that have already the modern display connector never got support by coreboot and for a better screen you have to take a bigger T440p or even go big and heavy like the W541. There was no good solution for people who need a small and portable device without heavy modifications required to the X230.

Since some months there is now support for the HP EliteBook 820 G2. Sadly it involves one more binary blob that is required to run and have of course not a Thinkpad keyboard. But maybe when more people get to this CPU platform, there would be someone who would reverse engineer the binary blobs. And hopefully the Thinkpad series with the same CPU get supported by coreboot also at some time later.

To get more people into this CPU generation that could potentially be able to run with as much free software as the x230, are there any plans adding heads support for this device? The build in TPM chip there is the SLB 9660.

More information available here: 
https://libreboot.org/docs/hardware/hp820g2.html
https://github.com/coreboot/coreboot/tree/main/src/mainboard/hp/elitebook_820_g2
https://review.coreboot.org/c/coreboot/+/46630

tlaurion, 2024-05-05:
@fhvyhjriur i followed and this is amazing work.

A reminder: Heads project is community effort driven.

The general idea here is that a more technical board owner is responsible to try to learn how to port a coreboot existing and maintained upstream port to Heads. There are multiple examples of such boards in the current board directory, on which git blame feature (github permits to see this visually) will lead you to respective commits and pull requests where effort was made to bring those boards on Heads.

That board owner needs to be motivated at least minimally to engage in such port for his own personal or business interests. That board owner is then responsible to become a willing tester when important changes happen under Heads so that the pull requests, maybe causing regression or bricks as opposed to master are tested before merge. 

The boards marked UNTESTED under the board directory will also link to pull requests that were untested by such previously responsible to test those boards by personal interests. 

The unmaintained directory shows boards that were moved there because untested for way too long by interested board owners, requiring way too much time from me, the maintainer, to poke people around before merging pull requests, which otherwise are rebased, conflicts needing resolving etc. So they were loved to unmaintained until someone personally interested jumps in and is willing to collaborate into bringing that board back maintained and tested under Heads. 

I have nothing against bringing platforms like without discrete TPM (dTPM) that would use Basic boot mode (not so relevant to be brought to Heads honestly, but if one is motivated to consent into testing in time, it will be maintained. 

The same applies to any board. Including this proposition. 

You lurked into Heads for a long time. You seem to have time. You seem to own many boards. But. You don't reply in time and don't seem to consent into testing in time in my prior efforts for the x200 etc. Therefore. I wait for a basic PR to be created and give a can to learn how to fish, as opposed to provide unlimited supplies of fish (without money compensation, without code contributions given back, without personal involvement into bringing those platforms supported.) 

If you are not technical enough, then you can contribute in the downstream communities to find such board owner that would consent into bringing that board supported, test pull requests requiring testing because chances of bricking, and then be a happy user of such platform. 

I do not own all the platforms. I don't intend to own all the platforms. And I do not intend to test all the platforms myself. This is why this is a community driven project, with people having personal interests involving themselves to bring those boards supported. 

Does that make sense?

tlaurion, 2024-05-06:
@fhvyhjriur i followed and this is amazing work.

A reminder: Heads project is community effort driven.

The general idea here is that a more technical board owner is responsible to try to learn how to port a coreboot existing and maintained upstream port to Heads. There are multiple examples of such boards in the current board directory, on which git blame feature (github permits to see this visually) will lead you to respective commits and pull requests where effort was made to bring those boards on Heads.

That board owner needs to be motivated at least minimally to engage in such port for his own personal or business interests. That board owner is then responsible to become a willing tester when important changes happen under Heads so that the pull requests, maybe causing regression or bricks as opposed to master are tested before merge. 

The boards marked UNTESTED under the board directory will also link to pull requests that were untested by such previously responsible to test those boards by personal interests. 

The unmaintained directory shows boards that were moved there because untested for way too long by interested board owners, requiring way too much time from me, the maintainer, to poke people around before merging pull requests, which otherwise are rebased, conflicts needing resolving etc. So they were loved to unmaintained until someone personally interested jumps in and is willing to collaborate into bringing that board back maintained and tested under Heads. 

I have nothing against bringing platforms like without discrete TPM (dTPM) that would use Basic boot mode (not so relevant to be brought to Heads honestly, but if one is motivated to consent into testing in time, it will be maintained. 

The same applies to any board. Including this proposition. 

You lurked into Heads for a long time. You seem to have time. You seem to own many boards. But. You don't reply in time and don't seem to consent into testing in time in my prior efforts for the x200 etc. Therefore. I wait for a basic PR to be created and give a can to learn how to fish, as opposed to provide unlimited supplies of fish (without money compensation, without code contributions given back, without personal involvement into bringing those platforms supported.) 

If you are not technical enough, then you can contribute in the downstream communities to find such board owner that would consent into bringing that board supported, test pull requests requiring testing because chances of bricking, and then be a happy user of such platform. 

I do not own all the platforms. I don't intend to own all the platforms. And I do not intend to test all the platforms myself. This is why this is a community driven project, with people having personal interests involving themselves to bring those boards supported. 

Does that make sense? 
But yes, those boards are interesting and pertinent alternatives to Lenovo boards for all reasons you mentioned.

Now wainting for such board owner to be willing to resolve open issue with above mentioned social contract.