linuxboot/heads #1672: Enable Lenovo Thinkpad T530-unmaintained-build for testing and reporting functionality of build

fhvyhjriur, 2024-05-13:
I bought for heads-testing a T530 like i announced it here https://github.com/linuxboot/heads/issues/925#issuecomment-2094795637 . 
T530 was supported by heads but was not tested for longer time and thats why it got disabled. Now there are no actual builds for the T530 to test it again.
@tlaurion 
Could you enable the T530-maximized images to test? You have done it few days ago for the kgpe-d16 here https://github.com/linuxboot/heads/commit/e4976e7882992c6b5182d2593f5d34bd4028e229

I could then flash the bottom and top image to the two spi chips of the T530 and report if its working.

tlaurion, 2024-05-13:
@fhvyhjriur see #1673

tlaurion, 2024-05-13:
And then, since the commit from where that board became untested and then unmaintained, check what changed in board config compared to its x230-maximized board equivalent under board config and make sure everything is in sync even if it was untested.

Report your findings. Iterate.

if this one works, it means that UNMAINTAINED-t530-hotp-maximized should work as well.

On top of my head, HOTP autoboot is missing in board config. 
To iterate with other changes comapring the board config files.
To iterate comparing coreboot config pointed by x230/t530 etc.

tlaurion, 2024-05-13:
[7d67d5a](https://github.com/tlaurion/heads/commit/7d67d5a58bc142ec4028cb36369544b0be050d56) oups. Bad board name.
Fixed under [1035a93](https://github.com/tlaurion/heads/commit/1035a93e79866b5483c5a35c3697d08fee4f8970)

fhvyhjriur, 2024-05-20:
Thanks for the images. Test report can be found here: https://github.com/linuxboot/heads/issues/1682

PR #1673 (2024-05-29): Build unmaintained t530 maximized
Builds UNMAINTAINED_t530-maximized for 

---

closes #1672 without putting it back as maintained for now.

@fhvyhjriur this is a single change showing you how to apply past work on top of master and test things for yourself doing such PR in the future for boards you want back in the build system. 

Doing such changes in your branch, combined with https://github.com/linuxboot/heads-wiki/issues/119 should get you going to bring back x200. Including having CircleCI build your branches, testing reproducible rom images and propose changes, as well as to collaborate [... cut]
closed 2024-05-20 by fhvyhjriur