linuxboot/heads #1674: OEM Factory Reset GPG user authentication: key_to_card fails with « invalid time »

headscontrib, 2024-05-13:
## Please identify some basic details to help process the report
Hi,

When setting GPG user authentication on x230, the oem-factory-reset function fails with « Invalid Time ».

The error is generated when using the keytocard_subkeys_to_smartcard function. 

### A. Provide Hardware Details

**1. What board are you using (see list of boards [here](https://github.com/eganonoa/heads/tree/master/boards))?**

x230-htop-maximized

**3. Who installed Heads on this computer?**
- [ ] Insurgo
- [ ] Nitrokey
- [ ] Purism
- [ ] Other provider
- [* ] Self-installed

**4. What PGP key is being used?**
- [ ] Librem Key
- [* ] Nitrokey Pro 2
- [ ] Nitrokey Storage
- [ ] Yubikey
- [ ] Other

**5. Are you using the PGP key to provide HOTP verification?**
- [* ] Yes
- [ ] No
- [ ] I don't know

### B. Identify how the board was flashed

**1. Is this problem related to updating heads or flashing it for the first time?**
- [ ] First-time flash
- [ *] Updating heads 

**2. If the problem is related to an update, how did you attempt to apply the update?**
- [ ]* Using the Heads GUI
- [ ] Flashrom via the Recovery Shell
- [ ] External flashing

**3. How was Heads initially flashed**
- [ *] External flashing
- [ ] Internal-only / 1vyrain
- [ ] Don't know

**4. Was the board flashed with a maximized or non-maximized/legacy rom?**
- [ *] Maximized
- [ ] Non-maximized / legacy
- [ ] I don't know

**5. If Heads was externally flashed, was IFD unlocked?**
- [*] Yes
- [ ] No
- [ *] Don't know

### C. Identify the rom related to this bug report

**1. Did you download or build the rom at issue in this bug report?**
- [ *] I downloaded it
- [ ] I built it

**2. If you downloaded your rom, where did you get it from?**
- [ *] Heads CircleCi
- [ ] Purism
- [ ] Nitrokey
- [ ] Somewhere else (please identify)

*Please provide the release number or otherwise identify the rom downloaded*

**3. If you built your rom, which repository:branch did you use?**
- [ ] Heads:Master
- [ ] Other (please identify)

**4. What version of coreboot did you use in building?**
- [ ] 4.8.1 (current default in heads:master)
- [ ] 4.13
- [ ] 4.14
- [ ] 4.15
- [ ] Other (please specify)
- [ ] I don't know

**5. In building the rom where did you get the blobs?**
- [ ] No blobs required
- [ ] Provided by the company that installed Heads on the device
- [ ] Extracted from a backup rom taken from this device
- [ ] Extracted from another backup rom taken from another device (please identify the board model)
- [ ] Extracted from the online bios using the automated tools provided in Heads
- [ ] I don't know

## Please describe the problem

**Describe the bug**

When setting GPG user authentication on x230, the oem-factory-reset function fails with « Invalid Time ».

The error is generated when using the keytocard_subkeys_to_smartcard function. 

When replaying the keytocard_subkeys_to_smartcard function, the error pops at line 283/290/294, each time one enters the Admin pin.

After discussing briefly the bug with @insurgo, I redid everything after running network-init-recovery. Same error, which indicates it may not be related to time synchronisation issues.

Looking briefly at gpg code (I'm really not an expert), I find that the the command.c file of https://github.com/gpg/gnupg/blob/master/agent/command.c, on line  3263 may be the one throwing this error: 
```c=
static const char hlp_keytocard[] =
  "KEYTOCARD [--force] <hexgrip> <serialno> <keyref> [<timestamp> [<ecdh>]]\n"
  "\n"
  "TIMESTAMP is required for OpenPGP and defaults to the Epoch.\n"
  "ECDH are the hexified ECDH parameters for OpenPGP.\n"
  "SERIALNO is used for checking; use \"-\" to disable the check.";
static gpg_error_t
cmd_keytocard (assuan_context_t ctx, char *line)
{
...
 err = agent_key_from_file (ctrl, NULL, ctrl->server_local->keydesc, grip,
                             &shadow_info, CACHE_MODE_IGNORE, NULL,
                             &s_skey, NULL, &timestamp);
...
 else if (timestamp == (time_t)(-1))
    timestamp = isotime2epoch ("19700101T000000");

  if (timestamp == (time_t)(-1))
    {
      err = gpg_error (GPG_ERR_INV_TIME);
      goto leave;
    }
```

I tried to understand why this may happen, but I'm lost at the moment.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to oem factory reset
2. Select Yes to « Would you like to format an encrypted USB Thumb drive to store GPG key material?  (Required to enable GPG authentication) [y/N]: » 
3. Select Yes to « Would you like in-memory generated subkeys to be copied to USB Security Dongle's smartcard? »
4. Finish and see error

**Expected behavior**
Oem factory reset should end up nicely.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

tlaurion, 2024-05-14:
@headscontrib was able to reproduce. Seems like culprit is version bump of gpg toolstack under https://github.com/linuxboot/heads/pull/1661

Investigating.

headscontrib, 2024-05-14:
Thanks! I spent a few hours but I could not find anything... I tested GPG user authentication on nv41 - same nitrokey - and issue was not showing on nv41.

tlaurion, 2024-05-14:
> Thanks! I spent a few hours but I could not find anything... I tested GPG user authentication on nv41 - same nitrokey - and issue was not showing on nv41. Also ran with debug logs on x230, nothing helpful on my side.

Again its simply because Nitrokey is using an older heads commit.

tlaurion, 2024-05-17:
@headscontrib 1677 fixes this

closed 2024-05-17 by tlaurion

headscontrib, 2024-05-18:
Thanks very much!