linuxboot/heads #1681: Bring colors to the shell console "'a la" systemd : "[...] doing action" -> "[OK] action" (green) || "[FAILED] action" (red)

tlaurion, 2024-05-17:
**Is your feature request related to a problem? Please describe.**
Heads has: warn, error and die as helpers but users are used to systemd output (liking or not systemd is not the point)

**Describe the solution you'd like**
Heads should include external helpers to unify output and use them so that the UX is uniform from firmware to OS.

**Describe alternatives you've considered**
Reinviting the wheel.

**Additional context**
[edit later with systemd console output on bootpatch of the OS]

- `[...] action`: ideally progress would write to the same console line, overwriting status update to the user, a la "flashrom_progress" we have under flash-gui.sh processing debug output of flashrom (I know, flashrom didn't upstream their internal progress bar as of 1.3.0).
- `[OK] action`: should be green and be eye reassuring to end users (I did a demo to a game designer. He said: ish. You don't have that?
- `[FAILED] action`: should be red and alarming.

All the rest of the current Heads output should be hidden behing DEBUG/TRACE if desired/needed.

tlaurion, 2024-05-17:
Do we know some UX designer that love this project enough to give some love to this project? If so please feel free to contact me over matrix https://osresearch.net/community/#matrix-channel

tlaurion, 2024-05-28:
Intesting library to be considered: https://github.com/Kicksecure/helper-scripts/blob/master/usr/libexec/helper-scripts/log_run_die.sh

To see how it looks: https://github.com/Kicksecure/usability-misc/actions/runs/9228546949/job/25392792206
Click on: Run default dist-installer-cli - Clearnet

----

src: https://forums.whonix.org/t/whonix-linux-installer-development-discussion/15917/182

tlaurion, 2024-05-28:
I have to find another one that even included whiptail+standard console output for prompts, including colors. Wished I took a trace if it back then, will look again.

mdrobnak, 2024-05-28:
I tried setting NEWT_COLORS and that didn't work to at least get some color to the dialog boxes for the text mode. Any idea why there's no color in the text mode NEWT/SLANG build?

tlaurion, 2024-05-31:
> I tried setting NEWT_COLORS and that didn't work to at least get some color to the dialog boxes for the text mode. Any idea why there's no color in the text mode NEWT/SLANG build?

Colors are hardcoded under newt, slang is unrelated.

Note that fbwhiptail depends on Cairo, not newt, to draw on FB instead of console. whiptail is aimed for server headless output mainly, and I use it extensively from qemu-whiptail-tpm(1/2) board configs for local tests, which is way faster then fbwhiptail on qemu tcg (where kvm is unavail under qubesos because of xen->kvm nested virt being impossible today. 

whiptail is deployed for kgpe-d16 server board and talos II only today, as can be seen per board configurations. 

Will have to dig into this, but just wanted to clarify that if you test this on your recent boards, this is fbwhiptail -> Cairo, not whiptail->newt.